networkupstools/nut #732: Incompatible salicru SPS home

javier-izquierdo-visualfy, 2019-09-23:
Hi I used nut, to monitor other sais, but I recently bought a salicru sps home, and I am not able to make it work with any driver, could someone help me?

clepple, 2019-09-23:
from https://networkupstools.org/support.html#_request_help :

* be sure to include the following information:
  * OS name and version,
  * exact NUT version,
  * NUT installation method: from source tarball, package or Subversion [git],
  * exact device name and related information (manufacturing date, web pointers, …),
* complete problem description, with any relevant traces, like system log excerpts, and driver debug output. You can obtain the latter using the following command, as root and after having stopped NUT:

```
/path/to/driver -DD -a <upsname>
```
If you don’t include the above information in your help request, we will not be able to help you!

RebelPhoton, 2019-11-28:
Hello. I have a similar unit: Salicru SPS Home 850 VA https://www.salicru.com/en/sps-home-eng.html
I got to the point of running the usb hid debug driver exploration.

`/lib/nut# ./usbhid-ups -DD  -x explore -x vendorid=2e66 -a salicru >& salicru-usbhid-explore.txt`

It gets into a loop. I stopped it after a while with ctrl+c. a The info is attached to this post.

[salicru-usbhid-explore.txt](https://github.com/networkupstools/nut/files/3902329/salicru-usbhid-explore.txt)

I'm running Promox 6, a debian buster derivative. uname -a returns:

`Linux pve 5.0.21-5-pve #1 SMP PVE 5.0.21-10 (Wed, 13 Nov 2019 08:27:10 +0100) x86_64 GNU/Linux`

Nut version is 2.7.4-8, installed via apt-get.

Unfortunately I don't know to proceed further, as the documentation starts talking about implementing new drivers and I wouldn't be able to tackle that... can I help with anything else?

RebelPhoton, 2020-02-10:
> If you don’t include the above information in your help request, we will not be able to help you!

This issue is still tagged as "need more info"... can I help with any more information?
Thanks.

gnomework, 2020-02-11:
Hi, I am trying to run the same unit as @RebelPhoton.

My OS is Raspbian and running uname -a retuns:
`Linux pi-hole 4.19.97+ #1294 Thu Jan 30 13:10:54 GMT 2020 armv6l GNU/Linux`

I have Nut version 2.7.4-8 installed via apt-get
`Package: nut                                                                                                                                                                                                       
Version: 2.7.4-8                                                                                                                                                                                                   
Priority: optional                                                                                                                                                                                                 
Section: metapackages                                                                                                                                                                                              
Maintainer: Arnaud Quette <aquette@debian.org>                                                                                                                                                                     
Installed-Size: 270 kB                                                                                                                                                                                             
Depends: nut-client, nut-server                                                                                                                                                                                    
Homepage: http://www.networkupstools.org/                                                                                                                                                                          
Tag: admin::monitoring, hardware::power, hardware::power:ups,                                                                                                                                                      
 interface::daemon, network::server, role::program, scope::utility                                                                                                                                                 
Download-Size: 245 kB                                                                                                                                                                                              
APT-Manual-Installed: yes                                                                                                                                                                                          
APT-Sources: http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages`

I ran the driver explore 
`/lib/nut/usbhid-ups -DD -x explore -x vendorid=2a66 -a salicru >& salicru-explore.txt`
[salicru_explore.txt](https://github.com/networkupstools/nut/files/4184842/salicru_explore.txt)

Tnix76, 2020-05-08:
Hi all!
I have this UPS unit and I can't get to work with vendor's supplied software PowerMaster. I think it needs the correct driver but I tried with _usbhid-ups_ and _blazer_usb_ without results. The UPS cannot be detected.
In Windows it's OK.

Any clue? Thanks.

xbgmsharp, 2021-11-20:
Same issue using the following:

SPS 850 HOME, https://www.salicru.com/sps-home.html

OS
```
# grep NAME /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_CODENAME=bullseye
```

Kernel:
```
# uname -a
Linux raspberrypi 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux
```

NUT version install from package.
```
# dpkg -l | grep nut-
ii  nut-client                           2.7.4-13                         armhf        network UPS tools - clients
ii  nut-server                           2.7.4-13                         armhf        network UPS tools - core system
# upsdrvctl -V
Network UPS Tools - UPS driver controller 2.7.4
```

usbhid-ups:
```
# /lib/nut/usbhid-ups -DD -x explore -x vendorid=2a66 -a salicru
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000	debug level is '2'
   0.002942	upsdrv_initups...
   0.005266	Checking device (2E66/0300) (001/004)
   0.006523	- VendorID: 2e66
   0.007462	- ProductID: 0300
   0.008361	- Manufacturer: unknown
   0.009257	- Product: unknown
   0.010172	- Serial Number: unknown
   0.010977	- Bus: 001
   0.011977	- Device release number: 0200
   0.012898	Trying to match device
   0.013865	Device does not match - skipping
   0.014821	Checking device (0424/EC00) (001/003)
   0.015712	- VendorID: 0424
   0.016093	- ProductID: ec00
   0.016519	- Manufacturer: unknown
   0.016695	- Product: unknown
   0.016868	- Serial Number: unknown
   0.017239	- Bus: 001
   0.017414	- Device release number: 0200
   0.017778	Trying to match device
   0.017974	Device does not match - skipping
   0.018393	Checking device (0424/9514) (001/002)
   0.018823	- VendorID: 0424
   0.019011	- ProductID: 9514
   0.019376	- Manufacturer: unknown
   0.019551	- Product: unknown
   0.019917	- Serial Number: unknown
   0.020091	- Bus: 001
   0.020462	- Device release number: 0200
   0.020637	Trying to match device
   0.021017	Device does not match - skipping
   0.021220	Checking device (1D6B/0002) (001/001)
   0.021681	- VendorID: 1d6b
   0.022056	- ProductID: 0002
   0.022410	- Manufacturer: unknown
   0.022596	- Product: unknown
   0.022953	- Serial Number: unknown
   0.023138	- Bus: 001
   0.023492	- Device release number: 0510
   0.023676	Trying to match device
   0.023862	Device does not match - skipping
   0.024254	No appropriate HID device found
   0.024630	No matching HID UPS found
```

xbgmsharp, 2021-11-22:
Created a generic salicru driver and biw getting additional information, however not sure where to go from there.
```
$ sudo ./drivers/usbhid-ups -DD -u root -x productid=0300 -a salicru
Network UPS Tools - Generic HID driver 0.43 (2.7.4-3304-g6135bb65)
USB communication driver 0.33
   0.000000	[D1] debug level is '2'
   0.002536	[D2] Initializing an USB-connected UPS with library (null) (NUT subdriver name='USB communication driver' ver='0.33')
   0.002634	[D1] upsdrv_initups (non-SHUT)...
   0.004074	[D2] Checking device (2E66/0300) (001/008)
   0.006768	[D2] - VendorID: 2e66
   0.006847	[D2] - ProductID: 0300
   0.006905	[D2] - Manufacturer: 1 
   0.006964	[D2] - Product:  850
   0.007021	[D2] - Serial Number:                 
   0.007079	[D2] - Bus: 001
   0.007137	[D2] - Device: 008
   0.007192	[D2] - Device release number: 0200
   0.007241	[D2] Trying to match device
   0.007358	[D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.007583	[D2] Device matches
   0.007730	[D2] failed to claim USB device: could not claim interface 0: Device or resource busy
   0.009234	[D2] detached kernel driver from USB device...
   0.010951	[D2] HID descriptor length 739
   0.015277	[D2] Report Descriptor size = 739
   0.015867	Using subdriver: Salicru HID 0.1
   0.015962	[D1] 73 HID objects found
   0.016524	[D1] Path: UPS.PowerSummary.iProduct, Type: Feature, ReportID: 0x01, Offset: 0, Size: 8, Value: 1
   0.017777	[D1] Path: UPS.PowerSummary.iSerialNumber, Type: Feature, ReportID: 0x02, Offset: 0, Size: 8, Value: 2
   0.019025	[D1] Path: UPS.PowerSummary.iOEMInformation, Type: Feature, ReportID: 0x03, Offset: 0, Size: 8, Value: 3
   0.020166	[D1] Path: UPS.PowerSummary.iDeviceChemistry, Type: Feature, ReportID: 0x04, Offset: 0, Size: 8, Value: 4
   0.021537	[D1] Path: UPS.PowerSummary.Rechargeable, Type: Feature, ReportID: 0x05, Offset: 0, Size: 8, Value: 1
   0.022780	[D1] Path: UPS.PowerSummary.CapacityMode, Type: Feature, ReportID: 0x06, Offset: 0, Size: 8, Value: 2
   0.023899	[D1] Path: UPS.PowerSummary.DesignCapacity, Type: Feature, ReportID: 0x07, Offset: 0, Size: 8, Value: 100
   0.024812	[D1] Path: UPS.PowerSummary.CapacityGranularity1, Type: Feature, ReportID: 0x07, Offset: 8, Size: 8, Value: 10
   0.025600	[D1] Path: UPS.PowerSummary.CapacityGranularity2, Type: Feature, ReportID: 0x07, Offset: 16, Size: 8, Value: 10
   0.026334	[D1] Path: UPS.PowerSummary.WarningCapacityLimit, Type: Feature, ReportID: 0x07, Offset: 24, Size: 8, Value: 20
   0.027068	[D1] Path: UPS.PowerSummary.FullChargeCapacity, Type: Feature, ReportID: 0x07, Offset: 32, Size: 8, Value: 100
   0.027785	[D1] Path: UPS.PowerSummary.RemainingCapacityLimit, Type: Feature, ReportID: 0x07, Offset: 40, Size: 8, Value: 10
   0.028534	[D1] Path: UPS.PowerSummary.RemainingCapacity, Type: Input, ReportID: 0x08, Offset: 0, Size: 8, Value: 100
   0.028688	[D1] Path: UPS.PowerSummary.RemainingCapacity, Type: Feature, ReportID: 0x08, Offset: 0, Size: 8, Value: 100
   0.028826	[D1] Path: UPS.PowerSummary.RunTimeToEmpty, Type: Input, ReportID: 0x08, Offset: 8, Size: 16, Value: 6000
   0.028962	[D1] Path: UPS.PowerSummary.RunTimeToEmpty, Type: Feature, ReportID: 0x08, Offset: 8, Size: 16, Value: 6000
   0.029350	[D1] Path: UPS.PowerSummary.RemainingTimeLimit, Type: Input, ReportID: 0x08, Offset: 24, Size: 16, Value: 300
   0.029486	[D1] Path: UPS.PowerSummary.RemainingTimeLimit, Type: Feature, ReportID: 0x08, Offset: 24, Size: 16, Value: 300
   0.030087	[D1] Path: UPS.PowerSummary.ConfigVoltage, Type: Feature, ReportID: 0x09, Offset: 0, Size: 16, Value: 12
   0.030647	[D1] Path: UPS.PowerSummary.Voltage, Type: Feature, ReportID: 0x0a, Offset: 0, Size: 16, Value: 13.5
   0.031269	[D1] Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Input, ReportID: 0x0b, Offset: 0, Size: 1, Value: 1
   0.031417	[D1] Path: UPS.PowerSummary.PresentStatus.Charging, Type: Input, ReportID: 0x0b, Offset: 1, Size: 1, Value: 1
   0.031551	[D1] Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Input, ReportID: 0x0b, Offset: 2, Size: 1, Value: 0
   0.031687	[D1] Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Input, ReportID: 0x0b, Offset: 3, Size: 1, Value: 0
   0.031820	[D1] Path: UPS.PowerSummary.PresentStatus.FullyCharged, Type: Input, ReportID: 0x0b, Offset: 4, Size: 1, Value: 1
   0.031954	[D1] Path: UPS.PowerSummary.PresentStatus.RemainingTimeLimitExpired, Type: Input, ReportID: 0x0b, Offset: 5, Size: 1, Value: 0
   0.032351	[D1] Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Feature, ReportID: 0x0b, Offset: 0, Size: 1, Value: 1
   0.032495	[D1] Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x0b, Offset: 1, Size: 1, Value: 1
   0.032628	[D1] Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Feature, ReportID: 0x0b, Offset: 2, Size: 1, Value: 0
   0.032952	[D1] Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x0b, Offset: 3, Size: 1, Value: 0
   0.033098	[D1] Path: UPS.PowerSummary.PresentStatus.FullyCharged, Type: Feature, ReportID: 0x0b, Offset: 4, Size: 1, Value: 1
   0.033232	[D1] Path: UPS.PowerSummary.PresentStatus.RemainingTimeLimitExpired, Type: Feature, ReportID: 0x0b, Offset: 5, Size: 1, Value: 0
   0.034036	[D1] Path: UPS.PowerSummary.AudibleAlarmControl, Type: Feature, ReportID: 0x0c, Offset: 0, Size: 8, Value: 2
[... cut ...]
   0.062117	[D1] Can't retrieve Report 2a: Value too large for defined data type
   0.062256	[D1] Path: UPS.ff01001d.ff01001b.ff010013, Type: Feature, ReportID: 0x2a, Offset: 0, Size: 1
   0.063941	[D2] libusb_get_report: error sending control message: Value too large for defined data type
   0.064104	[D1] Can't retrieve Report 2b: Value too large for defined data type
   0.064242	[D1] Path: UPS.ff01001d.ff01001b.ff010014, Type: Feature, ReportID: 0x2b, Offset: 0, Size: 1
   0.064417	[D2] Report descriptor retrieved (Reportlen = 739)
   0.064542	[D2] Found HID device

```

xbgmsharp, 2021-11-22:
Just push an initial release of the Salicru driver and now getting the following information. Would be great to confirm the output value. @Tnix76 @RebelPhoton @gnomework could you help to compare the data from the official software.

```
$ sudo ./clients/upsc salicru@localhost
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: 1 
battery.runtime: 1440
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage.nominal: 12
device.mfr: 1 
device.model:  850
device.serial:                 
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4-3304-g6135bb65
driver.version.data: Salicru HID 0.1
driver.version.internal: 0.43
input.frequency: 50.2
input.transfer.high: 270
input.transfer.low: 180
input.voltage: 224.5
input.voltage.nominal: 230
output.frequency: 50.2
output.voltage: 224.5
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 18
ups.mfr: 1 
ups.model:  850
ups.productid: 0300
ups.realpower.nominal: 490
ups.serial:                 
ups.status: OL
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 2e66
```

jimklimov, 2021-11-23:
Cool! Negative timer values look a bit weird though :\

xbgmsharp, 2021-11-24:
Latest output confirmed with the model I have.

```
$ sudo ./clients/upsc salicru@localhost 
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.runtime: 1440
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 13.50
battery.voltage.nominal: 12
device.mfr: 1 
device.model:  850
device.serial:                 
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4-3324-g845d808c
driver.version.data: Salicru HID 0.1
driver.version.internal: 0.43
input.frequency: 50.0
input.voltage: 224.5
input.voltage.nominal: 230
output.frequency: 50.0
output.voltage: 224.5
ups.beeper.status: enabled
ups.load: 18
ups.mfr: 1 
ups.model:  850
ups.productid: 0300
ups.realpower.nominal: 490
ups.serial:                 
ups.status: OL
ups.vendorid: 2e66
```

clepple, 2021-11-24:
> Cool! Negative timer values look a bit weird though :\

I think that's a HID PDC thing, where the timer values are negative when the timer isn't counting. From an MGE UPS:

```
ups.timer.shutdown: -1
ups.timer.start: -10
```

jimklimov, 2022-01-10:
In the report, is the `device.mfr` supposed to be `1`? (I see that in `usb-devices` and `lsusb` listings above as well... looks weird :) )

jimklimov, 2022-01-13:
With the changes mentioned above, merged and tested, the UPS model should not be "incompatible" anymore; so closed the issue.

mahonni, 2023-05-14:
Hi,

I have the same model, but the 650 VA version instead, and it doesn't work for me. VendorID and ProductID are the same, but it doesn't find a match. Any idea?

System info:
uname -a
Linux raspberrypi 5.15.84-v7l+ #1613 SMP Thu Jan 5 12:01:26 GMT 2023 armv7l GNU/Linux

dpkg -l | grep nut-server
nut-server                           2.7.4-13

Installation method: repository => apt get

/lib/nut/usbhid-ups -DD -x explore -x vendorid=2a66 -a salicru
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000     debug level is '2'
   0.002139     upsdrv_initups...
(...)
   0.176997     Checking device (2E66/0300) (001/006)
   0.177164     - VendorID: 2e66
   0.177251     - ProductID: 0300
   0.177309     - Manufacturer: unknown
   0.177365     - Product: unknown
   0.177444     - Serial Number: unknown
   0.177525     - Bus: 001
   0.177582     - Device release number: 0200
   0.177664     Trying to match device
   0.177750     Device does not match - skipping
(...)
   0.180710     No appropriate HID device found
   0.180770     No matching HID UPS found

xbgmsharp, 2023-05-14:
You would need to install the package by source, as the drivers is not included yet in Debian release. 
Please try using NUT v2.8.0.
https://github.com/networkupstools/nut/wiki/Links-to-distribution-packaging-recipes-and-repository-sections

jimklimov, 2023-05-15:
In the command you ask for 2a66 but the device is 2e66

mahonni, 2023-05-15:
> In the command you ask for 2a66 but the device is 2e66

Oh my!! My dyslexia... [facepalm]

Looks like now it's detecting it. I'll do some more checks.

BTW, is it also possible to compile it for a Synology NAS? The issue was solved a year and a half ago and it still doesn't have this version :(

Thanks for the help!

jimklimov, 2023-05-16:
I am not sure, probably it would be best to first bring it up with them as a customer, so it would be a "proper" release. Otherwise, they are obliged under GPL terms to provide at least recipes of how to build your own binary similar to theirs...

valsergi, 2023-11-05:
Hello, I think I have the same problem, with a UPS 650 home, I can't get it to work, but a ONE 700 is working without any problem NUT

Could someone help me get it up and running on my RPI 3b+?

THANK YOU!

jimklimov, 2023-11-05:
What NUT version do you run? (2.8.1 is finally out)
What messages do you see with elevated debug verbosity - why your driver build thinks it does not recognize the device?

valsergi, 2023-11-05:
> What NUT version do you run? (2.8.1 is finally out) What messages do you see with elevated debug verbosity - why your driver build thinks it does not recognize the device?

Hello, thanks for the response, I'm a bit of a newbie, I've seen that this version is finally available, but I don't know how to install it on my Raspberry Pi, maybe that's the solution, I'm currently Running an older version. Could you help me install that final version, maybe it will be the solution to my problem

Once again thanks for the help!!

jimklimov, 2023-11-05:
Currently not on PC to type much. Check the wiki, e.g. on "in-place" builds.

valsergi, 2023-11-05:
> Currently not on PC to type much. Check the wiki, e.g. on "in-place" builds.

Ok, thanks!

closed 2025-09-03 by jimklimov