networkupstools/nut #734: snmp-ups doesnt work for APC UPS: testOID provided and doesn't match MIB

harsh-w, 2019-10-09:
Running Network UPS Tools 2.7.4 on Ubuntu 18.04 and trying to work with an APC SMT2200 with an AP9631

```
$ sudo upsd -V
Network UPS Tools upsd 2.7.4
```

For some reason the upsd driver cannot seem to match any OID, either the ietf or the apcc ones.

The contents of the ups.conf file
```
[APC_UPS]
                   driver = snmp-ups
                   port = 192.168.42.63
                   snmp_version = v3
                   secLevel = authPriv
                   secName = shutdown_user
                   authPassword = *********
                   authProtocol = SHA
                   privPassword = *********
                   privProtocol = AES
                   desc = "APC UPS"
```
Running `$ sudo /lib/nut/snmp-ups -DD -a APC_UPS` gives 

```
Network UPS Tools - Generic SNMP UPS driver 0.97 (2.7.4)
   0.000000	debug level is '2'
   0.000772	SNMP UPS driver: entering upsdrv_initups()
   0.000790	SNMP UPS driver: entering nut_snmp_init(snmp-ups)
   0.005638	Setting SNMP retries to 5
   0.005658	Setting SNMP timeout to 1 second(s)
   0.031828	SNMP UPS driver: entering load_mib2nut(auto)
   0.031840	trying the new match_sysoid() method
   0.049989	Can't get sysOID value
   0.050026	load_mib2nut: trying classic method with 'apcc' mib
   0.050034	Testing ups.model using OID .1.3.6.1.4.1.318.1.1.1.1.1.1.0
   0.050087	load_mib2nut: testOID provided and doesn't match MIB 'apcc'!
   0.050111	load_mib2nut: trying classic method with 'mge' mib
   0.050238	Testing ups.model using OID .1.3.6.1.4.1.705.1.1.1.0
   0.050306	load_mib2nut: testOID provided and doesn't match MIB 'mge'!
   0.050393	load_mib2nut: trying classic method with 'netvision' mib
   0.050422	Testing ups.model using OID .1.3.6.1.4.1.4555.1.1.1.1.1.1.0
   0.050515	load_mib2nut: testOID provided and doesn't match MIB 'netvision'!
   0.050540	load_mib2nut: trying classic method with 'pw' mib
   0.050576	Testing ups.model using OID 1.3.6.1.4.1.534.1.1.2.0
   0.050605	load_mib2nut: testOID provided and doesn't match MIB 'pw'!
   0.050645	load_mib2nut: trying classic method with 'pxgx_ups' mib
   0.050657	Testing ups.model using OID 1.3.6.1.4.1.534.1.1.2.0
   0.050672	load_mib2nut: testOID provided and doesn't match MIB 'pxgx_ups'!
   0.050686	load_mib2nut: trying classic method with 'aphel_genesisII' mib
   0.050695	Testing ups.model using OID .1.3.6.1.4.1.17373.3.1.1.0
   0.050712	load_mib2nut: testOID provided and doesn't match MIB 'aphel_genesisII'!
   0.050723	load_mib2nut: trying classic method with 'aphel_revelation' mib
   0.050732	Testing ups.model using OID .1.3.6.1.4.1.534.6.6.6.1.1.12.0
   0.050744	load_mib2nut: testOID provided and doesn't match MIB 'aphel_revelation'!
   0.050755	load_mib2nut: trying classic method with 'eaton_epdu' mib
   0.050764	Testing ups.model using OID 1.3.6.1.4.1.534.6.6.7.1.2.1.2.0
   0.050782	load_mib2nut: testOID provided and doesn't match MIB 'eaton_epdu'!
   0.050792	load_mib2nut: trying classic method with 'pulizzi_switched1' mib
   0.050801	Testing ups.model using OID .1.3.6.1.4.1.20677.2.1.1.0
   0.050814	load_mib2nut: testOID provided and doesn't match MIB 'pulizzi_switched1'!
   0.050824	load_mib2nut: trying classic method with 'pulizzi_switched2' mib
   0.050833	Testing ups.model using OID .1.3.6.1.4.1.20677.2.1.1.0
   0.050845	load_mib2nut: testOID provided and doesn't match MIB 'pulizzi_switched2'!
   0.050855	load_mib2nut: trying classic method with 'raritan' mib
   0.050864	Testing ups.model using OID .1.3.6.1.4.1.13742.1.1.12.0
   0.050876	load_mib2nut: testOID provided and doesn't match MIB 'raritan'!
   0.050890	load_mib2nut: trying classic method with 'baytech' mib
   0.050899	Testing ups.model using OID .1.3.6.1.4.1.4779.1.3.5.2.1.24.1
   0.050912	load_mib2nut: testOID provided and doesn't match MIB 'baytech'!
   0.050922	load_mib2nut: trying classic method with 'cpqpower' mib
   0.050931	Testing ups.model using OID .1.3.6.1.4.1.232.165.3.1.2.0
   0.050942	load_mib2nut: testOID provided and doesn't match MIB 'cpqpower'!
   0.050952	load_mib2nut: trying classic method with 'bestpower' mib
   0.050962	Testing ups.model using OID .1.3.6.1.4.1.2947.1.1.2.0
   0.050973	load_mib2nut: testOID provided and doesn't match MIB 'bestpower'!
   0.050989	load_mib2nut: trying classic method with 'cyberpower' mib
   0.050999	Testing ups.model using OID .1.3.6.1.4.1.3808.1.1.1.1.1.1.0
   0.051011	load_mib2nut: testOID provided and doesn't match MIB 'cyberpower'!
   0.051021	load_mib2nut: trying classic method with 'delta_ups' mib
   0.051030	Testing ups.model using OID .1.3.6.1.4.1.2254.2.4.1.2.0
   0.051042	load_mib2nut: testOID provided and doesn't match MIB 'delta_ups'!
   0.051052	load_mib2nut: trying classic method with 'xppc' mib
   0.051061	Testing ups.model using OID .1.3.6.1.4.1.935.1.1.1.1.1.1.0
   0.051073	load_mib2nut: testOID provided and doesn't match MIB 'xppc'!
   0.051087	load_mib2nut: trying classic method with 'huawei' mib
   0.051096	Testing ups.model using OID .1.3.6.1.4.1.2011.6.174.1.2.100.1.2.1
   0.051108	load_mib2nut: testOID provided and doesn't match MIB 'huawei'!
   0.051118	load_mib2nut: trying classic method with 'ietf' mib
   0.051127	Testing ups.model using OID 1.3.6.1.2.1.33.1.1.2.0
   0.051140	load_mib2nut: testOID provided and doesn't match MIB 'ietf'!
   0.051149	load_mib2nut: trying classic method with 'eaton_ats' mib
   0.051160	Testing device.model using OID .1.3.6.1.4.1.534.10.2.1.2.0
   0.051173	load_mib2nut: testOID provided and doesn't match MIB 'eaton_ats'!
   0.051182	load_mib2nut: trying classic method with 'apc_ats' mib
   0.051198	Testing device.model using OID .1.3.6.1.4.1.318.1.1.8.1.5.0
   0.051211	load_mib2nut: testOID provided and doesn't match MIB 'apc_ats'!
   0.051220	load_mib2nut: trying classic method with 'ietf' mib
   0.051230	Testing ups.model using OID 1.3.6.1.2.1.33.1.1.2.0
   0.051241	load_mib2nut: testOID provided and doesn't match MIB 'ietf'!
   0.051251	No supported device detected
```

Yet running snmpwalk with the same values for authPassword and privPassword works

for the IETF OID
```
$ snmpwalk -v3 -u shutdown_user -l authpriv -a SHA -A ********* -x AES -X ********* 192.168.42.63 1.3.6.1.2.1.33.1.1.2.0
iso.3.6.1.2.1.33.1.1.2.0 = STRING: "Smart-UPS 2200"
```
and for the APCC OID

```
$ snmpwalk -c shutdown_user -v3 -u shutdown_user -l authpriv -a SHA -A  ********* -x AES -X  ********* 192.168.42.63 .1.3.6.1.4.1.318.1.1.1.1.1.1.0 
iso.3.6.1.4.1.318.1.1.1.1.1.1.0 = STRING: "Smart-UPS 2200"```

Why would getting the OID fail when snmpwalk can retrieve it?

harsh-w, 2020-02-04:
I found changing authProtocol to MD5 instead of SHA and privProtocol to DES instead of AES - and changing the values correspondingly on the NMC card -  will cause snmp-ups to work. Any idea of why AES is not supported?

Bi11, 2020-12-09:
> I found changing authProtocol to MD5 instead of SHA and privProtocol to DES instead of AES - and changing the values correspondingly on the NMC card - will cause snmp-ups to work. Any idea of why AES is not supported?

Thank you for the hint of AES, I finally connected to my UPS, and found out why AES is broken.

Let check the code from tag v2.7.4. It seems there is nothing different between DES and AES:
https://github.com/networkupstools/nut/blob/0b4bfddbc609c59ac0a117c758e6893635121166/drivers/snmp-ups.c#L474-L481

But wait for a second:
https://github.com/networkupstools/nut/blob/0b4bfddbc609c59ac0a117c758e6893635121166/drivers/snmp-ups.c#L60-L62

Related variables are defined in `transform_oids.h` from `libnetsnmp`.
```
NETSNMP_IMPORT oid      usmAESPrivProtocol[10]; /* == { 1,3,6,1,6,3,10,1,2,4 }; */
NETSNMP_IMPORT oid      *usmAES128PrivProtocol; /* backwards compat */
```

A _classic_ issue. `sizeof(usmAESPrivProtocol)/sizeof(oid)` becomes `sizeof(oid*)/sizeof(oid)`.

A FIXME comment was added recently in c7cf507454bcf6f40b24b58e61c130c7772c5bf2 and 7bcaa2599365350dbaa48f882c79f3e5d38cbb17. But this issue is not resolved yet.

`nut-scanner` with SNMPv3 authPriv is also broken too, it always produces `Error generating Ku from authentication pass phrase`.

jimklimov, 2020-12-16:
Thanks for the investigation. In the end, if this does work for you with current master branch codebase, which I hope will parent an official release sometime in visible future, can this issue be closed? :)

jflattery, 2021-07-13:
@jimklimov: What kind of time frame is the "visible future"? 🥺

jimklimov, 2021-09-21:
Vague, the future is. Always moving. Ever changing.

I do actually hope for a manageable timeframe, with more releases of finer-grained goals that were initially clumped as goals of 2.7.5; so now getting a warnings-clean codebase out (and half a decade of updates from contributors) that outlasted and survived 3 or more CI farm implementations, is a worthy milestone in its own :)

jeffgoh, 2022-01-04:
AP9640 - followed above didn't get anywhere until I added the following:

`secLevel=authPriv
`
for reference, here is what the AP9640/SU3000 returned via snmpwalk

`snmpwalk -v3 -u xxx -l authpriv -a MD5 -A <authprrace>  -x DES -X <privacyphrase> | grep APC
SNMPv2-MIB::sysDescr.0 = STRING: APC Web/SNMP Management Card (MB:v4.2.9 PF:v1.4.0.23 PN:apc_hw21_aos_1.4.0.23.bin AF1:v1.4.0.19 AN1:apc_hw21_su_1.4.0.19.bin MN:AP9640 HR:5 SN: ZA2145154071 MD:11/13/2021) (Embedded PowerNet SNMP Agent SW v2.2 compatible)
SNMPv2-SMI::mib-2.33.1.1.1.0 = STRING: "APC"
SNMPv2-SMI::mib-2.33.1.1.5.0 = STRING: "APC UPS"
`

lightingman117, 2023-06-28:
I have this issue too.

This is the config that eventually worked.

```
- name: ups-apc-x2200
  driver: snmp-ups
  port: 192.168.1.2
  snmp_version: v3
  secLevel: authPriv
  authProtocol: MD5
  authPassword: 0123456789ABCDEF
  privProtocol: DES
  privPassword: 0123456789ABCDEF
  desc: my description doesn not work
  config: []
```

jimklimov, 2023-06-29:
Thanks for the report.

Just in case: what is the expectation for `desc` to "work"? Per `docs/man/ups.conf.txt`: 
> *desc*::
>
> Optional.  This allows you to set a brief description that upsd will provide to clients that ask for a list of connected equipment.

So this allows clients like `ups*.cgi` or Python/X11/etc. UI or even data dumps with `upsc` to make it easier for you as the end-user to discern the devices, like "Beeping box near the fridge in the basement" :)

jimklimov, 2023-06-29:
@lightingman117 : also, which version of NUT is in the HA plugin? 2.7.4? 2.8.0? Custom build (newer than 2.8.0)?

lightingman117, 2023-07-24:
> Thanks for the report.
>
> Just in case: what is the expectation for `desc` to "work"? Per `docs/man/ups.conf.txt`:
>
>
> So this allows clients like `ups*.cgi` or Python/X11/etc. UI or even data dumps [...]

Sorry for taking so long to respond. Had some personal things to take care of.

[image]
I expected the description in the add-on to show up here when adding UPS via the integration.
I tried using quotes, no spaces, only letters, only lower case.
I had it work the very first time I loaded the integration with one UPS in the NUT configuration.
It never worked since.
Might be an issue on NUT side.

> @lightingman117 : also, which version of NUT is in the HA plugin? 2.7.4? 2.8.0? Custom build (newer than 2.8.0)?

```
Home Assistant 2023.7.1
Supervisor 2023.07.1
Operating System 10.3
Frontend 20230705.1 - latest
NUT Add-on 0.12.0
```

I can't figure out how to view the current version of integration(s)?
Can you point me in the right direction?

lightingman117, 2023-07-24:
So after looking at this some more.

Even with SNMPv3 disabled on either the Cyberpower or APC UPS and enabled in ups.conf file the SNMP-UPS.c driver still pulls values. This is due to pulling SNMPv1 and if I disable that, then the driver fails to load.

This is kinda a big deal that it is falling back to SNMPv1 as it is a privacy/security issue. If I setup SNMPv3 with encryption I'd expect to fail-safe (not working, but secure state) not fail-open (working, but insecure state).

Weird behavior.

Let me know how I can help.

I have a suite of UPS's available to test.

I can default them and grab wireshark/snmp-walk's if you need them, NUT logs, etc.
Got plenty of time as this is work related.
Thanks!

jimklimov, 2023-07-24:
So, by some googling I found that this plugin seems to be the official one, current (June 14) release with some (not all) NUT packages from a Debian distro (not sure quickly which one - a hassio debian base image 7.0.0), all per: https://github.com/hassio-addons/addon-nut/blob/main/nut/Dockerfile

So that seems reasonably fresh :)

jimklimov, 2023-07-24:
As for desc, did you try to use double-quotes in cases where you have spaces?

jimklimov, 2023-07-24:
As for debugging, NUT should be quite talkative with raised debug levels. Since 2.8.0 you can bump them via configs (see `debug_min` in docs), which is easier than older alternatives. See e.g. https://github.com/networkupstools/nut/wiki/Changing-NUT-daemon-debug-verbosity

You can also build NUT to e.g. add more logging in spots of interest or to step through the driver with an IDE (NetBeans and VSCode seemed to work), and generally have fewer moving parts (package tweaks, docker, HA...) to rule out (or implicate) influence of non-NUT code.

jimklimov, 2023-07-24:
Generally, I'd expect a driver setting of `v3` and lack of a usable community for `v1` to make the driver unable to query v1/v2c, the device usually won't reply to a wrong community.

That said, SNMPv3 was neglected for some time, with PRs coming as part of 2.8.0 release and some more maybe later. So there's a fair chance that current git tip behaves at least differently (maybe better) than 2.8.0 or older codebases...

jimklimov, 2023-07-31:
Just in case, can you post here the output of `snmp-ups -h`, the part where it suggests SNMPv3-related options (if supported by the build, based on net-snmp library used during compilation)? For example, on my build box it says:
````
...
Set the securityLevel used for SNMPv3 messages (default=noAuthNoPriv, allowed: authNoPriv,authPriv) : -x secLevel=<value>
Set the securityName used for authenticated SNMPv3 messages (no default) : -x secName
Set the authentication pass phrase used for authenticated SNMPv3 messages (no default) : -x authPassword
Set the privacy pass phrase used for encrypted SNMPv3 messages (no default) : -x privPassword
Set the authentication protocol (MD5, SHA, SHA256, SHA384, SHA512) used for authenticated SNMPv3 messages (default=MD5 if available) : -x authProtocol=<value>
Set the privacy protocol (DES, AES) used for encrypted SNMPv3 messages (default=DES if available) : -x privProtocol=<value>
````

Notably the code in NUT v2.8.0+ goes to great lengths to only report the values actually supported (and should error out if you ask it for say SHA512 when it is missing in the binary). It is remotely possible that somehow you specify what it can not enact, and that is why/how it falls back to SNMPv1 (if it does, is a bug).

arnaudquette-eaton, 2023-12-20:
ping up, and finally looking at this thread, since notified through #hassio-addons/addon-nut/issues/331

* first, on the why your UPS is not matched:
as per logs:
```
   0.049989	Can't get sysOID value
   0.050026	load_mib2nut: trying classic method with 'apcc' mib
   0.050034	Testing ups.model using OID .1.3.6.1.4.1.318.1.1.1.1.1.1.0
```

the matching system I made first try to get the sysOID (which doesn't work here! see below). Then also confirms the match with an additional test OID, in your APC case:
https://github.com/networkupstools/nut/blob/master/drivers/apc-mib.c#L31

The issue here is really the missing sysOID, which shouldn't be.
Could you please send the result of:
> snmpget ... .1.3.6.1.2.1.1.2.0

* 2nd, your conf is malformed:
```
- name: ups-apc-x2200
  driver: snmp-ups
  port: 192.168.1.2
  snmp_version: v3
  secLevel: authPriv
  authProtocol: MD5
  authPassword: 0123456789ABCDEF
  privProtocol: DES
  privPassword: 0123456789ABCDEF
  desc: my description doesn not work
  config: []
  ```
  
  should be
  ```
- name: ups-apc-x2200
  driver: snmp-ups
  port: 192.168.1.2
  desc: my description doesn not work
  config:
    snmp_version = v3
    secLevel = authPriv
    authProtocol = MD5
    authPassword = 0123456789ABCDEF
    privProtocol = DES
    privPassword = 0123456789ABCDEF
  
  ```
  pay attention to "=" Vs ":" if not copy/pasting ;)