networkupstools/nut #735: CyberPower OR2200PFCRT2U unstable connection?

abrownsword, 2019-10-16:
I recently acquired this UPS and install NUT to communicate with it.  I am using the usbhid-ups driver and it seems to work correctly, except that it loses connection after some amount of time (on the order of hours or days) and seems unable to regain it.  Restarting the driver gets them talking again, but it fails eventually.

I am running Ubuntu 18.04 on an aarch64 w/ nut 2.7.4-5.1ubutun2 installed via apt.

abrownsword, 2019-10-21:
If anyone else comes across this issue, the settings suggested in this link seem to have solved my issue:  https://nmaggioni.xyz/2017/03/14/NUT-CyberPower-UPS/

BradenM, 2019-11-04:
Hi,

I was also experiencing this issue with a CyberPower OR2200LCDRM2U and can confirm that the changes suggested in your link has solved it.

For reference, here are my (now working) settings:

**ups.conf**
```
[ups]
driver=usbhid-ups
port=auto
vendorid=0764
pollinterval=15
desc="CyberPower OR2200LCDRM2U"
offdelay=60
ondelay=120
```

**upsmon.conf**
```
MONITOR ups 1 ***** ****** master
SHUTDOWNCMD "/sbin/shutdown -p +0"
POWERDOWNFLAG /etc/killpower
NOTIFYCMD /usr/local/pkg/nut/nut_email.php
NOTIFYFLAG ONLINE   SYSLOG+WALL+EXEC
NOTIFYFLAG ONBATT   SYSLOG+WALL+EXEC
NOTIFYFLAG LOWBATT  SYSLOG+WALL+EXEC
NOTIFYFLAG FSD	 SYSLOG+WALL+EXEC
NOTIFYFLAG COMMOK   SYSLOG+WALL+EXEC
NOTIFYFLAG COMMBAD  SYSLOG+WALL+EXEC
NOTIFYFLAG SHUTDOWN SYSLOG+WALL+EXEC
NOTIFYFLAG REPLBATT SYSLOG+WALL+EXEC
NOTIFYFLAG NOCOMM   SYSLOG+WALL+EXEC
NOTIFYFLAG NOPARENT SYSLOG+WALL+EXEC

DEADTIME 25
MAXAGE 25
```

closed 2021-07-01 by abrownsword