networkupstools/nut #736: apcupsd-ups: Override battery.charge.low

Jonn3y, 2019-11-02:
Hey, 

i tried to setup Freenas as slave using a RaspberryPi on the UPS with apcupsd as master. 

Now I want to override the charge.low value. But it seems like it is not possible with the apcupsd-ups driver?

What am I doing wrong?

ups.conf:
```
[ups]
    driver = apcupsd-ups
    port = 10.0.10.51:3551
    desc = "SMT1500I"
    ignorelb =
    override.battery.charge.low = 30
```

root@freenas:~ # upsc ups
```
battery.charge: 40.0
battery.charge.low: 5          <-------- value from Master config: apcupsd.conf
battery.runtime: 4020.0
battery.runtime.low: 180
battery.voltage: 25.6
battery.voltage.nominal: 24.0
device.mfr: APC
device.model: Smart-UPS 1500
device.serial: AS1224310372
device.type: ups
driver.name: apcupsd-ups
driver.parameter.pollinterval: 60
driver.parameter.port: 10.0.10.51:3551
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.internal: 0.04
ups.date: 2019-11-02
ups.firmware: UPS 08.3
ups.firmware.aux: ID=18
ups.id: SMT1500I
ups.mfr: APC
ups.mfr.date: 2012-06-14
ups.model: Smart-UPS 1500
ups.serial: AS1224310372
ups.status: OL
ups.time: 17:58:45
```

And yes i could modify the apcupsd.conf but that's not what i want.
Freenas should shutdown earlier than the other clients.

Greetings

clepple, 2019-11-02:
You did restart the `apcupsd-ups` driver on the FreeNAS box after changing `ups.conf`, correct?

If not, I'm wondering if there was some sort of parse error on `ignorelb =` - usually we recommend just `ignorelb`.

I don't have things set up to test that driver right now, and nothing is jumping out at me when I look at the code. If the above two suggestions don't work, please stop the driver, and manually restart it with debug level 2 or more (e.g. `# /usr/local/libexec/nut/apcupsd-ups -DD -a ups > /tmp/apcupsd.log 2>&1`). We probably only need the first few seconds.

Jonn3y, 2019-11-02:
> You did restart the `apcupsd-ups` driver on the FreeNAS box after changing `ups.conf`, correct?

Well, i just did `service nut restart`, does it restart the apcupsd-ups driver ? 

> If not, I'm wondering if there was some sort of parse error on `ignorelb =` - usually we recommend just `ignorelb`.

replaced  `ignorelb =` with  `ignorelb` , nothing changed

___
Logs:
`root@freenas:~ # /usr/local/libexec/nut/apcupsd-ups -DD -a ups > /tmp/apcupsd.log 2>&1`
`
Ambiguous output redirect.
`

`root@freenas:~ # /usr/local/libexec/nut/apcupsd-ups -DD -a ups >& /tmp/apcupsd.log `
and 
`root@freenas:~ # cat /tmp/apcupsd.log `
```
   0.000000     debug level is '2'
   0.223968     using 'battery.charge' to set battery low state
   0.223983     using 'battery.runtime' to set battery low state
   0.224072     dstate_init: sock /var/db/nut/apcupsd-ups-ups open on fd 4
```

Feels like the log should be longer,?!

clepple, 2019-11-02:
> Well, i just did `service nut restart`, does it restart the apcupsd-ups driver ? 

Theoretically, if FreeNAS is using the same startup scripts as FreeBSD Ports. It should call `nut_poststop()` and `nut_prestart()` in `/usr/local/etc/rc.d/nut`, and those restart the driver through `upsdrvctl`.

I was looking at the wrong line for the debug level. You'll need `-DDDDD`. (As you noticed, there is not much debug output from the `apcupsd-ups` driver. I'm looking for output from [dstate.c](https://github.com/networkupstools/nut/blob/master/drivers/dstate.c#L194))

Jonn3y, 2019-11-02:
> Theoretically, if FreeNAS is using the same startup scripts as FreeBSD Ports. It should call `nut_poststop()` and `nut_prestart()` in `/usr/local/etc/rc.d/nut`, and those restart the driver through `upsdrvctl`.

`/usr/local/etc/rc.d/nut` looks good

> I was looking at the wrong line for the debug level. You'll need `-DDDDD`. (As you noticed, there is not much debug output from the `apcupsd-ups` driver. I'm looking for output from [dstate.c]

root@freenas:~ # cat /tmp/apcupsd.log
```
   0.000000     send_to_all: SETINFO driver.parameter.port "10.0.10.51:3551"
   0.000038     send_to_all: SETINFO driver.flag.ignorelb "enabled"
   0.000054     send_to_all: SETINFO battery.charge.low "30"
   0.000061     send_to_all: SETFLAGS battery.charge.low
   0.000075     debug level is '5'
   0.000893     send_to_all: SETINFO device.type "ups"
   0.000903     send_to_all: SETINFO driver.version "2.7.4"
   0.000911     send_to_all: SETINFO driver.version.internal "0.04"
   0.000919     send_to_all: SETINFO driver.name "apcupsd-ups"
   0.000930     send_to_all: DELINFO battery.charge.low
   0.109113     send_to_all: SETINFO ups.mfr "APC"
   0.109141     send_to_all: SETINFO ups.time "23:35:43"
   0.109152     send_to_all: SETINFO ups.date "2019-11-02"
   0.109174     send_to_all: SETINFO ups.id "SMT1500I"
   0.109203     send_to_all: SETINFO ups.model "Smart-UPS 1500 "
   0.109219     send_to_all: SETINFO ups.status "OL"
   0.109240     send_to_all: SETINFO battery.charge "100.0"
   0.109254     send_to_all: SETINFO battery.runtime "7620.0"
   0.109268     send_to_all: SETINFO battery.charge.low "5"
   0.109281     send_to_all: SETINFO battery.runtime.low "180"
   0.109303     send_to_all: SETINFO battery.voltage "26.8"
   0.109335     send_to_all: SETINFO ups.mfr.date "2012-06-14"
   0.109362     send_to_all: SETINFO battery.voltage.nominal "24.0"
   0.109375     send_to_all: SETINFO ups.firmware "UPS 08.3"
   0.109383     send_to_all: SETINFO ups.firmware.aux "ID=18"
   0.109413     send_to_all: DATAOK
   0.109423     send_to_all: DELINFO ups.serial
   0.109433     send_to_all: DELINFO ups.time
   0.109440     send_to_all: DELINFO ups.date
   0.109446     send_to_all: DELINFO ups.mfr.date
   0.109452     send_to_all: DELINFO ups.firmware
   0.109458     send_to_all: DELINFO ups.firmware.aux
   0.109464     send_to_all: DELINFO ups.id
   0.109473     send_to_all: DELINFO battery.charge
   0.109479     send_to_all: DELINFO battery.charge.low
   0.109486     send_to_all: DELINFO battery.voltage
   0.109492     send_to_all: DELINFO battery.voltage.nominal
   0.109497     send_to_all: DELINFO battery.runtime
   0.109503     send_to_all: DELINFO battery.runtime.low
   0.220392     send_to_all: SETINFO ups.time "23:35:43"
   0.220412     send_to_all: SETINFO ups.date "2019-11-02"
   0.220440     send_to_all: SETINFO ups.id "SMT1500I"
   0.220486     send_to_all: SETINFO battery.charge "100.0"
   0.220499     send_to_all: SETINFO battery.runtime "7620.0"
   0.220513     send_to_all: SETINFO battery.charge.low "5"
   0.220527     send_to_all: SETINFO battery.runtime.low "180"
   0.220548     send_to_all: SETINFO battery.voltage "26.8"
   0.220580     send_to_all: SETINFO ups.mfr.date "2012-06-14"
   0.220607     send_to_all: SETINFO battery.voltage.nominal "24.0"
   0.220619     send_to_all: SETINFO ups.firmware "UPS 08.3"
   0.220628     send_to_all: SETINFO ups.firmware.aux "ID=18"
   0.220655     using 'battery.charge' to set battery low state
   0.220663     using 'battery.runtime' to set battery low state
   0.220744     dstate_init: sock /var/db/nut/apcupsd-ups-ups open on fd 4
   0.220755     send_to_all: SETINFO driver.parameter.pollinterval "60"
   0.220762     send_to_all: SETINFO driver.parameter.synchronous "no"
   0.220770     send_to_all: SETINFO device.mfr "APC"
   0.220779     send_to_all: SETINFO device.model "Smart-UPS 1500 "
   0.220788     send_to_all: SETINFO device.serial "AS1224310372  "
   0.220795     send_to_all: DELINFO ups.serial
   0.220802     send_to_all: DELINFO ups.time
   0.220808     send_to_all: DELINFO ups.date
[... 50 lines cut ...]
 120.279529     send_to_all: DELINFO ups.mfr.date
 120.279536     send_to_all: DELINFO ups.firmware
 120.279542     send_to_all: DELINFO ups.firmware.aux
 120.279549     send_to_all: DELINFO ups.id
 120.279559     send_to_all: DELINFO battery.charge
 120.279566     send_to_all: DELINFO battery.charge.low
 120.279573     send_to_all: DELINFO battery.voltage
 120.279580     send_to_all: DELINFO battery.voltage.nominal
 120.279586     send_to_all: DELINFO battery.runtime
 120.279592     send_to_all: DELINFO battery.runtime.low
 120.390566     send_to_all: SETINFO ups.time "23:37:46"
 120.390586     send_to_all: SETINFO ups.date "2019-11-02"
 120.390612     send_to_all: SETINFO ups.id "SMT1500I"
 120.390659     send_to_all: SETINFO battery.charge "100.0"
 120.390673     send_to_all: SETINFO battery.runtime "7620.0"
 120.390687     send_to_all: SETINFO battery.charge.low "5"
 120.390701     send_to_all: SETINFO battery.runtime.low "180"
 120.390724     send_to_all: SETINFO battery.voltage "26.8"
 120.390756     send_to_all: SETINFO ups.mfr.date "2012-06-14"
 120.390783     send_to_all: SETINFO battery.voltage.nominal "24.0"
 120.390796     send_to_all: SETINFO ups.firmware "UPS 08.3"
 120.390805     send_to_all: SETINFO ups.firmware.aux "ID=18"
```

doesn't look like there is any override happening with a value of `30`

clepple, 2019-11-02:
Argh, I forgot that this driver does a `dstate_delinfo()` on all of the variables each time it polls. The override only gets set on the initial `battery.charge.low`, then it gets deleted.

Jonn3y, 2019-11-02:
Oh wow, yeah i see it. set on line 3 correctly, deleted override with the wrong values. 
Is there a way to stop the driver from doing it?

clepple, 2019-11-02:
In general, `dstate_delinfo()` should respect the `ST_FLAG_IMMUTABLE` flag that the `override.*` vars set. I'll create a separate issue for that.

You might be able to change `DU_FLAG_NONE` to `DU_FLAG_INIT` (on the `battery.charge.low` line) in [`drivers/apcupsd-ups.h`](https://github.com/networkupstools/nut/blob/v2.7.4/drivers/apcupsd-ups.h#L128). Do you have a way to rebuild the driver on FreeNAS?

Jonn3y, 2019-11-02:
> Do you have a way to rebuild the driver on FreeNAS?

I'm not sure if FreeNAS allows that. Because of its appliance state it's not designed to install compilers and other packages the way i know it from debian/apt-get ...
maybe i should not give it a try on my productive system. I'll not know what i'm doing exactly

clepple, 2019-11-02:
I think people have had success with finding the corresponding FreeBSD version, building NUT there, and copying the binaries over to FreeNAS [citation needed]. But probably best not to experiment without a backup system.

Jonn3y, 2019-11-03:
> You might be able to change `DU_FLAG_NONE` to `DU_FLAG_INIT` (on the `battery.charge.low` line) in [`drivers/apcupsd-ups.h`](https://github.com/networkupstools/nut/blob/v2.7.4/drivers/apcupsd-ups.h#L128). Do you have a way to rebuild the driver on FreeNAS?

Well, was kinda confusing to compile nut (or the apcupsd-ups driver) on FreeBSD 11.1 but i did it. 
replaced `/usr/local/libexec/nut/apcupsd-ups` with the new one. `chmod 0555 apcupsd-ups && service nut restart` and it works! 

`battery.charge.low` is now on `30`, awesome! 

I'll test if the shutdown at 30% battery capacity works as expected tomorrow.

clepple, 2019-11-03:
Cool. If you wouldn't mind, could you please create a wiki page with any tips on compilation, or links that you used? Feel free to use the Debian/Ubuntu one as a model: https://github.com/networkupstools/nut/wiki/Building-NUT-on-Debian,-Raspbian-and-Ubuntu

I will try to figure out whether we can add `DU_FLAG_INIT` for everyone - I think it might cause confusion if a given `apcupsd` connection doesn't expose `battery.charge.low` - but it might be fairly straightforward for us to modify the code to only delete a variable if it is really gone from apcupsd.

Jonn3y, 2019-11-03:
Done, it's not perfect and my english language skills might not be perfect, but this is what i did:
https://github.com/networkupstools/nut/wiki/Building-NUT-on-FreeBSD 

I could share my compiled apcupsd-ups driver if needed.

flobernd, 2020-06-15:
Can we expect a fix in the upstream release anytime soon? Just stumbled across the same issue. Building the driver itself is not a huge problem but it makes things a lot more complicated for automatic deployment using docker images :/

Jonn3y, 2020-10-07:
@flobernd issue #737 was closed. Maybe it works out of the box now

flobernd, 2020-10-07:
> @flobernd issue #737 was closed. Maybe it works out of the box now 

Thanks for the hint. It was actually me contributing this code change. It should work now :)