networkupstools/nut #738: riello_ser driver does not support ondelay/offdelay

Skarafaz, 2019-11-04:
Hi!

I'm monitoring a Riello NDG1500 with nut 2.7.4 (debian package) using riello_ser driver.
When I try to set ondelay/offdelay variables in ups.conf I get the error "variable not supported"; is this hardware related? I really need to set an off delay because my nas takes about 5 minutes to shutdown. Thanks.

clepple, 2019-11-04:
Per the protocol documentation, this should be supported by the hardware (NDG1500 uses the GPSER protocol): https://networkupstools.org/ups-protocols.html#_riello

However, it does not look like the driver implements any variables: https://github.com/networkupstools/nut/blob/v2.7.4/drivers/riello_ser.c#L758

clepple, 2019-11-04:
The authors of the driver seem to have been working on a patch, but it stalled:

* https://alioth-lists.debian.net/pipermail/nut-upsdev/2018-February/007374.html
* https://alioth-lists.debian.net/pipermail/nut-upsdev/2018-March/007388.html

@CentroricercheRPS do you have a version of the patch with `setvar`?

clepple, 2019-11-04:
@CentroricercheRPS Since the `ups.delay.shutdown` variable is sent at shutdown, and is not maintained in non-volatile memory on the UPS, @zykh's [2nd comment](https://alioth-lists.debian.net/pipermail/nut-upsdev/2018-March/007388.html) applies:
> - if they don't set values in the device, but only in NUT, add (and
> document in riello_{ser,usb} man pages) corresponding options to be
> set in ups.conf (or using the `-x var=value` command line option),
> since the driver, when launched with the `-k` flag or through
> `upsdrvctl shutdown`, cannot be aware of any 'transient' change made
> previously.

Skarafaz, 2019-11-05:
Thanks for the support @clepple!

Skarafaz, 2019-11-07:
@clepple just in case... what is the right way to deal with slow slaves without offdelay? Should I increase FINALDELAY variable in upsmon.conf?