networkupstools/nut #741: Powerpanel driver, no such file or directory.

thedarkb, 2019-11-12:
```
root@nutserver:/etc/nut# upsd
Network UPS Tools upsd 2.7.4
fopen /var/run/nut/upsd.pid: No such file or directory
listening on ::1 port 3493
listening on 127.0.0.1 port 3493
Can't connect to UPS [test1] (powerpanel-test1): No such file or directory
```
I'm trying to manage two CyberPower 1500AVR UPSs using NUT, but it responds with the preceding error message whenever I try to start upsd.

Here's the contents of my ups.conf, I've left off the second UPS for testing purposes, but it's identical and on /dev/ttyUSB1.

```
[test1]
       driver = powerpanel
       model = CP1500AVR
       port = /dev/ttyUSB0
       desc = "Mike's UPS"
```

clepple, 2019-11-12:
From https://networkupstools.org/docs/user-manual.chunked/ar01s02.html#_network_server :
> upsd is responsible for passing data from the drivers to the client programs via the network. It should be run immediately after `upsdrvctl` in your system’s startup scripts.

The driver(s) should be running before starting `upsd`, otherwise you will get the `Can't connect` message. (It should notice when a driver is restarted, though.)

Are you installing from packages or building from source? If the latter, is it possible that your driver was built from source, and was not configured to use the same directories as `upsd` from the packages? Or has the socket directory not been created yet? (Common when /var/run is a tmpfs.) The drivers and `upsd` communicate over a socket (in `/var/run/nut` for Debian/Ubuntu), and the socket name is in parentheses on the last line.

thedarkb, 2019-11-15:
I've installed from a package.

clepple, 2019-11-15:
If you run `upsdrvctl start`, do you get an error message from the driver?

clepple, 2019-11-15:
This likely isn't causing the issue you are seeing, but down the road, you may want to use a different path to the `/dev` node: https://unix.stackexchange.com/questions/382831/raspian-statically-map-dev-ttyusb#382913

The link says Raspbian, but it should apply to any distribution that uses `udev` (and there is one built into systemd, so that makes it fairly common).

That said, we can take a bit of guesswork out of the rest of the troubleshooting if you can provide a bit more detail about the distribution you are using.

PrplHaz4, 2019-11-15:
@thedarkb if you are connected over USB, you may want to try the usbhid driver

```
driver = usbhid-ups
model = CP1500AVR
port = /dev/ttyUSB0
desc = "Mike's UPS"
pollinterval = 15
```

clepple, 2019-11-15:
> if you are connected over USB, you may want to try the usbhid driver

Good point. What does lsusb show?

thedarkb, 2019-11-24:
I changed over to USB and I'm no longer having issues.

closed 2019-11-24 by thedarkb