networkupstools/nut #743: Warning: excessive poll failures adding APC NMC card

ssbarnea, 2019-12-07:
I tried adding the UPS and got:
```
[root@n0 ~]# snmp-ups -a apc
Network UPS Tools - Generic SNMP UPS driver 0.97 (2.7.4)
Duplicate driver instance detected! Terminating other driver!
No matching MIB found for sysOID '.1.3.6.1.4.1.318.1.3.27'!
Please report it to NUT developers, with an 'upsc' output for your device.
Going back to the classic MIB detection method.
Detected Smart-UPS RT 3000 XL on host 192.168.33.9 (mib: apcc 1.2)
[apc] Warning: excessive poll failures, limiting error reporting (OID = .1.3.6.1.4.1.318.1.1.1.9.2.3.1.5.1.1.3)
[apc] Warning: excessive poll failures, limiting error reporting (OID = .1.3.6.1.4.1.318.1.1.1.9.3.3.1.6.1.1.1)
```

How can I provide more info so we avoid the `No matching MIB found for sysOID '.1.3.6.1.4.1.318.1.3.27'!` line?

What is the purpose of these warnings, can we avoid it?

I would be happy to provide the output of upsc for my device but sadly that error line does not tel how to obtain this output.

``
cat /etc/ups/ups.conf
[apc]
driver = snmp-ups
port = 192.168.33.9
```

I tried to list output but all I got was:
```upsc -L
Error: Connection failure: No route to host
```

milkotodorov, 2021-03-05:
I have the same issue...

aquette, 2021-03-05:
Fyi, i never finished the fix but it's simply due to non existent OIDs. The reporting of your non supported sysoid is still valid though, to connect this fingerprint to the right OID map in nut

milkotodorov, 2021-03-05:
Thanks! It's an APC750I UPS with APC AP9630 NMC. This is not supported?

stevehop, 2021-03-06:
Same here with a SMT1500I and AP9631 NMC, OID is '.1.3.6.1.4.1.318.1.3.27'

Salvora, 2021-08-16:
same here with srt5kxli

linnemannr, 2021-09-19:
I'm really surprised nobody has done this yet, the 9630 UPS OIDs (on mine at least) are rooted at .1.3.6.1.2.1.33, so according to this[https://networkupstools.org/docs/developer-guide.chunked/ar01s04.html#snmp-subdrivers] it should be dirt simple to add support for this card. I'll check and if it hasn't been done or a PR isn't outstanding I'll give it a whack.

PR #1716 (2022-11-25): snmp-ups: fix "Warning: excessive poll failures"
Fix that long standing issue, which was tied to non-existent OIDs, not well handled in some part of the driver

Closes: #743
commit 857c057e (2022-11-25): snmp-ups: fix "Warning: excessive poll failures"

Fix that long standing issue, which was tied to non-existent OIDs,
not well handled in some part of the driver

Closes: #743
closed 2022-11-25 by jimklimov