networkupstools/nut #745: riello_usb driver: 'vendorid' is not a valid variable name for this driver

martin3142, 2019-12-13:
Using:

`nut-scanner -U`

gave the output:

```
Scanning USB bus.
[nutdev1]
        driver = "riello_usb"
        port = "auto"
        vendorid = "04B4"
        productid = "5500"
        product = "USB to Serial"
        vendor = "Cypress Semiconductor"
        bus = "003"
```

However, using that in /etc/nut/ups.conf causes the upsdrv service to fail with the error:

`'vendorid' is not a valid variable name for this driver`

The fix is to comment out or delete the offending lines to instead use:

```
[nutdev1]
        driver = "riello_usb"
        port = "auto"
        bus = "003"
```

And that works fine. (Aside: 'nutdev1' can be renamed as wished.)

Should the riello_usb driver be updated to accept those keyword values rather than error out? Or should nut-scanner be made aware to not output those superfluous lines?

Hopefully that is of help to other Riello UPS users.

Thanks,
Martin

tryfail, 2020-01-03:
> Using:
>
> `nut-scanner -U`
>
> gave the output:
>
> ```
> Scanning USB bus.
> [nutdev1]
>         driver = "riello_usb"
>         port = "auto"
>         vendorid = "04B4"
>         productid = "5500"
>         product = [...]

Just to comment that while the configuration successfully connects to the UPS it results in inaccurate data as reported by @martin3142 at  https://github.com/networkupstools/nut/issues/750

BrixSat, 2023-09-15:
Still having the same problem. Your post helped.

jimklimov, 2023-09-16:
I wonder if after #1763 got merged (currently only on master, not in a release) the matching options behaved better - e.g. the `vendorid` issue noted above.

closed 2024-08-10 by jimklimov