networkupstools/nut #746: Port PyNUT.py to python3

bigon, 2019-12-15:
Hello,

In addition to #521 PyNUT.py should also be ported to python3

The main problem here seems to be the fact that telnetlib is not accepting bytes:

```python
Traceback (most recent call last):
  File "/tmp/test.py", line 5, in <module>
    client.GetUPSList()
  File "/usr/lib/python3/dist-packages/PyNUT.py", line 127, in GetUPSList
    self.__srv_handler.write( "LIST UPS\n" )
  File "/usr/lib/python3.7/telnetlib.py", line 287, in write
    if IAC in buffer:
TypeError: 'in <string>' requires string as left operand, not bytes
```

Distributions are actively removing python2

seth-noaa, 2020-05-12:
Any update on the progress of this issue? It's nearing 6 months old.

jimklimov, 2021-11-14:
Should be fixed with #1115

closed 2021-11-14 by jimklimov