networkupstools/nut #747: Tecnoware UPS ERA PLUS 1100 supported by BLAZER_USB

bota87, 2019-12-15:
Tecnoware UPS ERA PLUS 1100
https://www.tecnoware.com/Prodotti/FGCERAPL1100/ups-era-plus-1100.aspx

```
Init SSL without certificate database
battery.charge: 100
battery.voltage: 13.60
battery.voltage.high: 13.00
battery.voltage.low: 10.40
battery.voltage.nominal: 12.0
device.type: ups
driver.name: blazer_usb
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.internal: 0.12
input.current.nominal: 2.0
input.frequency: 50.1
input.frequency.nominal: 50
input.voltage: 232.9
input.voltage.fault: 235.0
input.voltage.nominal: 230
output.voltage: 232.9
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 4
ups.productid: 5161
ups.status: OL
ups.type: offline / line interactive
ups.vendorid: 0665

```

```
Instant commands supported on UPS [ups]:

beeper.toggle - Toggle the UPS beeper
load.off - Turn off the load immediately
load.on - Turn on the load immediately
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
test.battery.start - Start a battery test
test.battery.start.deep - Start a deep battery test
test.battery.start.quick - Start a quick battery test
test.battery.stop - Stop the battery test
```

closed 2022-04-25 by jimklimov

jimklimov, 2025-01-15:
Regarding "Duplicate driver instance detected" - you probably have one running as a service, so should not be using `upsdrvctl` to start another - for more details see https://github.com/networkupstools/nut/wiki/nut%E2%80%90driver%E2%80%90enumerator-(NDE)

This may be why `upsc` takes a while, as the driver instances pull back and forth for the service to eventually revive after failure.

Another idea is that there may be lags with name resolution (`upsc upsname@hostname:port`)? Do other queries to that host (or `localhost` by default) work well? Does the server listen at only IPv4 or IPv6 while both protocol families are available (and `localhost` would point to both with only one working, so round-robining your attempts at best)?

jimklimov, 2025-01-15:
Guesstimation above seems reasonable, for runtimecal see the man pages.

jimklimov, 2025-01-16:
> So it has only one service running.

Actually, it has 3 services running in your report:
```
  nut-driver@tecnoware.service                                                         loaded active running   Network UPS Tools - device driver for tecnoware

  nut-monitor.service                                                                  loaded active running   Network UPS Tools - power device monitor and shutdown controller

  nut-server.service                                                                   loaded active running   Network UPS Tools - power devices information server
```

Just as their comments say, `nut-driver@tecnoware.service` instance wraps your driver (the instance that your `upsdrvctl` attempts kill off, and then are killed by as systemd revives the deceased unit).

The `nut-server.service` wraps your `upsd`, so starting another from command line in the latest post fails (ports are busy, packaged services apparently started the daemons in such a way that they did not leave a PID file, for other invocations to confirm the running process or communicate to it).

> It's expected to not have /run/nut/upsd.pid, because PID file is saved as /run/nut/nutdrv_qx-tecnoware.pid.
> Do I need to have upsd listening on IPv4/IPv6 at all for an USB connected UPS?

If I get your question right, you did not read up on NUT's 3-layer architecture? :)

In short, there are drivers that talk different protocols from a computer connected to the UPS using some media (dedicated like serial/USB, or networked). One or more drivers running on the computer are represented by one data server (`upsd`) with a common NUT protocol for clients over TCP/IP, regardless of whether they connect locally or remotely. The third layer is the clients, like `upsc`, `upscmd` or `upsmon`; some are read-only and anonymous, others need authenticated setup with certain roles on the data server.

jimklimov, 2025-01-16:
As for the logs, it can help to temporarily stop the running driver (service instance) and start a copy with increased debug verbosity, to post the log here, e.g.:
```
:; systemctl stop nut-driver@tecnoware.service

:; /lib/nut/nutdrv_qx -a tecnoware -DDDDDD -d 1
# Should pass one data-collection loop and exit; copy-paste the logs here

:; systemctl start nut-driver@tecnoware.service
```

This should clarify at least what it does about "guesstimation", etc.

If you did not yet set `runtimecal` - read about it in e.g. https://networkupstools.org/docs/man/nutdrv_qx.html and figure out the values valid for your UPS, battery, and attached load.

jimklimov, 2025-01-16:
To me, the bigger puzzle is the delays you report about starting the clients. Can the system be slow for whatever reason (age, storage, antiviruses, virtualisation, etc.) so there might be delays just about starting a program and dynamically linking `libnutclient.so` etc.?

Maybe a log of `strace -f -ttt -T upsc tecnoware` could be revealing about where it spends time...

jimklimov, 2025-01-16:
So, the driver start-up looks reasonable. NUT v2.8.2 has I think more debug in this area due to #1652 and related PRs. As noted before, I suggest trying to experiment, calibrate, time and configure the `runtimecal` for this driver to get a `battery.runtime` value as well.

As for the delay in connections, please double-check that `upsd` actually listens on `::1` port `3493`, and not some other random service; for example, by running `netstat -anp | grep 3493`. Also check what `localhost` resolves to and how (in `/etc/hosts` typically, but might also be in DNS; for many systems `getent localhost` should say).

Even if your `upsd` does not listen/respond on that port over IPv6 (maybe also due to firewall), but naming services tell the clients to go there, you may well get the timeout.

Also try `upsc tecnoware@127.0.0.1` explicitly, so it does not go to resolve `localhost` from string to numeric IP address - this one should go fast, I suppose, and rule out some other possible issues.