networkupstools/nut #749: nut-server.service fails to start because network is not brought up at unit start

Alphaprot, 2019-12-18:
**Issue**
The systemd-unit `nut-server.service` fails to start if the network is not brought up, so it cannot listen on network interfaces (neither localhost nor static IPs).

This is because the units all use the outdated ´network.target´ as their start requirement

**Recommended Solution**
Add the `network-online.target` at least in the `nut-server.service` like this:
```
[Unit]
Description=Network UPS Tools - power devices information server
After=local-fs.target network.target network-online.target nut-driver.service
Wants=nut-driver.service network-online.target
Before=nut-monitor.service
```
You can still keep `network.target` for compatibility purposes.

jimklimov, 2021-11-14:
Thanks for the suggestions, hope to have another look at that corner of the codebase: I believe this sort of changes was applied to 42ITy fork and hopefully NUT master branch over the past years, as the service units were redesigned to track each driver as a separate instance, and data-server and monitor too - each with their relevant dependency set on other units.

Etifloyd, 2023-01-04:
Thanks for the suggestion as I face this problem with my Raspberry pi running "Raspbian GNU/Linux 11 (bullseye)".
I made the change in the nut-server service, but no change. the network-online.target is triggered at the same time as network.target. I found in a forum (I don't find it again) that it's the same problem with debian distribution.
This only happen if the IP address is set to static in the dhcpcd.conf file. With a dynamic IP, it works fine.
As I want a fixed IP address, I have to uninstall dhcpcd and network manager and set a static IP in the /etc/network/interface file to get the nut-server starting correctly.

Etifloyd, 2023-01-04:
Thank you for your quick answer!
I listen to 127.0.0.1 and 192.168.10.2.
Nut server is version 2.7.4-13 for armhf which I simply install using "apt-get install nut" command.
I'll try listening to 0.0.0.0 by setting a static IP address in dhcpcd, but I have to setup an other raspberry for that. Will do that in a couple of days.
The output of the "apt-cache show nut" command:
> Package: nut-server
Source: nut
Version: 2.7.4-13
Architecture: armhf
Maintainer: Laurent Bigonville <bigon@debian.org>
Installed-Size: 4100
Pre-Depends: init-system-helpers (>= 1.54~)
Depends: adduser, lsb-base (>= 3.0-6), nut-client (= 2.7.4-13), udev, libc6 (>= 2.29), libnspr4 (>= 2:4.9-2~), libnss3 (>= 2:3.13.4-2~), libnutscan1 (>= 2.7.4-9~), libupsclient4 (>= 2.7.2), libusb-0.1-4 (>= 2:0.1.12), libwrap0 (>= 7.6-4~)
Suggests: nut-cgi, nut-ipmi, nut-snmp, nut-xml
Homepage: https://networkupstools.org/
Priority: optional
Section: admin
Filename: pool/main/n/nut/nut-server_2.7.4-13_armhf.deb
Size: 716080
SHA256: 6e9ea4b1e91615718d810d27132d25c5143bf5942af551d40e550e0c6d763f37
SHA1: a26ea981eef70ba7d9f2c9c98099e4c3c7284170
MD5sum: 14d5f5b115eaad14b3be377cbcde7030
Description: network UPS tools - core system
 Network UPS Tools (NUT) is a client/server monitoring system that
 allows computers to share uninterruptible power supply (UPS) and
 power distribution unit (PDU) hardware. Clients access the hardware
 through the server, and are notified whenever the power status
 changes.
 .
 This package provides NUT's core system, and the serial and USB UPS
 drivers.
Description-md5: cf50e3b4416f9edcc4a33c7f641661ef

jimklimov, 2023-01-05:
Thanks for the update.

Generally, whatever version is "released" and provided by a distro, is "set in stone" so there is not much we can fix about its code/logic if needed. So to try any non-configuration fixes on NUT side a custom build from Git may be required. (Not sure we currently need to, but it might come to that later).

Also note that since NUT 2.8.0 the (project-provided - distros may choose differently) service layout is different, with each driver being a `nut-driver@devicename.service` instance which may further depend each on different things according to communications media involved, and `nut-server` depending on the drivers via their target unit. (There is also a chicken-and-egg problem for e.g. dummy-ups and clone drivers retranslating something from a local upsd which was discussed elsewhere). Also now `upsd` has options to start even if drivers are not currently all running (or even defined), and can be `reload`'ed to connect to them later, which may be useful on certain deployments.

> I made the change in the nut-server service, but no change.

Just in case: did you apply it by `systemctl daemon-reload` (maybe reboot suffices, not sure) and `systemctl disable nut-server; systemctl enable nut-server` to generate the new "wants" symlinks?

> I'll try listening to 0.0.0.0 by setting a static IP address in dhcpcd, but I have to setup an other raspberry for that. 

That sentence sounds complicated...
* "listening to 0.0.0.0" - that you can try by editing `upsd.conf` and restarting the service/machine, independent of how it gets and address (or not)
* "setting a static IP address in dhcpcd" - I thought above you differentiated "setting a static/fixed IP address" in config files along with removing dhcpcd? The phrase here indicates still getting an IP address from a DHCP server, which just happens to be configured to reserve the same IP for certain MAC over eternity.

Etifloyd, 2023-01-11:
Dear Jim,

Thank you for your answer and sorry for the late response.
I was able to make a try today with an availabe Raspberry and a freshly installed OS.
Here are the step I did and the result:
1) Set a static IP in the /etc/dhcpcd.conf file (like it is explained on the Raspberry website)
2) Install nut with: apt-get install nut
3) Configure the UPS in the ups.conf file
4) Configure nut as netserver in the nut.conf file
5) Configure nut users in the upsd.user file
6) Configure the interface to listen in the upsd.conf file. At that moment I only put "LISTEN 127.0.0.1"
7) Reboot the Raspberry
8) The "upsc ups" command shows the information of the ups --> it's working
9) I add "LISTEN 192.168.10.120" in the upsd.conf file
10) Reboot the Raspberry
11) the upsc ups command fail --> nut server status is fail
12) I add the "network-online.target" in the nut-server.service file
13) stop, disable, enable and start the nut-server.service
14) Reboot the Rasberry
15) The "upsc ups" command shows the information of the ups --> it's working, the ups is accessible from network clients

In my first tests before I gave my feedback here, I didn't made the disable/enable trick to generate new "wants symlinks" as I wasn't aware of that stuff.
Today, I did this trick with disable/enable the service and it works.
Thank you for the tricks !!
Kind regards.

PR #1801 (2023-01-11): nut-{server,monitor,driver@}.service: clarify network-online.target dependency
Closes: #749
closed 2023-01-11 by jimklimov