networkupstools/nut #750: Bad data from riello_usb driver for Riello Net Power NPW 800

martin3142, 2019-12-18:
Using in ups.conf:

```
[riello]
       driver = "riello_usb"
       port = "auto"
#       vendorid = "04B4"
#       productid = "5500"
#       product = "USB to Serial"
#       vendor = "Cypress Semiconductor"
#       bus = "003"
       desc = "Muse08c Gateway Server"

```

with a usb connected Riello Net Power NPW 800 USB, the following reported data values are good/bad:

```
UPS Model: | USC6            Good?
Status: ONLINE                  Good (also indicates when running on battery, charging and fully charged)
Runtime: 1092:15:00          Bad
UPS temp: 255 °C              Bad
Battery: 13.6 V                    Good
Input: 239 V                        Good
Frequency: 50.00 Hz          Good
Output: 239  V                    Good
Load (phase1): 0 A             Bad
Load (phase2): 0 A             Bad
Load (phase3): 0 A             Bad
Frequency: 50.00 Hz           Good

Battery Charge: 255%         Bad
```

Note: The load values remain unchanged even when running from the battery (no mains).

Update needed for the Riello NPW series of UPS?

Thanks,
Martin

clepple, 2019-12-18:
You may find the discussion in issue #530 relevant. Apparently, 255 is the placeholder value used when a Riello UPS does not support a variable.

BrixSat, 2023-09-15:
Battery level a placeholder for a ups? Also a weird battery capacity.

Init SSL without certificate database
battery.capacity: 9
battery.charge: 255
battery.runtime: 3932100
battery.voltage: 26.4
battery.voltage.nominal: 24
device.mfr: RPS S.p.a.
device.model: USC6
device.serial:
device.type: ups
driver.name: riello_usb
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ttyUSB0
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.internal: 0.03
input.bypass.frequency: 409.50
input.bypass.voltage: 4095
input.frequency: 50.10
input.voltage: 239
output.frequency: 50.10
output.frequency.nominal: 50.0
output.L1.current: 0
output.L1.power: 0
output.L1.realpower: 0
output.L2.current: 0
output.L2.power: 0
output.L2.realpower: 0
output.L3.current: 0
output.L3.power: 0
output.L3.realpower: 0
output.power.percent: 0
output.voltage: 241
output.voltage.nominal: 230
ups.firmware: SWM048-01-04
ups.load: 0
ups.mfr: RPS S.p.a.
ups.model: USC6
ups.power.nominal: 2000
ups.productid: 5500
ups.realpower.nominal: 1200
ups.serial:
ups.status: OL
ups.temperature: 255
ups.vendorid: 04b4

jimklimov, 2023-09-16:
Well, other than `255` values, the `4095` or `409.5` above also look fishy (`4096-1`, where 4096 is a power of 2)...

Do you have a chance to test how it behaves with NUT v2.8.0 release, or with current master (there were some fixes for Qx drivers since the last release). For that matter, if a different driver like `nutdrv_qx` would fare better, by chance?

BrixSat, 2023-09-16:
Will do and report.

BrixSat, 2023-09-17:
Here you go.

```
upsdrvctl start
Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Riello USB driver 0.07 (2.8.0)
Warning: This is an experimental driver.
Some features may not function correctly.
```

```
Init SSL without certificate database
battery.capacity: 9
battery.voltage: 26.9
battery.voltage.nominal: 24
device.mfr: RPS S.p.a.
device.model: USC6           
device.serial:                 
device.type: ups
driver.name: riello_usb
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ttyUSB0
driver.parameter.synchronous: auto
driver.version: 2.8.0
driver.version.internal: 0.07
input.bypass.frequency: 409.50
input.bypass.voltage: 4095
input.frequency: 50.00
input.voltage: 241
output.frequency: 50.00
output.frequency.nominal: 50.0
output.L1.current: 0
output.L1.power: 0
output.L1.realpower: 0
output.L2.current: 0
output.L2.power: 0
output.L2.realpower: 0
output.L3.current: 0
output.L3.power: 0
output.L3.realpower: 0
output.power.percent: 0
output.voltage: 241
output.voltage.nominal: 230
ups.delay.reboot: 5
ups.delay.shutdown: 5
ups.firmware: SWM048-01-04
ups.load: 0
ups.mfr: RPS S.p.a.
ups.model: USC6           
ups.power.nominal: 2000
ups.productid: 5500
ups.realpower.nominal: 1200
ups.serial:                 
ups.status: OL
ups.vendorid: 04b4

```

BrixSat, 2023-09-18:
root@casinha:/home/brix/ups/nut# upsc ups
Init SSL without certificate database
battery.capacity: 9
battery.voltage: 26.9
battery.voltage.nominal: 24
device.mfr: RPS S.p.a.
device.model: USC6           
device.serial:                 
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: riello_usb
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.state: updateinfo
driver.version: 2.8.0-2512-gb94fc61ce
driver.version.internal: 0.10
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.bypass.frequency: 409.50
input.bypass.voltage: 4095
input.frequency: 50.10
input.voltage: 239
output.frequency: 50.00
output.frequency.nominal: 50.0
output.L1.current: 0
output.L1.power: 0
output.L1.realpower: 0
output.L2.current: 0
output.L2.power: 0
output.L2.realpower: 0
output.L3.current: 0
output.L3.power: 0
output.L3.realpower: 0
output.power.percent: 0
output.voltage: 239
output.voltage.nominal: 230
ups.delay.reboot: 5
ups.delay.shutdown: 5
ups.firmware: SWM048-01-04
ups.load: 0
ups.mfr: RPS S.p.a.
ups.model: USC6           
ups.power.nominal: 2000
ups.productid: 5500
ups.realpower.nominal: 1200
ups.serial:                 
ups.status: OL
ups.vendorid: 04b4

grifferz, 2024-03-01:
I've also got a Riello NPW 1500 and using nut as packaged in Debian 10

```
$ upsd -V
Network UPS Tools upsd 2.7.4
```

Using riello_usb I am getting similar strange values as OP for `battery.capacity`, `battery.charge` etc:

```
$ upsc riello 
Init SSL without certificate database
battery.capacity: 9
battery.charge: 255                                                                   
battery.runtime: 3932100
battery.voltage: 27.1
battery.voltage.nominal: 24
device.mfr: RPS S.p.a.
device.model: USC6         
device.serial:
device.type: ups                                                   
driver.name: riello_usb                 
driver.parameter.pollinterval: 2     
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4   
driver.version.internal: 0.03
input.bypass.frequency: 409.50
input.bypass.voltage: 4095
input.frequency: 49.80                                             
input.voltage: 241
output.frequency: 49.80     
output.frequency.nominal: 50.0      
output.L1.current: 0        
output.L1.power: 0
output.L1.realpower: 0
output.L2.current: 0
output.L2.power: 0                                             
output.L2.realpower: 0   
output.L3.current: 0                                 
output.L3.power: 0                            
output.L3.realpower: 0               
output.power.percent: 0          
output.voltage: 241             
output.voltage.nominal: 230              
ups.firmware: SWM048-01-05                           
ups.load: 0                                         
ups.mfr: RPS S.p.a.      
ups.model: USC6          
ups.power.nominal: 1500                                            
ups.productid: 5500
ups.realpower.nominal: 900  
ups.serial:                                                        
ups.status: OL
ups.temperature: 255                 
ups.vendorid: 04b4
```

Is this improved in any later version of nut or does this device simply not expose this information? Though like OP, a UPS that can't show battery charge seems weird.

I haven't yet tried Riello's own software to see if it sees useful values.

This device also has a serial port (as well as USB). Is there any chance that useful info might come from a serial cable? I'm assuming not as the USB is just a USB serial and the protocol is the same, right?

jimklimov, 2024-03-01:
CC @mzampieri70 : Would you by chance know if the "weird" numbers like 255 and 4095 (or 10-based multiples of that), which I suppose mean `-1` in 8-bit and 12-bit signed words, as seen in reports above are a hardware limitation with data not served by those models, or if there is some other protocol nuance to get them and the NUT drivers should be updated for Riello UPSes?

grifferz, 2024-03-01:
Built from HEAD using the instructions at https://github.com/networkupstools/nut/wiki/Building-NUT-on-Debian,-Raspbian-and-Ubuntu and while I do see a newer version of riello_usb, all it seems to do differently is filter out `battery.charge` and `battery.runtime` from being displayed. `battery.capacity` still says 9.

As mentioned, this device also has a DB9 serial port. If I connect that and use riello_ser would I expect any different behaviour?

This is a disappointment as looking at Riello NPW 1500 on the HCL it shows up with riello_usb as green and 5 stars. Wish I had done a more thorough search to find this issue. 🙁

jimklimov, 2024-03-02:
Well, I think it would not hurt to try the serial port, if you have the chance. I know *some* UPS vendors do have certain internal data representation in the controller, and filter code to represent that over Serial, USB, SNMP, Web, on-board LCD panel etc. and the results are sometimes different, both in precision and in the general collection of data points or commands made available.

I have no idea if specifically Riello devices have similar architecture/behavior.

grifferz, 2024-03-02:
I'll try the serial cable tomorrow as I just discovered that it needs a male-female serial cable and all my cables are female-female.

In the meantime I ran Riello's own PowerShield software for Linux and it only shows the same data as nut does.

In light of numerous issues I can find here in github it seems like the low end Riello devices mostly do not provide charge or runtime data. Is there some way to get that info into the HCL?

Over in issue #1692 I see someone contributed a riello_usb option to locally guess runtime and charge from nominal values and current battery voltage. Am I right in thinking this hasn't yet been committed and there is nothing like it currently in nut, so if I wanted to do that I'd need to patch that into riello_usb myself?

jimklimov, 2024-03-02:
Yes, so far that issue (and PR #1692 from it) has not got merged; I remember there was something in the review that stalled me - but gotta look again if it was anything serious.

For now, updated that PR's source branch with master-branch code, so testing with custom builds are easier.