networkupstools/nut #751: upsmon -c fsd command in pfsense master does not shutdown Synology slave

kevindd992002, 2019-12-31:
I have pfsense set as the NUT master and my Synology NAS as a slave client.

I tried recreating a shutdown sequence by issuing a /usr/local/sbin/upsmon -c fsd command in pfsense (so as to preserve battery lifetime while testing all these) but the weird thing is that my Syno doesn't seem to get a hold of the FSD event. My pfsense master and UPS shuts down properly though. As soon as I issue the FSD, the /var/log/messages in my Syno says this:

2019-12-31T02:05:03+08:00 Synology synoups: [UPS] Check Boot Status.
2019-12-31T02:05:03+08:00 Synology synoups: WARNING: UPS is On-Line

But when I recreate the shutdown sequence by unplugging the mains, all goes well and the Syno picks up the OB LB FSD event properly. Does this mean that Syno needs to see that the UPS is in OB LB state to issue a shutdown even if it sees an FSD event already?

kevindd992002, 2020-01-02:
@clepple Do you have any comments on this?

clepple, 2020-01-02:
I don't have a Synology, but `WARNING: UPS is On-Line` seems to indicate that they are looking for OB status.

Do they publish source for the Synology code?

kevindd992002, 2020-01-03:
I'm not sure with the Synology code but I would imagine it is not open-source as it is a commercial product.

clepple, 2020-01-03:
Right, but it's apparently a commercial product using open-source packages (NUT, for one). That's not to say they have to publish everything, but it sure would be handy to have access to some of the code for situations like this. It also could be a shell script logging that message, so it is probably worth poking around with `grep` (starting with some of the scripts referenced by e.g. `upsmon.conf`).

You could also ask in their support forum.

aquette, 2020-01-06:
hi fellows,
all my best wishes for 2020.

FYI, I've just asked to one of my contact at Synology, if some of their tech could consider this, and collaborate with us.

kevindd992002, 2020-01-06:
Thanks @aquette! Please keep us posted.

gdt, 2023-01-07:
Any news?

 * upsmon.c is GPL2, so if synology includes a binary from that source they have to deliver sources.
 * It may be that OL and FSD confuse the remote upsmon.   It's obviously a shutdown drill so the behavior seems buggy
 * If you have a problem with Synology behavior the right place to address that is with Synology.

jimklimov, 2023-01-07:
Technically, by license they should provide source to users of their binary upon request. Publishing tar/git on the internet is a convenience for both sides, but not a requirement.

vneno, 2023-06-17:
IMO, it is cleary Synology bug. 
Here is a simple and fast workaround:
on Synology:
add function to the /usr/syno/bin/synoups:
NASHalt() {
    /sbin/shutdown -h now
}
...
and edit cases (at the end of the script):
fsd)
    SynoUpsStateLog "synoups fsd"
    ...
    **NASHalt**
    ;;

you can also modify other cases (lowbatt...)
restart ups service:
systemctl restart ups-net.service

Then try FSD again.