networkupstools/nut #754: NSYS E series not supported

lkraider, 2020-01-08:
Hi, I am trying to setup the [NSYS SERIE E ONLINE 3kVA UPS](https://nsys.ind.br/produto/nobreak-serie-e-3kva/
), but it seems not to be supported.

As it creates a HID device, I tried the generic hid driver, here is the output:
https://gist.github.com/lkraider/729223f16025c1ff6223536896465e36

I also tried the blazer_usb driver, but all data is read as zeros:
https://gist.github.com/lkraider/21ab0de58d3cb6905fe658f30c1d47ec

Also the nutdrv_qx reads only zeros:
https://gist.github.com/lkraider/515e542b004538cc03313913f9b190a8

NSYS has their own software page, and they use UPSmart to connect over USB to the UPS:
https://nsys.ind.br/softwares/

Any suggestions on how to proceed, or what to try? I appreciate any help.

Environment:
- nut: version 2.7.4
- OS: Proxmox 5.4

lkraider, 2020-01-08:
I managed to get an USB trace from the windows software.

I sent a few commands using the interface, including switch to battery test mode, cancel battery test mode and switch buzz on/off:
[ups-trace.log](https://github.com/networkupstools/nut/files/4037602/ups-trace.log)

aquette, 2020-01-09:
hi @lkraider 
could you please post here (or attach a zip) of the generic hid output, since I can't currently access it (forbidden from work).
if it's indeed a HID device, very possibly with specific usages (i.e. non standard), we'll have to create a new HID subdriver, referencing these, as visible for example here: https://github.com/networkupstools/nut/blob/master/drivers/mge-hid.c#L813
And then create the variables mapping, according to that, like here: https://github.com/networkupstools/nut/blob/master/drivers/mge-hid.c#L1176

clepple, 2020-01-09:
I haven't had a chance to look at this thoroughly, but @zykh was helping with another MEC0003 device in #638. That branch might be a good starting point.

lkraider, 2020-01-09:
I think this is a Mega* (megatec?) protocol, as it is what the UPSmart program calls the connection type, but I don't know.

Here is the generichid output:
[ nut-nsys-hid-explore.zip](https://github.com/networkupstools/nut/files/4042264/729223f16025c1ff6223536896465e36-e2a1ca7b06a013cea7300da010c29f2fa4f68269.zip)
 output:

lkraider, 2020-06-16:
Is there any way to try and access the HID Report Descriptor output for each item? I parsed it using an online tool, this is the output: https://gist.github.com/lkraider/7feb68d6cd68c0e7091600bdfa3c5435

aquette, 2020-06-16:
When using usbhid-ups in debug mode, you can see the various USB and hid binary structs. Try with -DDDDD