networkupstools/nut #756: Support for PowerWalker VI 2200 SHL

schumar, 2020-01-22:
PowerWalker VI 2200 SHL
```
$ lsusb | grep 06da
Bus 003 Device 007: ID 06da:ffff Phoenixtec Power Co., Ltd 

$ upsc powerwalker
battery.charge: 98
battery.runtime: 3292
battery.type: PbAc
battery.voltage: 27
device.mfr: PPC
device.model: Offline UPS
device.serial: 000000000   
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: ffff
driver.parameter.synchronous: no
driver.parameter.vendorid: 06da
driver.version: 2.7.4
driver.version.data: Phoenixtec/Liebert HID 0.3
driver.version.internal: 0.41
ups.load: 0
ups.mfr: PPC
ups.model: Offline UPS
ups.productid: ffff
ups.serial: 000000000   
ups.status: OL CHRG
ups.vendorid: 06da

# /lib/nut/usbhid-ups -DD -a powerwalker
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000     debug level is '2'
   0.000335     upsdrv_initups...
   0.025897     Checking device (1D6B/0003) (004/001)
   0.025999     - VendorID: 1d6b
   0.026006     - ProductID: 0003
   0.026016     - Manufacturer: unknown
   0.026022     - Product: unknown
   0.026032     - Serial Number: unknown
   0.026036     - Bus: 004
   0.026041     - Device release number: 0504
   0.026046     Trying to match device
   0.026055     Device does not match - skipping
   0.026178     Checking device (06DA/FFFF) (003/007)
   0.049878     - VendorID: 06da
   0.049893     - ProductID: ffff
   0.049897     - Manufacturer: PPC
   0.049901     - Product: Offline UPS
   0.049905     - Serial Number: 000000000   
   0.049909     - Bus: 003
   0.049912     - Device release number: 0003
   0.049918     Trying to match device
   0.049976     Device matches
   0.053883     HID descriptor length 495
   0.117884     Report Descriptor size = 495
   0.118114     Using subdriver: Phoenixtec/Liebert HID 0.3
   0.118123     45 HID objects found
   0.120893     Path: UPS.BatterySystem.Battery.Test, Type: Feature, ReportID: 0x24, Offset: 0, Size: 8, Value: 6
   0.123886     Path: UPS.Flow.[4].ConfigApparentPower, Type: Feature, ReportID: 0x74, Offset: 0, Size: 16, Value: 9
   0.126885     Path: UPS.Flow.[4].ConfigFrequency, Type: Feature, ReportID: 0x36, Offset: 0, Size: 8, Value: 50
   0.129847     Path: UPS.Flow.[4].ConfigVoltage, Type: Feature, ReportID: 0x72, Offset: 0, Size: 8, Value: 230
   0.132871     Path: UPS.PowerConverter.ffff0041, Type: Feature, ReportID: 0x4a, Offset: 0, Size: 8, Value: 1
   0.135873     Path: UPS.PowerConverter.Input.[1].Voltage, Type: Feature, ReportID: 0x31, Offset: 0, Size: 16, Value: 233
   0.138871     Path: UPS.PowerConverter.Input.[1].PresentStatus.Boost, Type: Feature, ReportID: 0x2d, Offset: 0, Size: 8, Value: 0
   0.138889     Path: UPS.PowerConverter.Input.[1].PresentStatus.Boost, Type: Input, ReportID: 0x2d, Offset: 0, Size: 8, Value: 0
   0.138905     Path: UPS.PowerConverter.Input.[1].PresentStatus.Buck, Type: Feature, ReportID: 0x2d, Offset: 8, Size: 8, Value: 0
[... 84 lines cut ...]
 00 67 21 D1 F0 00 55 07 B1 83 09 02 A1 02 09 6E
 85 2D 75 08 26 FF 00 65 00 55 00 B1 83 09 6E 81
 83 09 6F B1 83 09 6F 81 83 C0 C0 05 84 09 1C A1
 00 05 84 09 32 85 42 75 10 27 FF FF 00 00 66 01
 F0 55 0F B1 83 05 84 09 30 85 42 75 10 27 FF FF
 00 00 67 21 D1 F0 00 55 07 B1 83 C0 C0 09 24 A1
 00 09 5A 85 13 75 08 15 00 26 FF 00 65 00 55 00
 B1 82 05 84 09 57 85 09 75 20 15 FF 27 FF FF FF
 7F 66 01 10 B1 82 09 56 85 0A B1 82 05 85 09 89
 85 10 75 08 15 00 26 FF 00 65 00 B1 03 05 84 09
 FD B1 03 06 FF FF 09 F0 B1 03 05 84 09 FE B1 03
 05 84 09 FF B1 03 06 FF FF 09 E2 85 E2 B1 83 05
 84 09 35 85 07 15 00 26 FF 00 65 00 B1 83 05 84
 09 02 A1 02 05 85 09 D0 85 01 75 08 B1 83 09 D0
 81 83 09 42 B1 83 09 42 81 83 09 44 B1 83 09 44
 81 83 05 85 09 45 B1 83 09 45 81 83 05 84 09 61
 B1 83 09 61 81 83 09 62 B1 83 09 62 81 83 05 84
 09 65 B1 83 09 65 81 83 C0 05 85 85 0B 75 08 95
 01 15 00 25 64 09 83 B1 03 09 8D B1 03 09 67 65
 00 B1 03 09 2C 65 00 B1 03 09 66 85 06 65 00 B1
 83 09 66 81 83 05 85 09 68 85 06 75 10 26 FF 7F
 66 01 10 B1 83 09 68 81 83 05 84 09 30 85 07 75
 10 26 FF 7F 67 21 D1 F0 00 55 06 B1 83 C0 C0

```

schumar, 2020-01-22:
p.s. this is indeed without any load connected yet, just unpacked it :)

tritonblaster, 2020-04-09:
same here..

I would love to be able to control for example the beep or get all values by upsc command

I did some quick and dirty fixes to get more information for my Monitoring:
`$ upsc powerwalker 2>/dev/null
battery.charge: 100
battery.runtime: 2073
battery.type: PbAc
battery.voltage: 27
device.mfr: PPC
device.model: Offline UPS
device.serial: 000000000   
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: ffff
driver.parameter.synchronous: no
driver.parameter.vendorid: 06da
driver.version: 2.7.4
driver.version.data: Phoenixtec/Liebert HID 0.3
driver.version.internal: 0.41
ups.load: 14
ups.mfr: PPC
ups.model: Offline UPS
ups.productid: ffff
ups.serial: 000000000   
ups.status: OL CHRG
ups.vendorid: 06da`

`$ echo "$(timeout 1 /lib/nut/usbhid-ups -a powerwalker -D 2>&1 | grep Path | sed -e "s/.*UPS\.//" -e "s/,.*Value//")" | uniq
BatterySystem.Battery.Test: 6
Flow.[4].ConfigApparentPower: 9
Flow.[4].ConfigFrequency: 50
Flow.[4].ConfigVoltage: 230
PowerConverter.ffff0041: 1
PowerConverter.Input.[1].Voltage: 237
PowerConverter.Input.[1].PresentStatus.Boost: 0
PowerConverter.Input.[1].PresentStatus.Buck: 0
PowerConverter.Output.Frequency: 49.6
PowerConverter.Output.Voltage: 237
PowerSummary.AudibleAlarmControl: 2
PowerSummary.DelayBeforeShutdown: 0
PowerSummary.DelayBeforeStartup: 0
PowerSummary.iDeviceChemistry: 5
PowerSummary.iManufacturer: 1
PowerSummary.ffff00f0: 3
PowerSummary.iProduct: 2
PowerSummary.iSerialNumber: 4
PowerSummary.ffff00e2: 4
PowerSummary.PercentLoad: 14
PowerSummary.PresentStatus.ACPresent: 1
PowerSummary.PresentStatus.BelowRemainingCapacityLimit: 0
PowerSummary.PresentStatus.Charging: 1
PowerSummary.PresentStatus.Discharging: 0
PowerSummary.PresentStatus.Good: 1
PowerSummary.PresentStatus.InternalFailure: 0
PowerSummary.PresentStatus.Overload: 0
PowerSummary.DesignCapacity: 100
PowerSummary.CapacityGranularity1: 1
PowerSummary.FullChargeCapacity: 100
PowerSummary.CapacityMode: 2
PowerSummary.RemainingCapacity: 100
PowerSummary.RunTimeToEmpty: 2073
PowerSummary.Voltage: 27.2`

closed 2022-04-25 by jimklimov