networkupstools/nut #764: How to configure when new bigger battery?

OhSoGood, 2020-03-10:
Hi,

I've replaced the 2 batteries of my old MGE Ellipse Premium 800 with 2 bigger ones, which last much longer. Is there a way to configure or tweak nut (and usbhid-ups) to manage this? 

With a continuous power consumer (a 50W lamp) the UPS lasts 2h20 but, if it is connected to a PC with nut active, the battery.charge value drops very fast and after 15 minutes nut shuts everything down... (battery.charge.low is currently 30 but even if I lower it, it may delay only a little the shutdown).

Any idea?

Just for info, my old battery: two CSB HR1224WF2F1 (5Ah), the new one: two KYOTO Ytx9-bs (8Ah)

clepple, 2020-03-10:
5Ah to 8Ah shouldn't confuse the UPS too much, but you should run a battery test on the UPS to recalibrate `battery.charge`. If your UPS is similar to this [Pulsar Ellipse Premium 800](https://networkupstools.org/ddl/MGE_UPS_SYSTEMS/Pulsar_Ellipse_Premium_800.html), then you should have the `test.battery.start` and `test.battery.stop` instant commands available through `upscmd`. In any case, there should be a button or something to trigger the test.

However, if the `battery.charge` value still drops quickly, you may want to check and make sure that the UPS is capable of charging at the higher current needed by the larger batteries.

OhSoGood, 2020-03-18:
Hi @clepple , we are indeed speaking of the same ups, MGE is the manufacturer, the full model name is Pulsar Ellipse Premium 800. I wonder if your link https://networkupstools.org/ddl/MGE_UPS_SYSTEMS/Pulsar_Ellipse_Premium_800.html is not wrong: my ups doesn't expose the test.battery.* commands, the ups just has a power on/off button, and the official documentation does not mention any testing or calibration feature http://lit.powerware.com/ll_download.asp?file=Pulsar%20Ellipse%20-%20Installation%20and%20User%20Manual%20-%20Multi%20Lang.pdf .

So the issue remains open... Any idea? Thank you!