networkupstools/nut #770: SNMPv3 AES support for APC AP96xx clarification

userrw, 2020-03-29:
This could be a documentation issue or something else.

[The hardware compatibility entry](http://www.networkupstools.org/stable-hcl.html?manufacturer=APC&connection=Network) for "AP9630 SNMP monitoring card" lists "privProtocol=AES" as a parameter for snmp-ups driver. According to [a message](https://alioth-lists.debian.net/pipermail/nut-upsuser/2014-April/008979.html) from the mailing list: "Typically, the HCL lists parameters which are needed to make things work."

However, at least for a few users, only "privProtocol=DES" works with AP96xx cards. This seems to have caused some confusion over the years:

[https://alioth-lists.debian.net/pipermail/nut-upsuser/2014-April/008982.html](https://alioth-lists.debian.net/pipermail/nut-upsuser/2014-April/008982.html) (AP9630)
[https://alioth-lists.debian.net/pipermail/nut-upsdev/2016-April/007174.html](https://alioth-lists.debian.net/pipermail/nut-upsdev/2016-April/007174.html) (AP9630)

[https://github.com/networkupstools/nut/issues/700](https://github.com/networkupstools/nut/issues/700) (AP9617)
[https://github.com/networkupstools/nut/issues/734](https://github.com/networkupstools/nut/issues/734) (AP9631)

If no AP96xx is able to use AES, maybe the HCL parameter should be changed to "privProtocol=DES".

What I have found, with the following setup, is NUT connecting to a AP9630 via SNMPv3 with AES as the privacy protocol, cannot connect to UPS. snmpwalk (5.6.2.1) using AES connects and NUT using DES connects. I have also seen the same behaviour (only DES works) on 1 other machine running FreeNAS and on another 3 machines running Synology DSM with 4 different AP9630 cards.

Please let me know if I can do anything to help. Thanks.

**APC AP9630 SNMP monitoring card
Hardware Revision 05
Application Module sumx v6.8.0
APC OS (AOS) v6.8.2
APC Boot Monitor v1.0.9**
 
**Network UPS Tools upsd 2.7.4
NUT UPS driver controller 2.7.4
NUT Generic SNMP UPS driver 0.97 (2.7.4)**

**pfsense 2.4.5-RELEASE
physical server**

**ups.conf:**
```
[UPS]
driver=snmp-ups
port=XXX.XXX.XXX.XXX
snmp_version=v3
secLevel=authPriv
secName=XXXX
authPassword=XXXX
privPassword=XXXX
authProtocol=SHA
privProtocol=AES
```

userrw, 2020-04-11:
@clepple do you maybe have any input as you replied on the 2014 thread?

clepple, 2020-04-12:
The snmp-ups driver is a bit of a black box to me - I was just asking for more information.

It's even more mystifying when `snmpwalk` works, and `snmp-ups` doesn't (since they use the same library underneath). However, that seems like something to compare. This may be naive, since I have only really looked at unencrypted SNMPv1 traffic over the network, but are there any obvious differences visible in Wireshark/tcpdump?

userrw, 2020-04-15:
@clepple I'm glad I'm not the only one who is mystified!

Before going the packet capture route, following your comment about it being the same library, I looked at snmp-ups a bit more. It would appear that when using AES, NUT can't get sysOID value which results in no supported device detected.

DES:
```
Network UPS Tools - Generic SNMP UPS driver 0.97 (2.7.4)
   0.000000	debug level is '5'
   0.000530	SNMP UPS driver: entering upsdrv_initups()
   0.000541	SNMP UPS driver: entering nut_snmp_init(snmp-ups)
   0.029699	Setting SNMP retries to 10
   0.029726	Setting SNMP timeout to 2 second(s)
   0.048672	SNMP UPS driver: entering load_mib2nut(auto)
   0.048683	trying the new match_sysoid() method
   0.048688	Entering nut_snmp_get_oid()
   0.048693	nut_snmp_get(.1.3.6.1.2.1.1.2.0)
   0.048697	nut_snmp_walk(.1.3.6.1.2.1.1.2.0)
   0.048701	nut_snmp_walk: max. iteration = 1
   0.101196	OID value: .1.3.6.1.4.1.318.1.3.27
```

AES:
```
Network UPS Tools - Generic SNMP UPS driver 0.97 (2.7.4)
   0.000000	debug level is '5'
   0.000527	SNMP UPS driver: entering upsdrv_initups()
   0.000538	SNMP UPS driver: entering nut_snmp_init(snmp-ups)
   0.029744	Setting SNMP retries to 10
   0.029770	Setting SNMP timeout to 2 second(s)
   0.048784	SNMP UPS driver: entering load_mib2nut(auto)
   0.048795	trying the new match_sysoid() method
   0.048800	Entering nut_snmp_get_oid()
   0.048806	nut_snmp_get(.1.3.6.1.2.1.1.2.0)
   0.048810	nut_snmp_walk(.1.3.6.1.2.1.1.2.0)
   0.048814	nut_snmp_walk: max. iteration = 1
   0.076320	Can't get sysOID value
```

snmpwalk (5.6.2.1 and 5.7.3) for .1.3.6.1.2.1.1.2.0 returns the following with both DES and AES:
`SNMPv2-MIB::sysObjectID.0 = OID: SNMPv2-SMI::enterprises.318.1.3.27`

Does that help or would a packet capture still be useful?

clepple, 2020-04-15:
In order to get past that point, we would need to know *why* the driver is returning `Can't get sysOID value`. If I am not mistaken, the driver is already sending requests using `authProtocol` and `privProtocol` in order to retrieve that OID. (There is a small possibility that you can force things by specifying `mibs=apcc` in the configuration file, but I think that will just move the error to the next place where the driver needs to call `nut_snmp_get`.)

An alternative to a packet capture would be to try setting some of the debug options in `snmp.conf`: http://www.net-snmp.org/docs/man/snmp.conf.html (not sure where this file would be located for pfsense, but probably under /usr/local if NUT and Net-SNMP were both built from the ports tree).

In either case, you might want to temporarily set the password to something trivial in case it is revealed.

userrw, 2020-04-17:
@clepple Of course, makes sense, thanks.

I had tried `mibs=apcc` and it does exactly what you expected.

So I did 4 packet captures between pfsense (`snmpwalk` 5.7.3) and the AP9630: `snmpwalk` DES, `snmpwalk` AES, `snmp-ups` DES and  `snmp-ups` AES:

For all successful attempts (`snmpwalk` and `snmp-ups` DES):
```
pfsense	AP9630	SNMP	106	get-request
AP9630	pfsense	SNMP	147	report 1.3.6.1.6.3.15.1.1.4.0
pfsense	AP9630	SNMP	173	encryptedPDU: privKey Unknown
AP9630	pfsense	SNMP	392	encryptedPDU: privKey Unknown
```

`snmp-ups` AES:
```
pfsense	AP9630	SNMP	106	get-request
AP9630	pfsense	SNMP	147	report 1.3.6.1.6.3.15.1.1.4.0
```

For all successful attempts the last 2 messages are then repeated until the end. For the unsuccessful attempt it stops there.

There doesn't seem to be any abnormal differences comparing the first 2 frames but here are the lines that differ:

`snmpwalk` DES `get-request`:
```
Simple Network Management Protocol
    msgGlobalData
        msgID: 2101821612
    msgData: plaintext (0)
        plaintext
            data: get-request (0)
                get-request
                    request-id: 1414842185
```

`snmpwalk` DES `report 1.3.6.1.6.3.15.1.1.4.0`:
```
Simple Network Management Protocol
    msgGlobalData
        msgID: 2101821612
    msgAuthoritativeEngineTime: 1418080
    msgData: plaintext (0)
        plaintext
            data: report (8)
                report
                    request-id: 1414842185
                    variable-bindings: 1 item
                        1.3.6.1.6.3.15.1.1.4.0: 36
                            Value (Counter32): 36
```

`snmp-ups` DES `get-request`:
```
Simple Network Management Protocol
    msgGlobalData
        msgID: 478653506
    msgData: plaintext (0)
        plaintext
            data: get-request (0)
                get-request
                    request-id: 1110883205
```

`snmp-ups` DES `report 1.3.6.1.6.3.15.1.1.4.0`:
```
Simple Network Management Protocol
    msgGlobalData
        msgID: 478653506
    msgAuthoritativeEngineTime: 1417412
    msgData: plaintext (0)
        plaintext
            data: report (8)
                report
                    request-id: 1110883205
                    variable-bindings: 1 item
                        1.3.6.1.6.3.15.1.1.4.0: 32
                            Value (Counter32): 32
```

`snmpwalk` AES `get-request`:
```
Simple Network Management Protocol
    msgGlobalData
        msgID: 58171569
    msgData: plaintext (0)
        plaintext
            data: get-request (0)
                get-request
                    request-id: 1810943879
```

`snmpwalk` AES `report 1.3.6.1.6.3.15.1.1.4.0`:
```
Simple Network Management Protocol
    msgGlobalData
        msgID: 58171569
    msgAuthoritativeEngineTime: 1418080
    msgData: plaintext (0)
        plaintext
            data: report (8)
                report
                    request-id: 1810943879
                    variable-bindings: 1 item
                        1.3.6.1.6.3.15.1.1.4.0: 37
                            Value (Counter32): 37
```

`snmp-ups` AES `get-request`:
```
Simple Network Management Protocol
    msgGlobalData
        msgID: 727884324
    msgData: plaintext (0)
        plaintext
            data: get-request (0)
                get-request
                    request-id: 1830754979
```

`snmp-ups` AES `report 1.3.6.1.6.3.15.1.1.4.0`:
```
Simple Network Management Protocol
    msgGlobalData
        msgID: 727884324
    msgAuthoritativeEngineTime: 1418080
    msgData: plaintext (0)
        plaintext
            data: report (8)
                report
                    request-id: 1830754979
                    variable-bindings: 1 item
                        1.3.6.1.6.3.15.1.1.4.0: 39
                            Value (Counter32): 39
```

No error is reported for any of the frames and the `Object Name` remains constant:

All `get-request` frames (`X` changes):
```
Simple Network Management Protocol
    msgData: plaintext (0)
        plaintext
            data: get-request (0)
                get-request
                    request-id: XXXXXXXXXX
                    error-status: noError (0)
                    error-index: 0
                    variable-bindings: 0 items
```

All `report 1.3.6.1.6.3.15.1.1.4.0` frames (`X` changes):
```
Simple Network Management Protocol
    msgData: plaintext (0)
        plaintext
            data: report (8)
                report
                    request-id: XXXXXXXXXX
                    error-status: noError (0)
                    error-index: 0
                    variable-bindings: 1 item
                        1.3.6.1.6.3.15.1.1.4.0: XX
                            Object Name: 1.3.6.1.6.3.15.1.1.4.0 (iso.3.6.1.6.3.15.1.1.4.0)
                            Value (Counter32): XX
```

I looked at `snmp.conf`, I guess the 2 interesting options are `dumpPacket` and `doDebugging`. However, I'm not entirely sure on what actions to do after setting them. Could you explain what tests to run?

Users are only for testing and are deleted.

If easier, I also have multiple VMs (Debian, CentOS, etc.) available for testing.

userrw, 2020-06-09:
Sorry to bother you @clepple , was this the info you were looking for?

userrw, 2020-12-10:
@Bi11 Thank you! Closure feels good, even if it isn't closed!

jimklimov, 2020-12-13:
I gather it should work in builds of current master branch. Knowing whether it does for you (and so is it a closure?) would help :)

userrw, 2021-02-03:
@jimklimov Tested today and ups.status: OL. Hooray!

closed 2021-02-03 by userrw