networkupstools/nut #771: Multiple identical UPS on the same USB bus

janipewter, 2020-04-09:
I have two identical basic APC UPS connected to a Raspberry Pi 4B. They show up in ```lsusb``` with the same vendor and product ID (as expected).

However all four of the USB ports on the Pi are on the same USB bus, the only thing that differs between the devices is the USB device ID.

In NUT you can tell it which USB bus to look for but not which device. The result is that the output of ```upsc UPS-1@localhost``` and ```upsc UPS-2@localhost``` are identical, because it's reading both from just one UPS and there is no way to tell NUT which one is which.

Is there any workaround for this? Such as spoofing the USB bus ID? I did try putting a USB hub between the Pi and one of the UPS but it still appears on the same bus. I am at a loss for what to do now.

clepple, 2020-04-09:
Are you sure there is no difference? Check for serial numbers - if you use `lsusb -v`, run it as root, because retrieving the serial number requires write access. (Or you can use `dmesg`)

Device IDs are not stable. If you unplug an UPS and plug it back in, you will get the next available device ID. Accessing port topology information is not exposed in libusb-0.1.

I am surprised that the hub did not change the bus ID. Does the hub show up in `lsusb` as well? If it has a switch for self-powered vs bus-powered, check that as well.

janipewter, 2020-04-10:
The output of ```lsusb -v``` shows the correct serial number for each device, but I can't see how to use this in NUT to specify which UPS to connect to

clepple, 2020-04-10:
https://networkupstools.org/docs/man/usbhid-ups.html and search for serial

janipewter, 2020-04-10:
Thank you that has worked wonderfully. Next time I will RTFM.

closed 2020-04-10 by janipewter