networkupstools/nut #772: Update nut-monitor.service to use /run for PIDFile instead of /var/run

WarmChocolateCake, 2020-04-10:
I'm seeing warnings in journald that nut-monitor.service is referring to /var/run 

This looks to be a change in systemd (see [systemd/systemd@a2d1fb8](https://github.com/systemd/systemd/commit/a2d1fb882c4308bc10362d971f333c5031d60069))

Editing the unit file resolves this for me: checked on Arch Linux & Linux Mint (Debian)

Thanks

ravron, 2020-11-29:
This actually changed in https://github.com/systemd/systemd/commit/4a66b5c9bf9aabae8b8ed2720f5f21a4a2a21061, but yeah that's the gist. 

I believe "all" we'd have to do is update the definition of `PIDPATH` here: https://github.com/networkupstools/nut/blob/163b0e1708b4f0c28432ca213d8f0bb11e94825c/configure.ac#L116

Alternatively, whoever creates the package for relevant systems could provide `--with-pidpath` when building the package: https://github.com/networkupstools/nut/blob/163b0e1708b4f0c28432ca213d8f0bb11e94825c/configure.ac#L1166-L1180

jimklimov, 2020-11-30:
Since the multiplatform default for ages was to have /var/run, and still in one way or another (as directory, mountpoint or symlink) the path is resolvable on all or most of the common systems, I wouldn't want to change the default value in the codebase.

But indeed, packagers of distros who have the recent enough versions of systemd to complain, can and should update the `configure` script arguments in their packaging recipe - that's what they are for :)

WarmChocolateCake, 2020-12-01:
My UPS's have moved around since creating this issue - looking again at Arch, the package has changed repositories and appears that they've repackaged it to use the correct PIDPath as it's no longer reporting an error.

Mint (Debian) is now only on a laptop, so I'm not really bothered with that one ;)

So, thanks for looking into this - for me, I'm happy to close this.

Thanks for the software!

closed 2020-12-01 by WarmChocolateCake