networkupstools/nut #773: APC UPS is stale when system is resumed from standby

auanasgheps, 2020-04-28:
I am using an APC Back-UPS 700 connected to my server running Debian 10 and nut 2.7.4-8.
NUT is configured to use the usbhid-ups driver and is mostly fine until the system goes to standby. When it's resumed, this happens:

```
Apr 28 18:00:02 nas upsd[1009]: Data for UPS [ups] is stale - check driver
Apr 28 18:00:02 nas upsd[1009]: UPS [ups] data is no longer stale
Apr 28 18:00:04 nas upsmon[1012]: Poll UPS [ups] failed - Write error: Broken pipe
Apr 28 18:00:04 nas upsmon[1012]: Communications with UPS ups lost
Apr 28 18:00:04 nas upssched[31185]: Executing command: notify
Apr 28 18:00:04 nas upssched-cmd: Communications with the UPS ups are lost
Apr 28 18:00:12 nas upsd[1009]: User monmaster@127.0.0.1 logged into UPS [ups]
Apr 28 18:00:12 nas upsmon[1012]: Communications with UPS ups established
Apr 28 18:00:12 nas upssched[32005]: Executing command: notify
Apr 28 18:00:12 nas upssched-cmd: Communications with the UPS ups are established
```
The system is idle most of the time and the UPS has very low load. This communication issue happens **only** when the host is resumed from sleep mode.

I have tried to set:

- UPSD - `MAXAGE = 25`
- UPS - `pollinterval = 10`
- UPSMON - `POLLFREQ = 8`
- pollinterval and maxage values together.

None of these helped and the error always occurs.
Any change at best delays the second health check (like in the log above, done 8s later instead of 5) and not the initial that seems to be too aggressive.

clepple, 2020-06-22:
This is pretty much the definition of stale data, though. `upsd` is reporting that the last good value from the driver was from before standby, and it looks like your standby/resume scripts (or the DHCP lease timeouts, indirectly) are interrupting the network connection (`upsmon` "Broken pipe" error).

I would recommend adding something to your notify script to silently drop the error if the system has recently resumed.

auanasgheps, 2020-06-22:
Why are you referring to network connection? The UPS is directly connected to the machine using USB.
I don't think the network connection is being dropped though, since I can wake the system when in standby using WOL.

jimklimov, 2020-06-22:
But still, the data is stale? Like, "last received measurement was 5 hours
ago with polling frequency of 30 sec - so something is fishy"? Does this
happen once and then soon after system resuming, the data becomes regularly
collected again? If yes, this is normal I suppose.

clepple, 2020-06-22:
> Why are you referring to network connection? The UPS is directly connected to the machine using USB.

but this is *Network* UPS Tools, and `upsmon` connects to `upsd` over a TCP network socket (often localhost if the UPS is locally attached).

The stale data that Jim and I mentioned is between the driver and `upsd`. There are two problems in the log excerpt you sent.
 
> I don't think the network connection is being dropped though, since I can wake the system when in standby using WOL.

WOL does not necessarily imply that TCP network connections will persist across standby/resume. (WOL packets are UDP or raw frames.)

jimklimov, 2020-06-24:
> Yes, it always happens when the system has just resumed, then everything
> is fine.
> Allright, so is this by design or there's something I could do to avoid
> this unnecessary alert?

Yes, this is (and acts) per design.

I suppose if your suspend/resume routine is scripted, you can stop NUT
monitoring as part of going to sleep mode and restart it from scratch when
awakening - then any errors seen would be more relevant ("can't connect
right now").

auanasgheps, 2020-06-24:
Thanks for the hint. Yes, my suspend/resume routine is scripted.
What is the correct way of stopping NUT monitoring?

closed 2022-11-16 by auanasgheps

jimklimov, 2022-11-16:
Sorry this one fell through the cracks :\

Generally I think you would stop the NUT driver as part of suspend, then start it again.

If the problem is just about messages of stale/unstale, and monitoring resuming with no harmful impact, I'd say leave it be as the best-responsive and least complex approach to resume the system.