networkupstools/nut #776: Tripp Lite SMART1500RM2U disconnects and reconnects repeatedly

cweagans, 2020-05-02:
(coming from https://www.reddit.com/r/homelab/comments/g7htfu/ups_connection_to_server_constantly/ -- another user is having this same problem with the same hardware as well)

I'm using nut via a Synology NAS that's connected directly to my UPS with the included USB cable. The Synology UI is reporting a disconnect and reconnect every 10 seconds or so. As far as I can tell (in `lsusb`), it's not _actually_ disconnecting and reconnecting, but instead, it's starting `upsmon`, that runs for a couple of seconds, and then fails (at which point the "disconnect" message is emitted). I see that there is only limited support for this UPS on the device compatibility list.

Is there something that I can do to help improve support for this UPS? Happy to help debug, but I don't know enough about how all the different pieces fit together to do so autonomously. I will happily debug in real time with somebody over Zoom (or whatever) if that would be helpful to the project!

I have several questions out of the gate (but feel free to steer me in a different direction if these are off base):

1. How do I find out if I'm using the old or new version of this nas? The HCL says that there are two models and they have different levels of compatibility. I just purchased this NAS, so I'm inclined to believe that it's the newer one, but probably better to be sure.

2. How do I find out why `upsmon` is failing? If you're not familiar with Synology, you can safely assume that this is an ordinary Linux box and I'll translate paths and such as needed.

3. How do I find out what version of nut is even installed? Maybe this issue has been resolved already?

4. Is there some kind of built-in debugger that I can look at to see messages being passed back and forth between the UPS and the NAS to see what's failing?

cweagans, 2020-05-02:
Oh, I can actually confirm that it's using the usbhid-ups driver if that makes a difference. Also, here's my `ups.conf` minus all the commented stuff at the top:

```
pollinterval = 5

[ups]
	driver = usbhid-ups
	port = auto
	#pollonly
	#community = name
	#snmp_version = v2c
	#mibs = auto
	#secName = Synology
	#secLevel = noAuthNoPriv
	#authProtocol = MD5
	#authPassword = 111111111111
	#privProtocol = DES
	#privPassword = 222222222222
```

mbuonaccorsi, 2021-03-10:
Did you ever find a resolution to this?  I am running into the exact same problem with a Synology RS1221+ and a Tripp Lite UPS.

cweagans, 2021-03-11:
I haven't gotten anywhere with this issue, unfortunately. I'm not entirely sure how to debug it.

bval, 2021-05-12:
I have this same problem with a SMART1500LCDXL.

kabaga, 2023-07-25:
> I have this same problem with a SMART1500LCDXL.

I bought the same model and planning to return it if I don't see a solution. This issue has not been resolved since 2019.

jimklimov, 2023-07-25:
CC @aquette @dzomaya - any ideas here, please?

aquette, 2023-07-27:
Hey,
For the basic: have you tried another (better, shielded) cable and possibly root hub on the Synology ?
Then, debug traces (including low level, libusb & Linux kernel) would help understanding more, i.e. if it's a HW or SW issue, from the host or device.