networkupstools/nut #777: I am unable to start upsd server.

Edison4mobile, 2020-05-06:
My UPS product is Eaton 5PX 1500.
I am going to get UPS information using nut in ubuntu 16.0.4.
I installed and tried upsc. but it seems it needs upsd server to be started before I call upsc.
So I tried to run upsd using upsdrvctl.
But it doesn't started.
It seems it needs ups.conf.
It requires driver and port but I am not sure I could input driver and port.
[image]
I hope this issue will be resolved. Thanks so much.

clepple, 2020-05-21:
Your OS vendor should provide the `/var/state/ups` directory - what distribution are you using?

Also, to determine the correct driver, go to the [NUT Hardware Compatibility](https://networkupstools.org/stable-hcl.html) list and select the UPS vendor, e.g. https://networkupstools.org/stable-hcl.html?manufacturer=Eaton .

As noted in https://networkupstools.org/docs/man/usbhid-ups.html#_implementation , you can just specify `auto` as the port.

clepple, 2020-06-22:
You mentioned Ubuntu 16.04 but the NUT version strings do not look anything like those in Ubuntu.

Also, `/var/state/ups` should be created properly on Ubuntu.

(Please add new information as a separate comment, otherwise we won't be notified of any edits.)