networkupstools/nut #778: IBM / Lenovo UPS 5395

NQRott, 2020-05-07:
Hi there, any chance to get NUT working with this UPS over USB?
With the Lenovo UPM / IPP it works. It is the rebranded Eaton software.

Vendor ID = 04b3
Product ID= 0001

aquette, 2020-05-12:
hi @NQRott 
for sure!
Would you be able to test a git branch?
Do you need some direction to compile with the right flags too? In which case, please point the name of the system you're using.
If so, I can make something quickly. If you need more assistance, that may be harder.

AndrewLinden, 2020-09-24:
Hey @aquette 
Support for 04b3:0001 has been added already here: https://github.com/networkupstools/nut/commit/d22cbd9ed25558535ea64909ed3a5f43b5d8f8ee 

I have also an IBM branded Eaton OEM UPS. But my UPS has vendor id 04b3, but product id 0002
Please add product id 0002 to the driver!

If you need I can test and send a dump too.

AndrewLinden, 2020-10-08:
It turns out my UPS has also product id 0001.
The real problem was that on freebsd the nut package is version 2.7.4, which does not have support for the IBM UPS id numbers. Solution was to use nut-devel package, which has more fresh nut version from github and all works!

nbriggs, 2021-04-15:
@aquette -- I'm also trying to get NUT running for one of these, on Solaris 11.4 on SPARC -- but it (at commit 88773359d2958585471fb6d5ebe8af633ac4936e, Sat Apr 10 00:13:24 2021)  chokes on the HID report:

```
   0.751928     [D2] Checking device (04B3/0001) (/dev/usb/4b3.1/0)
   1.280539     [D2] - VendorID: 04b3
   1.280660     [D2] - ProductID: 0001
   1.280721     [D2] - Manufacturer: IBM
   1.280781     [D2] - Product: IBM 1000VA/750W Tower UPS
   1.280836     [D2] - Serial Number: 23A4881                     
   1.280892     [D2] - Bus: /dev/usb
   1.280948     [D2] - Device release number: 0100
   1.281044     [D2] Trying to match device
   1.281105     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   1.281166     [D3] match_function_regex: matching a device...
   1.281231     [D2] Device matches
   1.281315     [D3] nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   1.316791     [D3] HID descriptor, method 1: (9 bytes) => 09 21 10 01 21 01 22 61 06
   1.316904     [D3] HID descriptor length (method 1) 1633
   1.316966     [D3] HID descriptor, method 2: (9 bytes) => 09 21 10 01 21 01 22 61 06
   1.317024     [D3] HID descriptor length (method 2) 1633
   1.317077     [D2] HID descriptor length 1633
   4.146064     [D2] Report Descriptor size = 1633
   4.146190     [D3] Report Descriptor: (1633 bytes) => 05 84 09 04 a1 01 09 10 a1 00 09 12 a1
   4.146265     [D3]  00 09 13 85 20 75 08 95 01 15 00 26 ff 00 65 00 55 00 b1 03 06 ff ff 09 90
   4.146329     [D3]  85 21 b1 82 05 85 09 83 85 23 75 20 27 ff ff ff 7f 67 01 10 10 00 b1 83 05
   4.146391     [D3]  84 09 02 a1 02 09 60 85 27 75 08 26 ff 00 65 00 b1 83 09 60 81 83 05 85 09
   4.146453     [D3]  41 b1 83 09 41 81 83 06 ff ff 09 74 b1 83 09 74 81 83 c0 05 84 09 58 85 24
   4.146512     [D3]  b1 82 06 ff ff 09 45 85 25 75 20 27 ff ff ff 7f 66 01 10 b1 82 c0 05 84 09
   4.146589     [D3]  11 85 20 75 08 26 ff 00 65 00 b1 03 09 14 a1 00 09 15 b1 03 06 ff ff 09 6b
   4.146659     [D3]  85 26 b1 83 09 94 b1 83 05 84 09 02 a1 02 06 ff ff 09 79 85 27 b1 83 09 79
   4.146719     [D3]  81 83 05 84 09 62 b1 83 09 62 81 83 09 6d b1 83 09 6d 81 83 c0 c0 c0 06 ff
   4.146777     [D3]  ff 09 e0 85 ff 95 ff b1 82 09 e0 85 fe 95 0a b1 82 05 84 09 1e a1 81 09 42
   4.146835     [D3]  85 14 95 01 66 01 f0 b1 83 09 40 85 15 67 21 d1 f0 00 55 07 b1 83 09 1f 85
   4.146902     [D3]  0b 65 00 55 00 b1 03 c0 09 1e a1 84 09 44 85 74 75 10 27 ff ff 00 00 66 21
   4.146967     [D3]  d1 55 07 b1 83 09 43 66 21 d1 b1 83 09 42 85 36 75 08 26 ff 00 66 01 f0 55
   4.147037     [D3]  00 b1 82 09 45 85 16 65 00 b1 83 09 40 85 72 67 21 d1 f0 00 55 07 b1 82 09
   4.147100     [D3]  1f 85 0b 65 00 55 00 b1 03 c0 09 18 a1 00 09 20 a1 82 06 ff ff 09 43 85 81
   4.147170     [D3]  b1 83 05 84 09 57 85 8d 75 20 15 ff 27 ff ff ff 7f 66 01 10 b1 82 09 56 85
   4.147230     [D3]  90 b1 82 09 1f 85 80 75 08 15 00 26 ff 00 65 00 b1 03 09 21 b1 03 09 02 a1
   4.147289     [D3]  02 09 6c 85 81 b1 03 09 6b 85 82 b1 83 09 6b 81 83 c0 05 85 09 29 85 84 65
   4.147347     [D3]  00 b1 82 06 ff ff 09 5d 85 87 75 20 27 ff ff ff 7f 66 01 10 b1 82 09 5e 85
   4.147405     [D3]  8a b1 82 c0 05 84 09 20 a1 83 06 ff ff 09 43 85 81 75 08 26 ff 00 65 00 b1
   4.147463     [D3]  83 05 84 09 57 85 8e 75 20 15 ff 27 ff ff ff 7f 66 01 10 b1 82 09 56 85 91
   4.147521     [D3]  b1 82 09 1f 85 80 75 08 15 00 26 ff 00 65 00 b1 03 09 21 b1 03 09 02 a1 02
   4.147579     [D3]  09 6c 85 81 b1 03 09 6b 85 82 b1 83 09 6b 81 83 c0 05 85 09 29 85 85 65 00
   4.147637     [D3]  b1 82 06 ff ff 09 5d 85 88 75 20 27 ff ff ff 7f 66 01 10 b1 82 09 5e 85 8b
   4.147696     [D3]  b1 82 c0 05 84 09 19 85 80 75 08 26 ff 00 65 00 b1 03 c0 09 16 a1 00 06 ff
   4.147755     [D3]  ff 09 41 85 4a b1 03 05 84 09 1a a1 81 06 ff ff 09 43 85 73 b1 83 05 84 09
   4.147815     [D3]  1f 85 30 b1 03 09 32 85 31 75 10 27 ff ff 00 00 66 01 f0 55 0f b1 83 09 1b
   4.147873     [D3]  85 30 75 08 26 ff 00 65 00 55 00 b1 03 09 02 a1 02 09 6e 85 2d b1 83 09 6e
   4.147931     [D3]  81 83 09 6f b1 83 09 6f 81 83 09 64 85 32 b1 83 09 64 81 83 09 62 b1 83 09
   4.147990     [D3]  62 81 83 09 6d b1 83 09 6d 81 83 09 63 b1 83 09 63 81 83 06 ff ff 09 74 b1
   4.148048     [D3]  83 09 74 81 83 09 75 b1 83 09 75 81 83 09 78 b1 83 09 78 81 83 c0 05 84 09
   4.148105     [D3]  30 85 31 75 10 27 ff ff 00 00 67 21 d1 f0 00 55 07 b1 83 06 ff ff 09 71 85
   4.148164     [D3]  3d 75 08 26 ff 00 65 00 55 00 b1 82 c0 05 84 09 1a a1 83 09 1f 85 30 b1 03
   4.148222     [D3]  09 1b b1 03 09 02 a1 02 09 6d 85 3b b1 83 09 6d 81 83 c0 c0 06 ff ff 09 14
   4.148281     [D3]  a1 00 09 15 85 40 b1 03 05 84 09 02 a1 02 09 6d 85 41 b1 83 09 6d 81 83 06
   4.148339     [D3]  ff ff 09 74 85 3e b1 83 09 74 81 83 09 75 b1 83 09 75 81 83 c0 c0 05 84 09
   4.148398     [D3]  1c a1 00 09 34 85 42 75 10 27 ff ff 00 00 66 21 d1 55 07 b1 83 09 33 66 21
   4.148456     [D3]  d1 b1 83 09 31 67 01 00 10 00 55 0f b1 83 09 1f 85 40 75 08 26 ff 00 65 00
   4.148514     [D3]  55 00 b1 03 06 ff ff 09 44 85 43 b1 82 05 84 09 32 85 42 75 10 27 ff ff 00
   4.148572     [D3]  00 66 01 f0 55 0f b1 83 06 ff ff 09 53 85 4d 67 21 d1 f0 00 55 07 b1 82 05
   4.148639     [D3]  84 09 54 85 48 b1 82 06 ff ff 09 50 85 4e b1 82 05 84 09 53 85 47 b1 82 09
   4.148697     [D3]  1d 85 40 75 08 26 ff 00 65 00 55 00 b1 03 09 65 a1 81 09 02 a1 02 06 ff ff
   4.148760     [D3]  09 7e 85 49 b1 83 09 7e 81 83 c0 09 7d 85 53 65 00 b1 82 c0 05 84 09 65 a1
[... cut ...]
   4.149942     [D3]  b1 82 09 6c 81 82 c0 05 85 09 66 85 06 65 00 b1 83 09 66 81 83 09 29 85 08
   4.150000     [D3]  b1 83 06 ff ff 09 4d 85 12 b1 82 05 85 09 2a 85 17 75 10 27 ff ff 00 00 66
   4.150058     [D3]  01 10 b1 82 09 68 85 06 b1 83 09 68 81 83 06 ff ff 09 48 85 11 75 08 26 ff
   4.150116     [D3]  00 65 00 b1 82 09 97 85 29 75 20 27 ff ff ff 7f 66 01 10 b1 82 05 84 09 30
   4.150172     [D3]  85 07 75 10 27 ff ff 00 00 67 21 d1 f0 00 55 07 b1 83 c0 c0
   4.154023     [D1] Failed to parse report descriptor!: Error 0
```
Any suggestions?

nbriggs, 2021-04-15:
OK... tracked it down.  Problem is bogus conversion for little-endian to big-endian in the hidparser code.