networkupstools/nut #779: Driver does not start when using `dummy-ups` (repeater-mode)

flobernd, 2020-05-08:
Hi there,

I have defined 2 UPS in my config: One APC one and one Dummy (repeater-mode) like this:
```sh
maxstartdelay = 150

maxretry = 30

retrydelay = 5

# UPS device(s)
[apc]
    # Back-UPS RS 900G
    driver = usbhid-ups
    port = auto
    desc = "APC Back-UPS RS 900G"
    offdelay = 120
    ondelay = 240

[ups]
    # Pretend to be a Synology NAS, so other DiskStations will connect here
    driver = dummy-ups
    port = apc@localhost
    desc = "Synology UPS server"
```

As you see I already fiddled with the `retry` options without any success:
```
May 08 14:08:31 RASPI systemd[1]: Starting Network UPS Tools - power device driver controller...
May 08 14:08:31 RASPI upsdrvctl[580]: Using subdriver: APC HID 0.96
May 08 14:08:31 RASPI upsdrvctl[580]: Network UPS Tools - Generic HID driver 0.41 (2.7.4)
May 08 14:08:31 RASPI upsdrvctl[580]: USB communication driver 0.33
May 08 14:08:32 RASPI usbhid-ups[633]: Startup successful
May 08 14:08:32 RASPI upsdrvctl[580]: Error: Connection failure: Connection refused
May 08 14:08:32 RASPI upsdrvctl[580]: Network UPS Tools - Device simulation and repeater driver 0.14 (2.7.4)
May 08 14:08:32 RASPI upsdrvctl[580]: Driver failed to start (exit status=1)
May 08 14:08:37 RASPI upsdrvctl[580]: Error: Connection failure: Connection refused
May 08 14:08:37 RASPI upsdrvctl[580]: Network UPS Tools - Device simulation and repeater driver 0.14 (2.7.4)
May 08 14:08:37 RASPI upsdrvctl[580]: Driver failed to start (exit status=1)
May 08 14:08:42 RASPI upsdrvctl[580]: Error: Connection failure: Connection refused
May 08 14:08:42 RASPI upsdrvctl[580]: Network UPS Tools - Device simulation and repeater driver 0.14 (2.7.4)
May 08 14:08:42 RASPI upsdrvctl[580]: Driver failed to start (exit status=1)
...
```

The "real" driver loads fine but for some reason the dummy is unable to connect.

When manually executing `upsdrvctl start ups` after boot, everything works fine. I could put this command in my `/etc/rc.local` but I was wondering if there is a more clean solution for my problem?

Best regards

aquette, 2020-05-10:
hi @flobernd 
The systemd integration that we ship with current deb is not as good as the one in our git.
I'd guess that your upsd is not started.
To assert it, simply "systemctl status nut-server", you'll see that it's off.
The issue here is that dummy-ups in repeater mode is essentially equivalent to "upsc apc@localhost".
however, nut-server (upsd) has a Wants on nut-driver, which makes a race cond preventing upsd to start, since driver failed to start (the dummy repeater), because it needs upsd.
To workaround this:
* edit /lib/systemd/system/nut-server.service and comment the line "#Wants=nut-driver.service"
* edit /lib/systemd/system/nut-driver.service and add "nut-server" at the end of the Wants and After
* do a "systemctl daemon-reload"

and all should be fine

flobernd, 2020-05-10:
Hi there,

thanks for the detailed reply! 

In the meantime I worked around by just renaming the `apc` device to `ups` and not using the `dummy-ups` driver at all (got a Synology device which always expects the ups device to be named `ups`). I will definitely try your instructions anyways as I would like to have the devices properly named.

> The systemd integration that we ship with current deb is not as good as the one in our git.

I already noticed that some files on my local Raspberry Pi are different than the files in this repo. Most notably the `nutshutdown` script was missing support for `POWEROFF_WAIT`. Hope there are not too many other things missing 😄

closed 2020-05-15 by flobernd

jackc94, 2022-01-06:
> hi @flobernd The systemd integration that we ship with current deb is not as good as the one in our git. I'd guess that your upsd is not started. To assert it, simply "systemctl status nut-server", [...]

This fix doesn’t work for me… the nut-driver.service fails to start at all and therefore both drivers don’t start. Running `upsdrvctl start ups` gets things running again but this is a manual process. Is there anything else I can try?

jimklimov, 2022-01-07:
Does it *fail* to start, or does it *not try* to start (e.g. per logs)? in the latter case, see if `systemctl enable nut-driver` would define the dependency symlinks (maybe needs a `disable` first to be sure).

Current NUT master branch includes the nut-driver-enumerator* family of services and scripts, to automatically define service units and appropriate dependencies  for each device driver; that might help too...

jackc94, 2022-01-07:
@jimklimov Thanks for the speedy response. It does try to start but fails… here are the logs after a failed start at boot:
[image]
I then manually started it using `systemctl start nut-driver` which was successful. 

I would need some assistance with your other solution of NUT master branch which includes the nut-driver-enumerator… both install and config!

jimklimov, 2022-01-07:
That may be a chicken-and-egg issue, corroborated by the single unit for all nut-driver's in that packaging: first the real UPS driver must start, for upsd (nut-server) to run as it depends on drivers. Then the dummy can start as it needs to talk to the upsd as the data source (and report back to it as the other NUT driver).

jackc94, 2022-01-08:
The workaround sounds like an option and makes complete sense. Any idea when NUT 2.7.5 might be released?

jimklimov, 2022-01-08:
It depends on time needed to brush up some things to push through CI, so I can only say "hopefully soon" since most of the goals for it were achieved or are close to that...

jackc94, 2022-01-08:
Great… I may just wait until then otherwise I’ll have to unpick the hack before upgrading. Thanks so much for your responses, so helpful and NUT is a great tool.

jackc94, 2022-01-09:
@jimklimov I just tried your workaround and got the following error
`/lib/systemd/system/nut-driver-qnapups.service:4: Failed to add dependency on nut-server, ignoring: Unknown error -22`

jimklimov, 2022-01-10:
Did you spell it fully - as nut-server.service (there are many other unit
types), and also, does your distro package deliver that?

jackc94, 2022-01-10:
I tried that after I left my comment which then meant that it added the dependency. The next issue is that it appears it’ll only allow me to have one of the services running at a time so starting `nut-driver-qnapups.service` (the dummy) then wipes out the ‘nut-driver.service’ and we’re back to square one again.

jimklimov, 2022-01-10:
That sounds weird. I can theorize there may be hard dependencies spelled in
the original (and copied) nut-driver.service, e.g.
RequiredBy=nut-server.service (or others like BindsTo or PartOf for
different nuances of depending), so starting either one somehow re-starts
nut-server and so stops the other, but that's just a guess at the moment.
Maybe some of those dependencies can be stripped away from the copy or
reduced (RequiredBy => WantedBy since dummy "knows" the server is running
as it depends on it); disable/enable the unit after that to apply.

One thing my suggestions earlier could have missed was to also run
`systemctl daemon-reload` after every edit, so the run-time systemd daemon
learns about the changes.

For debugging, `journalctl -xl` IIRC was useful, though I never got systemd
to bluntly state, "event 1 happened so I react with event 2" when
troubleshooting who stops whom, etc.

jackc94, 2022-01-10:
So what should I strip out of the config above for `nut-driver-qnapups.service` (the dummy) before I disable then enable?

I always run `systemctl daemon-reload` once I've made changes to any systemd services... thanks for checking though.

jimklimov, 2022-01-11:
Got a theory that without an 
````
[Install]
WantedBy=nut-server.service
````

(or similar) section in nut-driver-qnapups.service (i assume the opposite Wants=nut-driver.service is in nut-server.service), the StopWhenUnneeded kicks in and calls `upsdrvctl stop` that lacks a specific argument so indiscretionately stops all drivers.

Can you test if that happens and the `upsdrvctl stop CyberPower` etc would fare better?

jackc94, 2022-01-11:
How do I see what commands are sent… apologies, this level of debugging is new to me.

jackc94, 2022-02-02:
@jimklimov Is NUT 2.7.5 any closer to release?

jimklimov, 2022-02-04:
It is, a few PRs away now hopefully. Unfortunately got sick and that
delayed things.

jackc94, 2022-02-05:
Great news… keeping an eye out for it! 

Sorry to hear you’ve not been well… nothing serious I hope.

jimklimov, 2022-02-15:
FWIW, while testing other PRs, ran a spin-off of current NUT master branch on a clean debian system, built with `./ci_build.sh` and `sudo make install` afterwards (so most of it landed into /usr/local/ups/... tree by default - PATH in my root profile amended accordingly, but the systemd units went to /lib/systemd/... locations).

Then as root I enabled the NDE path unit (causes namesake service to start and create (or forget) `nut-driver@somename.service` instances whenever ups.conf is changed):
````
# systemctl enable nut-driver-enumerator.path
# systemctl start nut-driver-enumerator.path
````
...and the general umbrella target for driver+server+monitor setup:
````
# systemctl enable nut.target
# systemctl start nut.target
````

And "edited" `ups.conf` by scanning my USB ups into it:
````
# nut-scanner -U > /usr/local/ups/etc/ups.conf 
````

And the device driver service instance appeared (NDE path unit started and stopped that namesake service unit); the latter had this to log:
````
# journalctl -u nut-driver-enumerator.service

-- Logs begin at Tue 2022-02-15 09:45:24 UTC, end at Tue 2022-02-15 12:15:59 UTC. --
Feb 15 12:04:48 mirabox systemd[1]: Starting Network UPS Tools - enumeration of configure-file devices into systemd unit instances...
Feb 15 12:04:48 mirabox nut-driver-enumerator[19124]: Error reading the list of systemd service instances for UPS drivers, or none are defined - before manipulations
Feb 15 12:04:48 mirabox nut-driver-enumerator[19124]: Tue Feb 15 12:04:48 UTC 2022 : Detected changes in global section of '/usr/local/ups/etc/ups.conf', will restart all drivers
Feb 15 12:04:48 mirabox nut-driver-enumerator[19124]: OK
Feb 15 12:04:48 mirabox nut-driver-enumerator[19124]: Adding new systemd service instance for power device [nutdev1]...
Feb 15 12:04:48 mirabox nut-driver-enumerator[19124]: Created symlink from /etc/systemd/system/nut-driver.target.wants/nut-driver@nutdev1.service to /lib/systemd/system/nut-driver@.service.
Feb 15 12:04:49 mirabox nut-driver-enumerator[19124]: Enabled instance: 'nut-driver@nutdev1' for NUT configuration section 'nutdev1'
Feb 15 12:04:49 mirabox nut-driver-enumerator[19124]: Adding 'Wants'+After dependency for 'nutdev1' on 'systemd-udev.service systemd-udev-settle.service'...
Feb 15 12:04:49 mirabox nut-driver-enumerator[19124]: OK
Feb 15 12:04:49 mirabox nut-driver-enumerator[19124]: OK
Feb 15 12:04:49 mirabox nut-driver-enumerator[19124]: Started instance: 'nut-driver@nutdev1' for NUT configuration section 'nutdev1'
Feb 15 12:04:50 mirabox nut-driver-enumerator[19124]: === The currently defined service instances are:
Feb 15 12:04:50 mirabox nut-driver-enumerator[19124]: nutdev1
Feb 15 12:04:50 mirabox nut-driver-enumerator[19124]: === The currently defined configurations in '/usr/local/ups/etc/ups.conf' are:
Feb 15 12:04:50 mirabox nut-driver-enumerator[19124]: nutdev1
Feb 15 12:04:50 mirabox nut-driver-enumerator[19124]: Tue Feb 15 12:04:50 UTC 2022 : OK: No more changes to reconcile between systemd service instances and device configurations in '/usr/local/ups/etc/ups.conf'
Feb 15 12:04:50 mirabox systemd[1]: Started Network UPS Tools - enumeration of configure-file devices into systemd unit instances.
````

Units that appear in view can be grepped by "nut", e.g.:
````
# systemctl -a | grep nut
  nut-driver-enumerator.path                                                                                                          loaded    active   waiting   nut-driver-enumerator.path
  nut-driver-enumerator.service                                                                                                       loaded    inactive dead      Network UPS Tools - enumeration of configure-file devices into systemd unit instances
  nut-driver@nutdev1.service                                                                                                          loaded    active   running   Network UPS Tools - device driver for NUT device 'nutdev1'
  nut-monitor.service                                                                                                                 loaded    active   running   Network UPS Tools - power device monitor and shutdown controller
  nut-server.service                                                                                                                  loaded    active   running   Network UPS Tools - power devices information server
  system-nut\x2ddriver.slice                                                                                                          loaded    active   active    system-nut\x2ddriver.slice
  nut-driver.target                                                                                                                   loaded    active   active    Network UPS Tools - target for power device drivers on this system
  nut.target                                                                                                                          loaded    active   active    Network UPS Tools - target for power device drivers, data server and monitoring client (if enabled) on this system
````

So for users with Linux systemd (and Solaris/illumos SMF) the automagic should be mostly there.

jackc94, 2022-02-16:
Hey @jimklimov 

I must admit you’ve lost me a little there but it sounds like things are moving on very well. Will there be many steps to make this work when it is released or just an update and it’s taken care of?

jimklimov, 2022-02-16:
"Define 'update'" :)

Regarding update for people who are installing by building from source, at the very least existing configs should keep working (otherwise we have a bug), and new smartiness may need some setup/enablement like above.

Regarding update for users of distro-provided packages, technically it is same story, but the adjustments would hopefully be done by those OS packager teams.

jackc94, 2022-02-16:
Haha! Sorry… via distro-provided packages. 

So what setup would I be looking to do on a Raspberry Pi?

jimklimov, 2022-02-17:
Played around with the current state of NUT master, seems it needs a small bit of chiseling for such use-case with dummy-ups and the chicken-and-egg around it, which I'll "productize" shortly and explain below.

My test `ups.conf` was like this:
````
[nutdev1]
        driver = "usbhid-ups"
        port = "auto"
        # other args specific to the device

[echo]
        driver = dummy-ups
        port = nutdev1@localhost
````

With `nut-driver-enumerator.path` enabled and started, when I edited the `echo` section into the file, the `nut-driver@echo.service` unit became defined, and started well. The device got instantly seen in `upsc -l` listing and `upsc echo` queries.

However during reboot it failed to start (logically, because no upsd was running to talk to yet). systemd retried a few times and gave up ("start request repeated too quickly, refusing to start"), since it has a throttle to not swamp the system with vain attempts, and by default tries with a short delay. 

A working solution was to add `RestartSec=15s` (and `Restart=always` for good measure) to the `/lib/systemd/system/nut-driver@.service` file (so you can try that as a quick fix in your custom-built deployment).

The `nut-server` unit only `Wants` the `nut-driver.target`, and that only `Wants` each of the driver instances, so triggers them to start but does not abort if they don't. As a result, the slower pace of `nut-driver@echo` unit retries starts when `nut-server` for `upsd` is not running yet (because `nut-driver.target` is not ready), but with a later retry the dummy-ups driver has the server to connect to and so its unit starts well then.