networkupstools/nut #782: richcomm_usb - Digitus - Can't open USB device

dragon299, 2020-05-15:
I try to make a Digitus DN-170010 to work with nut using a serial to usb adapter and the richcomm_usb driver.

I know that DN-170010 is not in the compatibility list, but at least DN-170014 is, which is very similar.

ups.conf
```
[serverups]
    driver = richcomm_usb
    port = /dev/ttyUSB0
    desc = "Server UPS"
```

Output of upsdrvctl start:
```
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Richcomm dry-contact to USB driver 0.04 (2.7.4)
Warning: This is an experimental driver.
Some features may not function correctly.

Communications with UPS lost: Can't open USB device, retrying ...

```

Maybe someone would be so kind and give me a hint what I am doing wrong?

Thanks a lot.

clepple, 2020-12-31:
@dragon299 The `richcomm_usb` driver only has one USB VID:PID combination listed in the source code, which is for a Sweex 1000VA UPS.

However, if your UPS has a serial port, and you are using a generic USB-to-serial adapter that shows up as `/dev/ttyUSB0`, your kernel already knows how to do the USB-to-serial translation. Thus, you need a NUT driver that can talk to that virtual serial device. You should check out the [nutdrv_qx](https://networkupstools.org/docs/man/nutdrv_qx.html) driver, and counterintuitively, ignore everything about USB on that page. In theory, that might just mean changing to `driver = nutdrv_qx`, and maybe setting some more options.

clepple, 2020-12-31:
Another idea: what software does the vendor recommend for your model? That can be a useful keyword when searching.