networkupstools/nut #783: APC SmartUPS SMT series

flobernd, 2020-05-16:
Hi there, 

I did some research and planned to get me a proper APC SmartUPS to have all the advanced features like the power-off delay and power-on min capacity.

The SMT series catched my eyes but I’m not sure if it works with NUT. Reading some confusing statements regarding modbus protocol.

- Is modbus USB supported now or can I at least get all the features using the microlink USB?
- If not, am I able to use NUT with apcupsd driver to inform the clients while controlling the UPS with apcupsd?

Best regards

clepple, 2020-05-21:
Modbus support is being tracked in #503 

See https://networkupstools.org/docs/man/apcupsd-ups.html#_limitations - does that cover what you are trying to do?

clepple, 2020-05-21:
(fixed typo in link to #50 for ModBus - GitHub's autocomplete is kicking in)

flobernd, 2020-05-21:
Thanks for the response! I saw that specific driver and I already planned to use the native `apcupsd` driver to perform the shutdown and mirror the UPS state using NUT to inform the other clients.

Sadly Debian (or more specific Raspbian) does not seem to allow me to run both drivers at the same time. `apt get install apcupsd` forcefully removed `nut-client` and `nut-server` while installing one of the later packages forcefully removes `apcupsd`. :/

The UPS will be connected using USB. Would the USB-HID driver work for the basic stuff until the Modbus protocol makes it into NUT (I guess that will take some time as the issue exists for 7 years right now)?

clepple, 2020-05-22:
> Sadly Debian (or more specific Raspbian) does not seem to allow me to run both drivers at the same time.

Ah, right, you mentioned this in #779. I don't have that specific issue, but if I did, I would want to file a bug with Debian to remove the `Conflicts:` from both packages, and somehow turn it into a warning.

>  Would the USB-HID driver work for the basic stuff until the Modbus protocol makes it into NUT (I guess that will take some time as the issue exists for 7 years right now)?

I always cringe at these questions, since everyone has a different idea of "the basic stuff". I only bought a Modbus-based APC UPS about two years ago, and I have been logging data on it with `usbhid-ups` for most of that time. (I also moved and rebuilt the home network over that period, which partially explains why I haven't done any Modbus development yet.) The only thing I have noticed is that occasionally battery voltages are reported 10x higher than actual for about one polling cycle, and then it returns to normal. See the [DDL](https://networkupstools.org/ddl/APC/) for examples of what various APC UPSes return. (I realize it is not labeled well for answering your question, but look at the serial numbers and firmware versions to find similar units.)

flobernd, 2020-05-22:
Thanks for your response again! 

Alright, I guess I will just give it a try. The "basic functionality" which I need at the moment is just the detection of low-battery/critical-runtime and the execution of the delayed shutdown command.

closed 2020-05-22 by flobernd