networkupstools/nut #786: need to disable hiddev support ?

evilgreen, 2020-05-25:
I see that my UPS (Ippon UPS "Smart Power Pro II 1200", 0x06da:0xffff) is constantly reconnecting on my Linux server. That is not a power or cable issue, exactly the same combination was working for some time with usbhid-ups driver, then something probably changed on the host OS. It was advised [here](https://github.com/networkupstools/nut/issues/701#issuecomment-509117414) to use MGE driver, this is what I'm trying to do. I'm reading in the [installation notes](https://networkupstools.org/docs/man/usbhid-ups.html) that
> On Linux with MGE equipment, you will need at least a 2.4.25 or 2.6.2 kernel as well as libusb-0.1.8 or later to **disable hiddev support** and avoid conflict.

How that should be done? I suppose that could be related to my issue.

clepple, 2020-05-26:
The `usbhid-ups` driver automatically detaches (temporarily disables) the kernel `hiddev` driver from the UPS' primary interface, but the API for detaching drivers from userspace wasn't present before libusb-0.1.8 (or with kernels prior to the versions mentioned).

I don't know what causes constant reconnections, but this is most likely not related (unless the UPS firmware is brittle, and expects exactly the same timing that other HID implementations use).

evilgreen, 2020-05-26:
Thank you for the explanation. It looks like the reconnects could be related to the kernel - I do not see any on my old laptop with 4.4.0 kernel, but this is what I have on different hosts with newer kernels (4.14.71 and 4.14.66):
````[ 2199.188999] usb 1-1: new low-speed USB device number 2 using musb-hdrc
[ 2199.351510] usb 1-1: New USB device found, idVendor=06da, idProduct=ffff
[ 2199.351563] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[ 2199.351590] usb 1-1: Product: Offline UPS
[ 2199.351615] usb 1-1: Manufacturer: PPC
[ 2199.351641] usb 1-1: SerialNumber: 000000000   
[ 2199.412854] hid-generic 0003:06DA:FFFF.0001: hiddev0,hidraw0: USB HID v1.00 Device [PPC Offline UPS] on usb-musb-hdrc.1-1/input0
[ 2213.439938] usb 1-1: USB disconnect, device number 2
[ 2217.817451] usb 1-1: new low-speed USB device number 3 using musb-hdrc
[ 2217.980133] usb 1-1: New USB device found, idVendor=06da, idProduct=ffff
[ 2217.980185] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[ 2217.980213] usb 1-1: Product: Offline UPS
[ 2217.980238] usb 1-1: Manufacturer: PPC
[ 2217.980263] usb 1-1: SerialNumber: 000000000   
[ 2218.037867] hid-generic 0003:06DA:FFFF.0002: hiddev0,hidraw0: USB HID v1.00 Device [PPC Offline UPS] on usb-musb-hdrc.1-1/input0
[ 2232.059966] usb 1-1: USB disconnect, device number 3
[ 2236.441928] usb 1-1: new low-speed USB device number 4 using musb-hdrc
````

clepple, 2020-05-26:
Do the disconnects happen regardless of whether the NUT driver is running?

Greg Oliver suggested adding `usbcore.autosuspend=-1` on the kernel command line.

https://alioth-lists.debian.net/pipermail/nut-upsuser/2020-May/011870.html

evilgreen, 2020-05-26:
Yes, even without NUT driver running, the last example is from the host with `usbcore.autosuspend=-1 ` added.

evilgreen, 2020-05-27:
I guess the explanation from [this](https://www.linuxquestions.org/questions/slackware-14/usb-device-keeps-connecting-and-disconnecting-slackware-14-specific-4175471128/) forum post is correct:
> Apparently the firmware on these units is dodgy, and if you don't connect within 20 seconds, you get the infinite disconnect/reconnect loop. 

However it is not clear why there're no reconnects on the old laptop.
I now see that NUT driver is working and there is no _dmesg_ spam anymore:
```
systemd[1]: Starting Network UPS Tools - power device driver controller...
upsdrvctl[1166]: Using subdriver: MGE HID 1.44
upsdrvctl[1166]: Network UPS Tools - Generic HID driver 0.43 (2.7.4-747-g3d88a972)
upsdrvctl[1166]: USB communication driver 0.33
upsdrvctl[1166]: Network UPS Tools - UPS driver controller 2.7.4
usbhid-ups[1168]: Startup successful
systemd[1]: Started Network UPS Tools - power device driver controller.
````

closed 2020-05-27 by evilgreen