networkupstools/nut #787: TS Shara / UPS Senoidal 3200va

bpbastos, 2020-05-26:
Hello,

I am able to use this UPS using blazer_ser driver, but according to the docs TS Shara UPSs is supposed to use nutdrv_qx with sgs subdriver. Tried all sort of configurations but no success. At the moment I am using these configs:

[tsshara]
       driver = blazer_ser
       desc = "TS Shara"
       port = /dev/ttyACM0
       protocol = "megatec"

Any chance to get this UPS working?

Vendor ID = 0483
Product ID= 5740

Thanks!

favarete, 2020-09-07:
My UPS is almost the same one (2200va in my case) and I can't make it to work using either blazer_ser not nutdrc_qx

nanometer5088, 2023-03-04:
Same here - got a UPS TS Shara Senoidal 2200va and it isn't working on any driver, nor is it detected using `nut-scanner -U`

jimklimov, 2023-03-29:
FWIW, which NUT versions are you using? 2.7.4 (7 year old but still packaged by many distros), 2.8.0, or newer (e.g. custom master build)?

dantti, 2023-10-02:
I have one and it just works, one thing that http://mylowtechstuff.blogspot.com/2018/02/utilizando-nut-para-controle-de-nobreak.html did not mention was that I needed to start the nut server `sudo systemctl start nut-server`, after this both blazer_ser and nutdrv_qx worked.
BTW I recommend using /dev/serial/by-id/ instead of ttyACM which can have random numbers if some other device is attached.

pedrovanzella, 2024-04-12:
@dantti what's the firmware version on your UPS? Mine's `23.002.001` and it works with `blazer_ser` but not with `nutdrv_qx`.

bpbastos, 2024-06-18:
Just a little update, it seems like the firmware is the culprit.

# /lib/nut/nutdrv_qx  -DDDD -a tsshara
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D3] main_arg: var='driver' val='nutdrv_qx'
   0.000041     [D3] main_arg: var='desc' val='TS Shara'
   0.000048     [D3] main_arg: var='port' val='/dev/ttyACM1'
   0.000056     [D3] main_arg: var='protocol' val='megatec'
   0.000065     [D3] main_arg: var='runtimecal' val='2169,100,4339,50,8678,25,11108,20,17357,10'
   0.000077     [D3] main_arg: var='default.battery.voltage.high' val='27.00'
   0.000084     [D3] main_arg: var='default.battery.voltage.low' val='20.80'
   0.000095     [D1] debug level is '4'
   0.001125     [D1] upsdrv_initups...
   0.101687     [D2] Skipping protocol Voltronic 0.06
   0.101699     [D2] Skipping protocol Voltronic-QS 0.07
   0.101702     [D2] Skipping protocol Voltronic-QS-Hex 0.10
   0.101706     [D2] Skipping protocol Mustek 0.07
   0.101709     [D2] Skipping protocol Megatec/old 0.07
   0.101712     [D2] Skipping protocol BestUPS 0.06
   0.101715     [D2] Skipping protocol Mecer 0.07
   0.101786     [D3] send: 'Q1'
   0.310485     [D3] read: '(129.0 129.0 126.0 013 60.0 27.4 30.0 00001001'
   0.310630     [D3] send: 'I'
   0.482153     [D3] read: 'TS Shara        UPS Seno.  Saida 115V '
   0.482177     [D2] qx_process_answer: ups.firmware - invalid start character [54], expected [23]
   0.482258     [D3] send: 'Q1'
   0.691182     [D3] read: '(128.0 128.0 126.0 013 60.0 27.4 30.0 00001001'
   0.691315     [D3] send: 'I'
   0.862786     [D3] read: 'TS Shara        UPS Seno.  Saida 115V '
   0.862827     [D2] qx_process_answer: ups.firmware - invalid start character [54], expected [23]
   0.862907     [D3] send: 'Q1'
   1.071754     [D3] read: '(129.0 129.0 126.0 013 60.0 27.4 30.0 00001001'
   1.071895     [D3] send: 'I'
   1.243377     [D3] read: 'TS Shara        UPS Seno.  Saida 115V '
   1.243402     [D2] qx_process_answer: ups.firmware - invalid start character [54], expected [23]
   1.243411     [D2] Skipping protocol Zinto 0.06
   1.243415     [D2] Skipping protocol Masterguard 0.02
   1.243418     [D2] Skipping protocol Hunnox 0.01
   1.243422     [D2] Skipping protocol Ablerex 0.01
   1.243428     [D2] Skipping protocol Q1 0.07
   1.243433     Device not supported!
   1.243452     Device not supported!

jimklimov, 2024-06-18:
Interesting... FWIW, ASCII 0x23 is the hash `#` character and it sees 0x54 `T`  (from `TS Shara...`) in the reply.
Not sure OTOH which is correct, if the spec says anything about it. Commuting now.

bpbastos, 2024-06-18:
Is there anything I can do to help @jimklimov?

bpbastos, 2024-06-18:
Maybe it could be accents like "ãéïòû" that are commonly used in Brazilian Portuguese.

iskigow, 2024-09-19:
I'm using Ubuntu 24.04.1 LTS and my UPS is the same as @favarete's and I managed to get it working using both `blazer_ser` and `nutdrc_qx` drivers, even if the `nut-scanner` cannot detect this UPS. I read the guide that @dantti mentioned and another one (https://github.com/rbernardes/tsshara-ubuntu-server), as well as some other issues in the repository and driver's man (https://networkupstools.org/docs/man/blazer_ser.html, https://networkupstools.org/docs/man/nutdrv_qx.html). 

I installed NUT by `apt package` and its version is 2.8.1

For `blazer_ser` I use following configuration:
```INI
[tsshara]
  driver = blazer_ser
  desc = "TS Shara"
  protocol = "megatec"
  port = /dev/ttyACM0 # or /dev/serial/by-id/ like
```
For `nutdrc_qx`:
```INI
[tsshara]
  driver = nutdrv_qx
  desc = "TS Shara"
  protocol = megatec
  novendor
  port = /dev/ttyACM0 # or /dev/serial/by-id/ like
```
With the `novendor` config, the `nutdrv_qx` skips vendor discovery and continues monitoring. It is not the ideal solution, but it allows to use `nutdrv_qx` instead of `blazer_ser`.

upsc's output:
```CONSOLE
$ upsc tsshara
Init SSL without certificate database
battery.charge: 100
battery.voltage: 26.2
battery.voltage.high: 26.00
battery.voltage.low: 20.80
battery.voltage.nominal: 24.0
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.flag.novendor: enabled
driver.name: nutdrv_qx
driver.parameter.offdelay: 60
driver.parameter.ondelay: 0
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ttyACM0
driver.parameter.protocol: megatec
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.1
driver.version.data: Megatec 0.07
driver.version.internal: 0.36
driver.version.usb: libusb-1.0.27 (API: 0x100010a)
input.current.nominal: 100.0
input.frequency: 60.0
input.frequency.nominal: 60
input.voltage: 132.0
input.voltage.fault: 132.0
input.voltage.nominal: 115
output.voltage: 132.0
ups.beeper.status: enabled
ups.delay.shutdown: 60
ups.delay.start: 0
ups.load: 2
ups.status: OL
ups.temperature: 45.0
ups.type: offline / line interactive
``` 

[Off Topic] I haven't tested yet the shutdown and recovery behavior that was mentioned in the guide linked by @dantti, if it makes any difference depending on the driver.