networkupstools/nut #791: Unable to build nut on MacMini G4(PPC) running 10.5.8(Leopard)

abodnar, 2020-06-04:
Obviously this is a fairly old hardware and OS that I'm trying to build nut for. I am only wanting to setup the nut client so it can receive instructions from my Synology(which is on a UPS) on when to shutdown.

I am configuring it using the following command(not sure if there are different flags I should use):
`./configure --with-usb=no --with-serial=no --with-neon=no`

And I get the following error:
```
/bin/sh ../libtool  --tag=CC   --mode=link gcc -I../include  -DNETSNMP_ENABLE_IPV6 -g -Os -pipe -DBUILD=122 -DMACOSX_DEPLOYMENT_TARGET=10.5 -F/System/Library/PrivateFrameworks/ -F/System/Library/Frameworks/ -pipe -Udarwin9 -Ddarwin9=darwin9 -I/usr/local/include    -g -O2 -Wall -Wsign-compare   -o snmp-ups snmp-ups.o apc-mib.o baytech-mib.o compaq-mib.o eaton-mib.o ietf-mib.o mge-mib.o netvision-mib.o powerware-mib.o raritan-pdu-mib.o bestpower-mib.o cyberpower-mib.o delta_ups-mib.o xppc-mib.o huawei-mib.o eaton-ats-mib.o apc-ats-mib.o ../common/libcommon.la ../common/libparseconf.la main.o dstate.o -lnetsnmp 
libtool: link: gcc -I../include -DNETSNMP_ENABLE_IPV6 -g -Os -pipe -DBUILD=122 -DMACOSX_DEPLOYMENT_TARGET=10.5 -F/System/Library/PrivateFrameworks/ -F/System/Library/Frameworks/ -pipe -Udarwin9 -Ddarwin9=darwin9 -I/usr/local/include -g -O2 -Wall -Wsign-compare -o snmp-ups snmp-ups.o apc-mib.o baytech-mib.o compaq-mib.o eaton-mib.o ietf-mib.o mge-mib.o netvision-mib.o powerware-mib.o raritan-pdu-mib.o bestpower-mib.o cyberpower-mib.o delta_ups-mib.o xppc-mib.o huawei-mib.o eaton-ats-mib.o apc-ats-mib.o main.o dstate.o  ../common/.libs/libcommon.a ../common/.libs/libparseconf.a -lnetsnmp
depbase=`echo macosx-ups.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
	gcc -DHAVE_CONFIG_H -I. -I../include    -I../include  -DNETSNMP_ENABLE_IPV6 -g -Os -pipe -DBUILD=122 -DMACOSX_DEPLOYMENT_TARGET=10.5 -F/System/Library/PrivateFrameworks/ -F/System/Library/Frameworks/ -pipe -Udarwin9 -Ddarwin9=darwin9 -I/usr/local/include    -g -O2 -Wall -Wsign-compare -MT macosx-ups.o -MD -MP -MF $depbase.Tpo -c -o macosx-ups.o macosx-ups.c &&\
	mv -f $depbase.Tpo $depbase.Po
macosx-ups.c: In function 'upsdrv_initinfo':
macosx-ups.c:121: error: syntax error before 'kIOPSTypeKey'
macosx-ups.c:122: error: syntax error before 'kIOPSInternalBatteryType'
make[1]: *** [macosx-ups.o] Error 1
make: *** [all-recursive] Error 1
```

I've tried installing nut through MacPorts and Homebrew(tigerbrew) and both are encountering the same error.

Please let me know if there's more information I can provide.

clepple, 2020-06-22:
Sorry, I didn't see this issue until now. It doesn't sound like you need the `macosx-ups` driver (it would be for a locally connected UPS or laptop battery) but can you try the following patch?

```diff
diff --git a/drivers/macosx-ups.c b/drivers/macosx-ups.c
index 4873855a..73e7fbd5 100644
--- a/drivers/macosx-ups.c
+++ b/drivers/macosx-ups.c
@@ -22,6 +22,7 @@
 
 #include <regex.h>
 
+#include "CoreFoundation/CFString.h"
 #include "CoreFoundation/CoreFoundation.h"
 #include "IOKit/ps/IOPowerSources.h"
 #include "IOKit/ps/IOPSKeys.h"
```

Otherwise, you should be able to add `--without-macosx_ups` to the `./configure ...` parameters.

seanm, 2020-08-09:
Including `CoreFoundation/CFString.h` shouldn't help, since it's included by `CoreFoundation/CoreFoundation.h`.

The problem is that kIOPSTypeKey didn't exist in the 10.5 SDK, it was only added in the 10.6 SDK:

https://github.com/phracker/MacOSX-SDKs/blob/master/MacOSX10.5.sdk/System/Library/Frameworks/IOKit.framework/Versions/A/Headers/ps/IOPSKeys.h

vs

https://github.com/phracker/MacOSX-SDKs/blob/master/MacOSX10.6.sdk/System/Library/Frameworks/IOKit.framework/Versions/A/Headers/ps/IOPSKeys.h

clepple, 2020-08-09:
@seanm thanks for looking up the headers. The `CFString.h` idea was a bit of a shot in the dark, but I'm confused as to why it would show up as a syntax error, rather than some sort of unknown identifier error. Then again, I haven't looked at how `CFSTR()` is expanded. *shrug*

Do you have an older OS X system to test on? It doesn't sound like @abodnar needs the `macosx-ups` driver to connect to a Synology master, so I am tempted to just add `kIOPSTypeKey` to the list of things that autoconf checks for, and skip that driver if the key is missing from the SDK. Otherwise, I would want to see what keys are present pre-10.6 to make sure we don't introduce a logic error when fewer keys are present than expected.

seanm, 2020-08-09:
I don't have a 10.5 system these days, but it should work in VMWare should one be so inclined.

Personally, I wouldn't bother complicating the build script for a 12 year old OS for something the bug reporter doesn't even seem to need, as you say.  But if you were to do so, I'd recommend testing the SDK version (`MAC_OS_X_VERSION_MAX_ALLOWED > 1050').

abodnar, 2020-09-27:
Yeah, I wouldn't make any changes. I'm actually pretty close to retiring the system.

jimklimov, 2022-09-12:
FWIW, past years leading up to NUT v2.8.0 release (and some fixes after that) did address MacOS a few times, primarily to pass CI builds of environments available for free automated testing. Wondering if any of that work improved the situation described here?

Alternately, if it is no longer relevant to the OP (was system retired after all?) then please close the issue :)

abodnar, 2022-09-12:
I think I did get it to compile at some point. But no longer relevant and the system has been retired.

closed 2022-09-12 by abodnar