networkupstools/nut #794: The -K flag doesn't work if upsmon is running

VasiliPupkin256, 2020-06-12:
These lines 

https://github.com/networkupstools/nut/blob/3d88a972774f23f3567ed3696745794187e9fa59/clients/upsmon.c#L1951-L1952

should probably go before these

https://github.com/networkupstools/nut/blob/3d88a972774f23f3567ed3696745794187e9fa59/clients/upsmon.c#L1938-L1942

But also `check_pdflag` rely on config and `loadconfig` should be called before it. Please take a look.

gdt, 2023-01-07:
I ran into this too.  This should either be fixed, or the upsmon man page changed to document the restriction.

jimklimov, 2023-01-07:
Can you please check if current master (after #1762 fix for #1680 among others) still suffers any variant of this problem? Think I'm off the dev env for the weekend.

gdt, 2023-01-09:
Fixed in master.   I guess ticket doctrine is we close when fixed there?  (That's fine with me.)

I lack privs to hit close.

```
Network UPS Tools upsmon 2.8.0.1
Note: A previous upsmon instance is already running!
Usually it should not be running during OS shutdown,
which is when checking POWERDOWNFLAG makes most sense.
UPS: foo@localhost (primary) (power value 1)
Using power down flag file /etc/killpower
Power down flag is not set
```

closed 2023-01-09 by jimklimov