networkupstools/nut #798: usbhid-ups doesn't find Eaton on multi-ups USB system

gnudiff, 2020-06-17:
Environment:

- Ubuntu 18.04 LTS server
- Linux monitor 4.15.0-106-generic #107-Ubuntu SMP Thu Jun 4 11:27:52 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
- default NUT installed from apt repository: v 2.7.4-5.1ubuntu2

I have 2 UPSes connected to both via USB.

APC Smart UPS 1000 as `/dev/bus/usb/001/003` (symlinked by udev to /dev/usb/apc_smart_ups )
Eaton 5P 650iR as `/dev/bus/usb/001/002` (symlinked by udev to /dev/usb/eaton_ups )

The relevant lsusb:
```
Bus 001 Device 003: ID 051d:0002 American Power Conversion Uninterruptible Power Supply
Bus 001 Device 002: ID 0463:ffff MGE UPS Systems UPS
```

The appropriate entries in /etc/ups.conf are:

[eaton_cross]
driver = usbhid-ups
port = /dev/usb/eaton_ups
desc = "Eaton 5P 650iR @ cross"

[apc_cross]
driver = usbhid-ups
port = /dev/usb/apc_smart_ups
desc = "APC Smart UPS @ cross"

**However, NUT uses and shows APC UPS for both configs when starting!**
# upsdrvctl stop
Network UPS Tools - UPS driver controller 2.7.4
# upsdrvctl start eaton_cross
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
Using subdriver: APC HID 0.96
# upsc eaton_cross
Init SSL without certificate database
...
battery.mfr.date: 2015/00/05
device.mfr: American Power Conversion
device.model: Smart-UPS 1000
device.serial: AS0451130939
device.type: ups
driver.name: usbhid-ups
...
driver.parameter.port: /dev/usb/eaton_ups
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: APC HID 0.96
driver.version.internal: 0.41
...
ups.load: 10.4
ups.mfr: American Power Conversion
ups.mfr.date: 2004/12/13
ups.model: Smart-UPS 1000
ups.productid: 0002
ups.serial: AS0451130939
ups.status: OL
ups.test.result: No test initiated
ups.timer.reboot: -1
ups.timer.shutdown: -1
ups.timer.start: -1
ups.vendorid: 051d

# upsdrvctl start apc_cross
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
Using subdriver: APC HID 0.96
# upsc apc_cross
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
...
device.mfr: American Power Conversion
device.model: Smart-UPS 1000
device.serial: AS0451130939
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/usb/apc_smart_ups
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: APC HID 0.96
driver.version.internal: 0.41
...
ups.firmware: 600.3.I
ups.firmware.aux: 1.5
ups.load: 10.4
ups.mfr: American Power Conversion
ups.mfr.date: 2004/12/13
ups.model: Smart-UPS 1000
ups.productid: 0002
ups.serial: AS0451130939
ups.status: OL
ups.test.result: No test initiated
ups.timer.reboot: -1
ups.timer.shutdown: -1
ups.timer.start: -1
ups.vendorid: 051d

gnudiff, 2020-06-17:
It worked for Eaton, after I added specific vendor ID to eaton config: `vendorid = 0463`

However, it then tried to also register same device for the APC UPS (and failing obviously):
```
$ upsdrvctl start apc_cross
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
Using subdriver: MGE HID 1.39
libusb_get_report: error sending control message: Device or resource busy
Got disconnected by another driver: Device or resource busy
Can't initialize data from HID UPS
Driver failed to start (exit status=1)`
```

So I am leaving this as error, as I think NUT really shouldn't try to attach to already attached UPS, but realize that there are more than 1 entry in config file.

aquette, 2020-06-17:
Hi
Please use the serial number matching, as per the manage. This guarantee to match the right unit.

gnudiff, 2020-06-17:
> Hi
> Please use the serial number matching, as per the manage. This guarantee to match the right unit.

I can understand getting wrong UPS and having 2 UPSes mixed up. That would be normal. But I think error is getting both entries to attach to and start *same actual UPS*.

clepple, 2020-06-18:
> But I think error is getting both entries to attach to and start same actual UPS, especially as I have written the different device links.

As far as I know, it is not possible for NUT to tell the difference between the kernel claiming the device (as the kernel `usbhid` driver does automatically, which we have to undo in order to use libusb) and another NUT driver having the device open.

It is explicitly mentioned in the man page that the driver ignores `port`: https://github.com/networkupstools/nut/blob/master/docs/man/usbhid-ups.txt#L155

clepple, 2020-06-18:
closing as this is basically a duplicate of #763 and #771.

closed 2020-06-18 by clepple