networkupstools/nut #801: again usbhid-ups does not disable beeper with Eaton 3S550

GanFall, 2020-06-22:
Hi, I have problem to turn off beeper on my EATON 3S, under windows it is working well, but when use it on Linux with nut it does not work.

Is there any inteligent way how to fix it? :o)
Great thanks, MiP

Same issue as mentioned on website bellow:
https://www.gitmemory.com/issue/networkupstools/nut/706/504001849
https://github.com/networkupstools/nut/issues/9

Device: rpi4 + EATON 3S
OS:
Linux raspberrypi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linu
Nut package version:
root@raspberrypi:/home/pi# dpkg -s nut | grep -i version
Version: 2.7.4-8

jimklimov, 2020-12-17:
Is this with a packaged release 2.7.4 (sadly quite aged, working on that) - not with a recent Git revision?

clepple, 2020-12-28:
@GanFall It looks like Jim may have missed your answer about using a packaged v2.7.4. You would need the fix from https://github.com/networkupstools/nut/issues/9 which is only available in git at the moment. Instructions for compiling the driver from source are here: https://github.com/networkupstools/nut/wiki/Building-NUT-on-Debian,-Raspbian-and-Ubuntu