networkupstools/nut #803: Salicru SPS SOHO+ 500 VA supported by usbhid-ups

tiagofreire-pt, 2020-07-10:
This one: https://www.salicru.com/en-uk/sps-500-soho-.html

Connected by USB type-B port only, as this model don't have any serial port.

Logs:

```
# upsc salicru@localhost
Init SSL without certificate database
battery.charge: 65
battery.runtime: 3427
battery.type: PbAc
battery.voltage: 13
device.mfr: PPC
device.model: Offline UPS
device.serial: 000000000   
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: Phoenixtec/Liebert HID 0.3
driver.version.internal: 0.41
ups.load: 6
ups.mfr: PPC
ups.model: Offline UPS
ups.productid: ffff
ups.serial: 000000000   
ups.status: OL CHRG
ups.vendorid: 06da
```

```
# upsrw salicru@localhost

```

```
# upscmd -l salicru@localhost
Instant commands supported on UPS [salicru]:

```

Tutorial for this UPS with NUT, on Proxmox VE (Debian 10): https://github.com/tiagofreire-pt/Home_Assistant_Salicru_SOHO_UPS

clepple, 2020-07-11:
Thanks for the information. One note of caution: it doesn't look like we support a shutdown command for this UPS.

tiagofreire-pt, 2020-07-11:
More info:

```
# /lib/nut/usbhid-ups -DD -u root -x explore -x vendorid=06da -a salicru
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000     debug level is '2'
   0.002286     upsdrv_initups...
(...)
   0.119393     Checking device (06DA/FFFF) (001/006)
   0.124645     - VendorID: 06da
   0.124700     - ProductID: ffff
   0.124705     - Manufacturer: unknown
   0.124710     - Product: Offline UPS
   0.124715     - Serial Number: 000000000   
   0.124720     - Bus: 001
   0.124724     - Device release number: 0003
   0.124729     Trying to match device
   0.124774     Device matches
   0.125573     HID descriptor length 495
   0.143600     Report Descriptor size = 495
   0.143996     Using subdriver: EXPLORE HID 0.1
   0.144050     45 HID objects found
   0.144523     Path: UPS.BatterySystem.Battery.Test, Type: Feature, ReportID: 0x24, Offset: 0, Size: 8, Value: 6
   0.145147     Path: UPS.Flow.[4].ConfigApparentPower, Type: Feature, ReportID: 0x74, Offset: 0, Size: 16, Value: 2
   0.145599     Path: UPS.Flow.[4].ConfigFrequency, Type: Feature, ReportID: 0x36, Offset: 0, Size: 8, Value: 50
   0.146083     Path: UPS.Flow.[4].ConfigVoltage, Type: Feature, ReportID: 0x72, Offset: 0, Size: 8, Value: 230
   0.146541     Path: UPS.PowerConverter.ffff0041, Type: Feature, ReportID: 0x4a, Offset: 0, Size: 8, Value: 1
   0.147125     Path: UPS.PowerConverter.Input.[1].Voltage, Type: Feature, ReportID: 0x31, Offset: 0, Size: 16, Value: 231
   0.147600     Path: UPS.PowerConverter.Input.[1].PresentStatus.Boost, Type: Feature, ReportID: 0x2d, Offset: 0, Size: 8, Value: 0
   0.147629     Path: UPS.PowerConverter.Input.[1].PresentStatus.Boost, Type: Input, ReportID: 0x2d, Offset: 0, Size: 8, Value: 0
   0.147639     Path: UPS.PowerConverter.Input.[1].PresentStatus.Buck, Type: Feature, ReportID: 0x2d, Offset: 8, Size: 8, Value: 0
   0.147649     Path: UPS.PowerConverter.Input.[1].PresentStatus.Buck, Type: Input, ReportID: 0x2d, Offset: 8, Size: 8, Value: 0
   0.148125     Path: UPS.PowerConverter.Output.Frequency, Type: Feature, ReportID: 0x42, Offset: 0, Size: 16, Value: 50.3
   0.148156     Path: UPS.PowerConverter.Output.Voltage, Type: Feature, ReportID: 0x42, Offset: 16, Size: 16, Value: 231
   0.148565     Path: UPS.PowerSummary.AudibleAlarmControl, Type: Feature, ReportID: 0x13, Offset: 0, Size: 8, Value: 2
   0.149118     Path: UPS.PowerSummary.DelayBeforeShutdown, Type: Feature, ReportID: 0x09, Offset: 0, Size: 32, Value: 0
   0.149580     Path: UPS.PowerSummary.DelayBeforeStartup, Type: Feature, ReportID: 0x0a, Offset: 0, Size: 32, Value: 0
   0.150290     Path: UPS.PowerSummary.iDeviceChemistry, Type: Feature, ReportID: 0x10, Offset: 0, Size: 8, Value: 5
   0.150327     Path: UPS.PowerSummary.iManufacturer, Type: Feature, ReportID: 0x10, Offset: 8, Size: 8, Value: 1
   0.150337     Path: UPS.PowerSummary.ffff00f0, Type: Feature, ReportID: 0x10, Offset: 16, Size: 8, Value: 3
   0.150347     Path: UPS.PowerSummary.iProduct, Type: Feature, ReportID: 0x10, Offset: 24, Size: 8, Value: 2
   0.150399     Path: UPS.PowerSummary.iSerialNumber, Type: Feature, ReportID: 0x10, Offset: 32, Size: 8, Value: 4
   0.150890     Path: UPS.PowerSummary.ffff00e2, Type: Feature, ReportID: 0xe2, Offset: 0, Size: 8, Value: 4
   0.151352     Path: UPS.PowerSummary.PercentLoad, Type: Feature, ReportID: 0x07, Offset: 0, Size: 8, Value: 5
   0.151931     Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Feature, ReportID: 0x01, Offset: 0, Size: 8, Value: 1
   0.151975     Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Input, ReportID: 0x01, Offset: 0, Size: 8, Value: 1
   0.151986     Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x01, Offset: 8, Size: 8, Value: 0
   0.151997     Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Input, ReportID: 0x01, Offset: 8, Size: 8, Value: 0
   0.152008     Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x01, Offset: 16, Size: 8, Value: 1
   0.152018     Path: UPS.PowerSummary.PresentStatus.Charging, Type: Input, ReportID: 0x01, Offset: 16, Size: 8, Value: 1
   0.152028     Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Feature, ReportID: 0x01, Offset: 24, Size: 8, Value: 0
   0.152055     Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Input, ReportID: 0x01, Offset: 24, Size: 8, Value: 0
   0.152067     Path: UPS.PowerSummary.PresentStatus.Good, Type: Feature, ReportID: 0x01, Offset: 32, Size: 8, Value: 1
   0.152091     Path: UPS.PowerSummary.PresentStatus.Good, Type: Input, ReportID: 0x01, Offset: 32, Size: 8, Value: 1
   0.152101     Path: UPS.PowerSummary.PresentStatus.InternalFailure, Type: Feature, ReportID: 0x01, Offset: 40, Size: 8, Value: 0
   0.152111     Path: UPS.PowerSummary.PresentStatus.InternalFailure, Type: Input, ReportID: 0x01, Offset: 40, Size: 8, Value: 0
   0.152121     Path: UPS.PowerSummary.PresentStatus.Overload, Type: Feature, ReportID: 0x01, Offset: 48, Size: 8, Value: 0
   0.152131     Path: UPS.PowerSummary.PresentStatus.Overload, Type: Input, ReportID: 0x01, Offset: 48, Size: 8, Value: 0
   0.152785     Path: UPS.PowerSummary.DesignCapacity, Type: Feature, ReportID: 0x0b, Offset: 0, Size: 8, Value: 100
   0.152827     Path: UPS.PowerSummary.CapacityGranularity1, Type: Feature, ReportID: 0x0b, Offset: 8, Size: 8, Value: 1
[... cut ...]
  36.678054     Quick update...
  38.427720     upsdrv_updateinfo...
  38.678708     libusb_get_interrupt: Connection timed out
  38.678749     Got 0 HID objects...
  38.678762     Quick update...
  40.428451     upsdrv_updateinfo...
  40.679697     libusb_get_interrupt: Connection timed out
  40.679739     Got 0 HID objects...
  40.679748     Quick update...
  42.430301     upsdrv_updateinfo...
  42.681057     libusb_get_interrupt: Connection timed out
  42.681097     Got 0 HID objects...
  42.681105     Quick update...
```

clepple, 2020-07-12:
This might be addressed by #775.

tiagofreire-pt, 2020-07-13:
Additional info:

```
# /lib/nut/usbhid-ups -a salicru -DDD
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000     debug level is '3'
   0.002327     upsdrv_initups...
(...)
   0.135006     Checking device (06DA/FFFF) (001/043)
   0.138725     - VendorID: 06da
   0.138762     - ProductID: ffff
   0.138768     - Manufacturer: PPC
   0.138772     - Product: Offline UPS
   0.138776     - Serial Number: unknown
   0.138781     - Bus: 001
   0.138785     - Device release number: 0003
   0.138790     Trying to match device
   0.138801     Device matches
   0.138840     nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.139498     HID descriptor, method 1: (9 bytes) => 09 21 00 01 00 01 22 ef 01
   0.139529     HID descriptor length (method 1) 495
   0.139543     HID descriptor, method 2: (9 bytes) => 09 21 00 01 00 01 22 ef 01
   0.139551     HID descriptor length (method 2) 495
   0.139556     HID descriptor length 495
   0.157434     Report Descriptor size = 495
   0.157474     Report Descriptor: (495 bytes) => 05 84 09 04 a1 01 09 10 a1 00 09 12 a1 00
   0.157485      05 84 09 58 85 24 75 08 95 01 65 00 15 00 26 ff 00 55 00 b1 82 c0 c0 09 1e
   0.157501      a1 84 09 43 85 74 75 10 95 01 15 00 27 ff ff 00 00 66 21 d1 55 07 b1 83 09
   0.157514      42 85 36 75 08 26 ff 00 66 01 f0 55 00 b1 82 09 40 85 72 67 21 d1 f0 00 55
   0.157524      07 b1 82 c0 09 16 a1 00 06 ff ff 09 41 85 4a 65 00 55 00 b1 03 05 84 09 1a
   0.157533      a1 81 05 84 09 30 85 31 75 10 27 ff ff 00 00 67 21 d1 f0 00 55 07 b1 83 09
   0.157543      02 a1 02 09 6e 85 2d 75 08 26 ff 00 65 00 55 00 b1 83 09 6e 81 83 09 6f b1
   0.157552      83 09 6f 81 83 c0 c0 05 84 09 1c a1 00 05 84 09 32 85 42 75 10 27 ff ff 00
   0.157562      00 66 01 f0 55 0f b1 83 05 84 09 30 85 42 75 10 27 ff ff 00 00 67 21 d1 f0
   0.157571      00 55 07 b1 83 c0 c0 09 24 a1 00 09 5a 85 13 75 08 15 00 26 ff 00 65 00 55
   0.157582      00 b1 82 05 84 09 57 85 09 75 20 15 ff 27 ff ff ff 7f 66 01 10 b1 82 09 56
   0.157592      85 0a b1 82 05 85 09 89 85 10 75 08 15 00 26 ff 00 65 00 b1 03 05 84 09 fd
   0.157602      b1 03 06 ff ff 09 f0 b1 03 05 84 09 fe b1 03 05 84 09 ff b1 03 06 ff ff 09
   0.157619      e2 85 e2 b1 83 05 84 09 35 85 07 15 00 26 ff 00 65 00 b1 83 05 84 09 02 a1
   0.157630      02 05 85 09 d0 85 01 75 08 b1 83 09 d0 81 83 09 42 b1 83 09 42 81 83 09 44
   0.157639      b1 83 09 44 81 83 05 85 09 45 b1 83 09 45 81 83 05 84 09 61 b1 83 09 61 81
   0.157649      83 09 62 b1 83 09 62 81 83 05 84 09 65 b1 83 09 65 81 83 c0 05 85 85 0b 75
   0.157659      08 95 01 15 00 25 64 09 83 b1 03 09 8d b1 03 09 67 65 00 b1 03 09 2c 65 00
   0.157668      b1 03 09 66 85 06 65 00 b1 83 09 66 81 83 05 85 09 68 85 06 75 10 26 ff 7f
   0.157678      66 01 10 b1 83 09 68 81 83 05 84 09 30 85 07 75 10 26 ff 7f 67 21 d1 f0 00
   0.157684      55 06 b1 83 c0 c0
   0.157813     Using subdriver: Phoenixtec/Liebert HID 0.3
   0.157828     45 HID objects found
   0.158263     Report[get]: (2 bytes) => 24 06
   0.158366     Path: UPS.BatterySystem.Battery.Test, Type: Feature, ReportID: 0x24, Offset: 0, Size: 8, Value: 6
   0.158796     Report[get]: (3 bytes) => 74 02 00
   0.158819     Path: UPS.Flow.[4].ConfigApparentPower, Type: Feature, ReportID: 0x74, Offset: 0, Size: 16, Value: 2
   0.159220     Report[get]: (2 bytes) => 36 32
   0.159241     Path: UPS.Flow.[4].ConfigFrequency, Type: Feature, ReportID: 0x36, Offset: 0, Size: 8, Value: 50
   0.159700     Report[get]: (2 bytes) => 72 e6
   0.159729     Path: UPS.Flow.[4].ConfigVoltage, Type: Feature, ReportID: 0x72, Offset: 0, Size: 8, Value: 230
   0.160138     Report[get]: (2 bytes) => 4a 01
   0.160164     Path: UPS.PowerConverter.ffff0041, Type: Feature, ReportID: 0x4a, Offset: 0, Size: 8, Value: 1
   0.160720     Report[get]: (3 bytes) => 31 e4 00
   0.160762     Path: UPS.PowerConverter.Input.[1].Voltage, Type: Feature, ReportID: 0x31, Offset: 0, Size: 16, Value: 228
   0.161196     Report[get]: (3 bytes) => 2d 00 00
   0.161223     Path: UPS.PowerConverter.Input.[1].PresentStatus.Boost, Type: Feature, ReportID: 0x2d, Offset: 0, Size: 8, Value: 0
   0.161230     Report[buf]: (3 bytes) => 2d 00 00
   0.161238     Path: UPS.PowerConverter.Input.[1].PresentStatus.Boost, Type: Input, ReportID: 0x2d, Offset: 0, Size: 8, Value: 0
   0.161244     Report[buf]: (3 bytes) => 2d 00 00
   0.161252     Path: UPS.PowerConverter.Input.[1].PresentStatus.Buck, Type: Feature, ReportID: 0x2d, Offset: 8, Size: 8, Value: 0
   0.161259     Report[buf]: (3 bytes) => 2d 00 00
   0.161267     Path: UPS.PowerConverter.Input.[1].PresentStatus.Buck, Type: Input, ReportID: 0x2d, Offset: 8, Size: 8, Value: 0
   0.161723     Report[get]: (5 bytes) => 42 f7 01 e3 00
[... 157 lines cut ...]
  45.466364     upsdrv_updateinfo...
  45.716670     libusb_get_interrupt: Connection timed out
  45.716724     Got 0 HID objects...
  45.716733     Quick update...
  45.717312     Report[get]: (8 bytes) => 01 01 00 01 00 01 00 00
  45.717347     Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Feature, ReportID: 0x01, Offset: 0, Size: 8, Value: 1
  45.717357     Report[buf]: (8 bytes) => 01 01 00 01 00 01 00 00
  45.717364     Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x01, Offset: 8, Size: 8, Value: 0
  45.717372     Report[buf]: (8 bytes) => 01 01 00 01 00 01 00 00
[... cut ...]
  60.729508     Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x01, Offset: 8, Size: 8, Value: 0
  60.729517     Report[buf]: (8 bytes) => 01 01 00 01 00 01 00 00
  60.729523     Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x01, Offset: 16, Size: 8, Value: 1
  60.729531     Report[buf]: (8 bytes) => 01 01 00 01 00 01 00 00
  60.729537     Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Feature, ReportID: 0x01, Offset: 24, Size: 8, Value: 0
```

Arcnor, 2020-07-15:
I have the 2200 model and the output of those commands is different, starting with the first one:

```
Init SSL without certificate database
battery.charge: 28
battery.runtime: 804
battery.type: PbAc
device.mfr: PPC
device.model: Offline UPS HID UPS
device.serial: 000000000
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: MGE HID 1.40
driver.version.internal: 0.41
input.voltage: 236.0
outlet.1.status: on
output.frequency: 50.0
output.frequency.nominal: 50
output.voltage: 236.0
output.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 9
ups.mfr: PPC
ups.model: Offline UPS HID UPS
ups.power.nominal: 9
ups.productid: ffff
ups.serial: 000000000
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: 0
ups.timer.start: 0
ups.type: offline / line interactive
ups.vendorid: 06da
```

It looks like there is support for commands and other stuff, I can provide the full rundown of commands if there is value on doing so.

forast3r, 2020-10-06:
Looks like the same vendorid (06da) starts different subdrivers (MGE and Phoenixtec/Liebert). By the way the MGE-HID gives more sensor information.

How can we force an specific subdriver?

forast3r, 2020-10-07:
> I have the 2200 model and the output of those commands is different, starting with the first one:
> 
> ```
> Init SSL without certificate database
> battery.charge: 28
> battery.runtime: 804
> battery.type: PbAc
> device.mfr: PPC
> device.model: Offline UPS HID UPS
> device.serial: 000000000
> device.type: ups
> driver.name: usbhid-ups
> driver.parameter.pollfreq: 30
> driver.parameter.pollinterval: 2
> driver.parameter.port: auto
> driver.parameter.synchronous: no
> driver.version: 2.7.4
> driver.version.data: MGE HID 1.40
> driver.version.internal: 0.41
> input.voltage: 236.0
> outlet.1.status: on
> output.frequency: 50.0
> output.frequency.nominal: 50
> output.voltage: 236.0
> output.voltage.nominal: 230
> ups.beeper.status: enabled
> ups.delay.shutdown: 20
> ups.delay.start: 30
> ups.load: 9
> ups.mfr: PPC
> ups.model: Offline UPS HID UPS
> ups.power.nominal: 9
> ups.productid: ffff
> ups.serial: 000000000
> ups.status: OL
> ups.test.result: No test initiated
> ups.timer.shutdown: 0
> ups.timer.start: 0
> ups.type: offline / line interactive
> ups.vendorid: 06da
> ```
> 
> It looks like there is support for ups commands and other stuff, I can provide the full output rundown if there is value on doing so.

Could you please post your config for ups.conf ? Thanks in advance!

forast3r, 2020-10-07:
> I have the 2200 model and the output of those commands is different, starting with the first one:
>
> ```
> Init SSL without certificate database
> battery.charge: 28
> battery.runtime: 804
> battery.type: [...]

Could you post your ups.conf contents? Thanks in advance!

jimklimov, 2020-11-07:
Note: PR #775 referenced above, which may be relevant to this issue, was now merged.

forast3r, 2020-11-11:
> Note: PR #775 referenced above, which may be relevant to this issue, was now merged.

Will this PR evolve into a new stable version?

jimklimov, 2020-11-11:
Note: PR #775 <https://github.com/networkupstools/nut/pull/775> referenced
above, which may be relevant to this issue, was now merged.

> Will this PR evolve into a new stable version?

Not sure I got the question... That PR's change is now part of master
branch code, so people who build their NUT can benefit (or test and perhaps
fix) already.

Eventually we hope to tie up enough loose ends to cut a release from master
branch, for distros to pick up "officially". That's long overdue (

Jim

forast3r, 2020-11-11:
Thanks, understood! Will try to build NUT myself

forast3r, 2020-11-12:
I confirm that after compiling and installing the PR #775 adds support for almost every sensor in the Salicru SPS SOHO+

jimklimov, 2020-11-12:
Great, thanks for the confirmation!

jimklimov, 2021-12-01:
Heads-up: with PR #1199 merged into NUT master branch, there should now be a new subdriver for Salicru devices "out of the box". Can you please check if it also works for your model?

Arcnor, 2025-04-29:
Hi, I forgot I even commented on this issue 5 years ago :).

I've gotten a new Salicru SOHO+ 2200 model, and now I'm getting the same output as the original poster of this issue. However, I'm running NUT 2.8.0 this time (as part of TrueNAS) which is supposed to have a new subdriver according to that last comment.

Monitoring works, but things like input load are not working, although I don't know what the subdriver will provide. That said, the USB VID/PID is `06da:ffff` so I'm not sure how the subdriver will identify this model at all...

TrueNAS simply adds the following configuration, I'm not sure if there is anything else required to make the new subdriver work:

```
[ups]
        driver = usbhid-ups
        port = auto
        desc = ""
```

Thanks in advance.