networkupstools/nut #808: Please add support for Tripp-Lite 3024 Protocol UPS' Protocol docs attached

objecttothis, 2020-07-24:
I have an Tripp-Lite OMNIVSX1000D that uses protocol 3024.  Tripp-Lite has sent me the protocol documents (see attached).
[3024 Protocol.pdf](https://github.com/networkupstools/nut/files/4970293/3024.Protocol.pdf)

alexmerm, 2020-12-28:
If you manually specify this in your ups.conf file and use the usbhid-ups driver it seems to work fine (see below)
```
[netups]
        driver = usbhid-ups 
        port = auto
        vendorid = 09ae
        productid = 3024
```
I also had to add a line to /lib/udev/rules.d/62-nut-usbups.rules to allow my nut to access the devcice and restart
`ATTR{idVendor}=="09ae", ATTR{idProduct}=="3024", MODE="664", GROUP="nut"`

I looked around and am not sure how to integrate this into nut, but so far I haven't had any issues

scolby33, 2020-12-29:
This is the solution I came up with as well, but I was unable to get the full functionality working. In particular, have you managed to go through a simulation of a power failure (look under "Simulating Power Failures" [here](https://networkupstools.org/docs/man/upsmon.html)) and gotten the UPS to turn off the outlets and then turn them back on after a delay/after power is restored?

jimklimov, 2021-11-14:
Possibly related to:
* #962
* #955
* #963