networkupstools/nut #813: not that ecuret battery.charge will wall power is disconnected

eshchar, 2020-08-24:
when the power from the wall is disconnected and i am on the battery the percentage is displayed as so:
from 100 to 80 and then it jump to 30 and continue to drop (the UPS is NEW).
i now that the percentage is reported by the UPS.

but you can calculate the percentage According to the Voltage :
[image]

sow the calculation cod be (in my case i have a 4cells in line sow 4 * 12 = 48) :
if i use the picture info about battery then the 100% is 50.92V and the minimum recommended is 46.04V
then 50.92 - 46.04 = 4.88 => 4.88/100 = 0.0488 so every drop by 0.0488 volts is 1%.
i now that the the ups isn't that accurate so you can say that every 0.1V is 2%.

-so can you please give an option to choose my option (with a disclaimer) to show the percentage.
-and please add to the info displayed the battery Voltage.

jimklimov, 2020-08-24:
Hello, I edited the post to fix English the way I got the original phonetic writing (most people here are not native speakers, so better make an effort and write properly to be understood).

It would be curious to know the UPS model and driver involved, and if there are options to try changing like one of several protocols. For example with SNMP UPSes some devices offer different precision data and different sets of variables from different MIBs.

That said, generally NUT can not reliably guess such conversions once and for all for different vendors. Battery technologies differ, relation of voltage to energy that can be extracted is not linear originally and depends on microclimate around the device, etc. 

If vendors have installed devices and logic and protocol to report the charge - it should probably be trusted. If they report voltage and preferably battery tech (PbAc? LiPo? Etc? - or user might provide that in driver options otherwise) your suggestion can be a reasonable approximation of the charge. NUT should report the voltage, so if it is not in upsc output, the vendor HW or protocol or driver do not see/pass/give it.

As for "NEW" UPS - is it newly bought, or newly made? Are there date stickers? Shops should store PbAc batteries in certain conditions, recharge every half a year etc. but many skip the hassle so older batteries are not that good on arrival to customer. Maybe some complete charge-discharge cycles are needed to "train" the battery and the sensors in the UPS, often such need is documented.

jimklimov, 2020-08-26:
The screenshot is not that easy to search in, and I can't find a battery voltage in it.

It also seems curious that you have a relatively aged version of master (MGE HID 1.44 version  was defined a couple of years ago) but with libusb-1.0.23. There are a number of branches and PRs that aim to add NUT support for libusb newer than 0.1 original, but AFAIK none yet merged. I don't think it causes your issue, unless there is a loss of communication due to e.g. library changes and it takes long (50% of battery charge) to recover. 

That all said, the commit by @aquette which defines version 1.43 offers just an example of calculated approximation as a virtual field : https://github.com/networkupstools/nut/commit/b7f7043c8cc6f84d801e12d3db133c7cb1ae1463

jimklimov, 2020-08-26:
Also I think I remember my first UPSes (small cheap Powercoms probably) some 20 years ago to have completely different readings and remaining runtime estimates when they changed from online to onbattery state...

eshchar, 2020-08-26:
when i use the program(power manneger lite) that was provided with the UPS it show the voltage and quite accurate.
i am using NUT on an Unraid.

gdt, 2023-01-07:
Estimating State of Charge and hence runtime from voltage is extremely difficult.   The idea of 100% at 12.9V and 0% at 11.4V is only valid for a battery that has had no current into or out of it for 24h, obviously not the UPS case.    SOC/voltage relationship is highly dependent on both discharge current and battery health, and really needs cal from previous discharges.

FWIW, my Best LI660V (from 1995!) does a pretty good job of runtime estimation *after* being on battery a few minutes.  It reports 99min when on line power.

So this is not a bug in nut.   There might be actionable features in here, like reporting voltage on your UPS, or adding some voltage/load to SOC history and mapping from that.  IMHO those should be separate feature tickets.

jimklimov, 2023-01-07:
For cross-reference, #1027 addressed a similar issue in nutdrv_qx driver.

jimklimov, 2023-01-09:
> when i use the program(power manneger lite) that was provided with the UPS it show the voltage and quite accurate.

@eshchar : Can you ask the program or device vendor for protocol specs, or ideally to contribute a NUT driver so more people can buy and use their products? Possibly there are some extended data points that NUT subdriver does not query or parse. Maybe they implement different protocols with different data models in the device so USB-HID is not the only way to talk to it?

Screenshots in http://www.powerinspired.com/brochures/PowerManagerLite_USB(EN).pdf mention "richcomm" - which suggests a brand name that has a (somewhat sta[b]le) dedicated driver in NUT for some models; possibly it might fare better? (Worth trying a quick test)