networkupstools/nut #815: Wrong HighVoltageTransfer value on CyberPower PR1000ELCDRT2U

LucidityCrash, 2020-09-06:
Just got a CyberPower PR1000ELCDRT2U and trying to make NUT work, I noticed that the High Voltage Transfer Value is incorrect
Running : /lib/nut/usbhid-ups -a cyberpower1 -DDDD the relevant section shows : 
   0.618726     string_to_path: depth = 3
   0.618744     Report[buf]: (3 bytes) => 11 1a 01
   0.618760     Path: UPS.Input.HighVoltageTransfer, Type: Feature, ReportID: 0x11, Offset: 0, Size: 16, Value: 90

The Hex is correct 011a = 282 but the value reported shows as 90 

The Low Voltage transfer is correct : 

   0.618674     string_to_path: depth = 3
   0.618689     Report[buf]: (2 bytes) => 10 aa
   0.618705     Path: UPS.Input.LowVoltageTransfer, Type: Feature, ReportID: 0x10, Offset: 0, Size: 8, Value: 170

I've been looking through the code but I'm struggling to see where the value is set from the raw hex, could I get some pointers ?

clepple, 2020-09-07:
This is probably the same underlying issue as in #482, just with a different set of HID parameters left open on the HID parsing stack. Think of it as the HID equivalent of mismatched parentheses. If you post the value of the HID Report Descriptor, I can explain a little more of what is going on. I forget what debug level shows that descriptor, but it's the wall of hex at either 4 or 5 `-D`s.

`HIDGetDataValue()` calls `logical_to_physical()` on the raw hex value from the report. These are both in [libhid.c](https://github.com/networkupstools/nut/blob/v2.7.4/drivers/libhid.c#L371).

I don't think we currently have code to set high/low transfer on CyberPower units (#438) so this is basically a display issue at this point. So you can paper over the problem with an `override.*` directive. To solve the underlying HID descriptor problem, we need a way to replace HID Report Descriptors: #169

jimklimov, 2023-08-29:
Cheers, per clues here and there, this issue may be related to the fixes that landed into NUT v2.8.0 release (and addressed more in "master" branch afterwards). Do you have a chance to test any of those builds - whether NUT compatibility with the device has improved?

jimklimov, 2024-12-18:
PR #1245 is presumed to have fixed the broken CPS high voltage transfer reports discussed in this issue. Both high and low values were reported reasonable in #1512 and #2718 testing, though possibly on models too different to this issue?...

closed 2024-12-18 by jimklimov