networkupstools/nut #818: PowerWalker Basic VI 1000 SB supported by usbhid-ups

mrfrenzy, 2020-09-24:
This PowerWalker ups works fine with the usbhid-ups, it seems to be built by Cyber Power.

**Proposed HCL
"PowerWalker"	"ups"	"2"	"Basic VI 1000 SB"	"Cyber Power PR1500LCDRT2U"	"usbhid-ups"

**Output from lsusb:**
ID 0764:0601 Cyber Power System, Inc. PR1500LCDRT2U UPS

**Output from upsc:**
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: 1
battery.runtime: 3120
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 27.1
battery.voltage.nominal: 24
device.mfr: 1
device.model: 1000
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.transfer.high: 29
input.transfer.low: 16
input.voltage: 1.2
input.voltage.nominal: 230
output.voltage: 231.4
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 9
ups.mfr: 1
ups.model: 1000
ups.productid: 0601
ups.realpower.nominal: 600
ups.status: OL
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764

devZer0, 2021-05-28:
hello, what did you do to get this run?  

i have powerwalker vi 2200 rle which lists as 
Bus 004 Device 002: ID 0764:0601 Cyber Power System, Inc. PR1500LCDRT2U UPS
in lsusb

but - i don't get it work

mrfrenzy, 2021-05-29:
In the future after the nut developers add this in auto will work, for now you have to specify driver manually:

I have this in ups.conf:
```
[powerwalker]
        driver = usbhid-ups
        port = auto
        desc "Powerwalker basic ups"
```

upsd.users:
```
[upsmon]
        password = password1234
        upsmon master
```

upsmon.conf
`MONITOR powerwalker@localhost 1 upsmon password1234 master`

devZer0, 2021-05-30:
i'm getting this output. 

```
/usr/local/ups/bin/usbhid-ups -a powerwalker -D
Network UPS Tools - Generic HID driver 0.43 (2.7.4.1)
USB communication driver 0.33
   0.000000	[D1] debug level is '1'
   0.005437	[D1] upsdrv_initups (non-SHUT)...
   0.848183	Using subdriver: CyberPower HID 0.5
   0.848278	[D1] 73 HID objects found
   0.849286	[D1] Path: UPS.PowerSummary.iProduct, Type: Feature, ReportID: 0x01, Offset: 0, Size: 8, Value: 1
   0.851185	[D1] Path: UPS.PowerSummary.iSerialNumber, Type: Feature, ReportID: 0x02, Offset: 0, Size: 8, Value: 2
   0.853188	[D1] Path: UPS.PowerSummary.iOEMInformation, Type: Feature, ReportID: 0x03, Offset: 0, Size: 8, Value: 3
   0.855198	[D1] Path: UPS.PowerSummary.iDeviceChemistry, Type: Feature, ReportID: 0x04, Offset: 0, Size: 8, Value: 4
   0.857278	[D1] Path: UPS.PowerSummary.Rechargeable, Type: Feature, ReportID: 0x05, Offset: 0, Size: 8, Value: 1
   0.859295	[D1] Path: UPS.PowerSummary.CapacityMode, Type: Feature, ReportID: 0x06, Offset: 0, Size: 8, Value: 2
   0.861290	[D1] Path: UPS.PowerSummary.DesignCapacity, Type: Feature, ReportID: 0x07, Offset: 0, Size: 8, Value: 100
   0.861411	[D1] Path: UPS.PowerSummary.CapacityGranularity1, Type: Feature, ReportID: 0x07, Offset: 8, Size: 8, Value: 10
   0.861445	[D1] Path: UPS.PowerSummary.CapacityGranularity2, Type: Feature, ReportID: 0x07, Offset: 16, Size: 8, Value: 10
   0.861468	[D1] Path: UPS.PowerSummary.WarningCapacityLimit, Type: Feature, ReportID: 0x07, Offset: 24, Size: 8, Value: 20
   0.861499	[D1] Path: UPS.PowerSummary.FullChargeCapacity, Type: Feature, ReportID: 0x07, Offset: 32, Size: 8, Value: 100
   0.861522	[D1] Path: UPS.PowerSummary.RemainingCapacityLimit, Type: Feature, ReportID: 0x07, Offset: 40, Size: 8, Value: 0
   0.863408	[D1] Path: UPS.PowerSummary.RemainingCapacity, Type: Input, ReportID: 0x08, Offset: 0, Size: 8, Value: 100
   0.863506	[D1] Path: UPS.PowerSummary.RemainingCapacity, Type: Feature, ReportID: 0x08, Offset: 0, Size: 8, Value: 100
   0.863539	[D1] Path: UPS.PowerSummary.RunTimeToEmpty, Type: Input, ReportID: 0x08, Offset: 8, Size: 16, Value: 7200
   0.863564	[D1] Path: UPS.PowerSummary.RunTimeToEmpty, Type: Feature, ReportID: 0x08, Offset: 8, Size: 16, Value: 7200
   0.863595	[D1] Path: UPS.PowerSummary.RemainingTimeLimit, Type: Input, ReportID: 0x08, Offset: 24, Size: 16, Value: 300
   0.863617	[D1] Path: UPS.PowerSummary.RemainingTimeLimit, Type: Feature, ReportID: 0x08, Offset: 24, Size: 16, Value: 300
   1.219462	[D1] Path: UPS.PowerSummary.ConfigVoltage, Type: Feature, ReportID: 0x09, Offset: 0, Size: 16, Value: 24
   1.221319	[D1] Path: UPS.PowerSummary.Voltage, Type: Feature, ReportID: 0x0a, Offset: 0, Size: 16, Value: 25.9
   1.223182	[D1] Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Input, ReportID: 0x0b, Offset: 0, Size: 1, Value: 1
   1.223235	[D1] Path: UPS.PowerSummary.PresentStatus.Charging, Type: Input, ReportID: 0x0b, Offset: 1, Size: 1, Value: 1
   1.223256	[D1] Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Input, ReportID: 0x0b, Offset: 2, Size: 1, Value: 0
   1.223277	[D1] Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Input, ReportID: 0x0b, Offset: 3, Size: 1, Value: 0
   1.223305	[D1] Path: UPS.PowerSummary.PresentStatus.FullyCharged, Type: Input, ReportID: 0x0b, Offset: 4, Size: 1, Value: 1
   1.223325	[D1] Path: UPS.PowerSummary.PresentStatus.RemainingTimeLimitExpired, Type: Input, ReportID: 0x0b, Offset: 5, Size: 1, Value: 0
   1.223345	[D1] Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Feature, ReportID: 0x0b, Offset: 0, Size: 1, Value: 1
   1.223375	[D1] Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x0b, Offset: 1, Size: 1, Value: 1
   1.223395	[D1] Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Feature, ReportID: 0x0b, Offset: 2, Size: 1, Value: 0
   1.223414	[D1] Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x0b, Offset: 3, Size: 1, Value: 0
   1.223434	[D1] Path: UPS.PowerSummary.PresentStatus.FullyCharged, Type: Feature, ReportID: 0x0b, Offset: 4, Size: 1, Value: 1
   1.223453	[D1] Path: UPS.PowerSummary.PresentStatus.RemainingTimeLimitExpired, Type: Feature, ReportID: 0x0b, Offset: 5, Size: 1, Value: 0
   1.225173	[D1] Path: UPS.PowerSummary.AudibleAlarmControl, Type: Feature, ReportID: 0x0c, Offset: 0, Size: 8, Value: 2
   1.227175	[D1] Path: UPS.PowerSummary.iManufacturer, Type: Feature, ReportID: 0x0d, Offset: 0, Size: 8, Value: 3
   1.229176	[D1] Path: UPS.Input.ConfigVoltage, Type: Feature, ReportID: 0x0e, Offset: 0, Size: 16, Value: 230
   1.231178	[D1] Path: UPS.Input.Voltage, Type: Feature, ReportID: 0x0f, Offset: 0, Size: 16, Value: 225.6
   1.233179	[D1] Path: UPS.Input.LowVoltageTransfer, Type: Feature, ReportID: 0x10, Offset: 0, Size: 16, Value: 165
   1.235177	[D1] Path: UPS.Input.HighVoltageTransfer, Type: Feature, ReportID: 0x11, Offset: 0, Size: 16, Value: 290
   1.237320	[D1] Path: UPS.Input.Frequency, Type: Feature, ReportID: 0x1c, Offset: 0, Size: 16, Value: 50
   1.239172	[D1] Path: UPS.Output.Voltage, Type: Feature, ReportID: 0x12, Offset: 0, Size: 16, Value: 0
   1.241177	[D1] Path: UPS.Output.PercentLoad, Type: Feature, ReportID: 0x13, Offset: 0, Size: 8, Value: 0
   1.243241	[D1] Path: UPS.Output.Test, Type: Feature, ReportID: 0x14, Offset: 0, Size: 8, Value: 0
   1.245183	[D1] Path: UPS.Output.DelayBeforeShutdown, Type: Feature, ReportID: 0x15, Offset: 0, Size: 16, Value: -60
[... cut ...]
  10.877111	[D1] Quick update...
  10.878047	[D1] Can't retrieve Report 16: Value too large for defined data type
  10.880011	[D1] Can't retrieve Report 15: Value too large for defined data type
  10.882063	[D1] Can't retrieve Report 0b: Value too large for defined data type
  10.884014	[D1] Can't retrieve Report 0b: Value too large for defined data type
  10.886061	[D1] Can't retrieve Report 0b: Value too large for defined data type
  10.888015	[D1] Can't retrieve Report 0b: Value too large for defined data type

ant-thomas, 2021-11-09:
> **input.voltage: 1.2** //These are relative to input.voltage.nominal

Does this mean your input voltage is 230 + 1.2 = 231.2?

I've got the same UPS and want to calculate/track input voltage and it does look to be relative to something rather than exact.

mrfrenzy, 2021-11-10:
That is correct, you can see that if you add up **input.voltage.nominal + input.voltage** it is rougly equal to **output.voltage**

jimklimov, 2022-04-25:
@devZer0 : Hi, there were some issues with USB HID encoding fixed last year (starting with approx #1040 and #1024) - I wonder if "too large for defined type" is related to that. Can you re-check if the current NUT codebase handles your device better?

closed 2022-04-25 by jimklimov