networkupstools/nut #819: Status of 2.7.5 release

MicroTransactionsMatterToo, 2020-10-01:
Was just wondering what the status of 2.7.5's release is. Seems the main blocker is the move from libusb-0.1 to libusb-1.0, but I can't find much information on exactly whats going on. Anyone care to fill me in?

jimklimov, 2020-10-01:
We are also discussing this internally and it seems the consensus is to
merge one those branches (yet to be decided which) and keep the original
implementation 0.1 mode as safe default but expose the new 1.0 ones for
wider testing by community. Main problem is to choose which one of these to
propose, because the original authors had some reservations against each of
their works, I believe.

Not sure if it is easy to merge all of them (lots of ifdefs for configure
script options? Separately named source and binary files - might be better
and easier while we are unsure which is best?) in a better effort to get
each approach testable in the wild and pick the best default for subsequent
releases.

Also, moving forward needs time and all are currently short on that :(

So as ever, PRs are welcome! :) In this particular case, it is likely more
about arranging sources from several branches into one and extending build
recipes...

Jim

sgrayban, 2020-10-16:
Is this release ready yet? I am able to build the current master branch.

jimklimov, 2020-10-19:
Sadly, not quite; at least the libusb bump is not resolved yet.

There was a surge in getting code tested in more setups now available to
Travis and a private buildbot, and cleaned from compiler warnings that
arise (help would be welcome with PRs to handle the cases still red), and
several device driver supports were merged over this month or will likely
soon be. So value to be brought by new release is growing :)

Temtaime, 2020-12-03:
IMO it's better to release than to wait. Let the users decide to update or not.
There're tons of new upses which are not supported by latest release, but supported with current master.

jimklimov, 2020-12-03:
Thanks for the nudge, we are getting closer I hope.

jimklimov, 2021-09-21:
Vague, the future is. Always moving. Ever changing.

I do actually hope for a manageable timeframe, with more releases of finer-grained goals that were initially clumped as goals of 2.7.5; so now getting a warnings-clean codebase out (and half a decade of updates from contributors) that outlasted and survived 3 or more CI farm implementations, is a worthy milestone in its own :)

So the libusb-1.x support per #300 would wait until a next release, hopefully not that far behind 2.7.5 though.

Updating the docs for the release at #486 branch every once in a while to keep track of all the changes that are landing.

jimklimov, 2022-01-18:
UPDATE: Merge of libusb-1.0 support actually happened recently, because the branch was quite in practical use by many community members, and as the least-resistance path for another milestone in build-warnings resolution, so it would be part of 2.7.5 after all.

jimklimov, 2022-01-18:
Closing in favor of newer duplicate question.

closed 2022-01-18 by jimklimov