networkupstools/nut #834: Support for Salicru ADV 1500 RT2

pcamelo, 2020-10-18:
I've been trying to configure NUT with a Salicru UPS, Model ADV 2500 RT2. It came with ViewPower and seems that I get something working after giving 0666 permission to `/dev/bus/usb/001/002` [just for testing, I know this is not the right way to do it].

I get the following when I run `sudo ./usbhid-ups -DD -x explore -x vendorid=2e66 -a salicru`:

```$ sudo ./usbhid-ups -DD -x explore -x vendorid=2e66 -a salicru
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000	debug level is '2'
   0.001580	upsdrv_initups...
   0.133331	Checking device (1D6B/0003) (002/001)
   0.133553	- VendorID: 1d6b
   0.133577	- ProductID: 0003
   0.133613	- Manufacturer: unknown
   0.133650	- Product: unknown
   0.133676	- Serial Number: unknown
   0.133694	- Bus: 002
   0.133721	- Device release number: 0504
   0.133735	Trying to match device
   0.133758	Device does not match - skipping
   0.133905	Checking device (8087/0A2B) (001/003)
   0.133947	- VendorID: 8087
   0.133958	- ProductID: 0a2b
   0.133967	- Manufacturer: unknown
   0.133985	- Product: unknown
   0.134020	- Serial Number: unknown
   0.134038	- Bus: 001
   0.134058	- Device release number: 0010
   0.134086	Trying to match device
   0.134111	Device does not match - skipping
   0.134145	Checking device (2E66/0101) (001/002)
   0.135480	- VendorID: 2e66
   0.135497	- ProductID: 0101
   0.135506	- Manufacturer: Salicru
   0.135519	- Product: ADV1500-RT
   0.135543	- Serial Number: 0000000000000000
   0.135555	- Bus: 001
   0.135576	- Device release number: 0002
   0.135592	Trying to match device
   0.135630	Device matches
   0.135845	HID descriptor length 27
   0.136083	Report Descriptor size = 27
   0.136202	Using subdriver: EXPLORE HID 0.1
   0.136215	2 HID objects found
   0.136363	libusb_get_report: error sending control message: Operation not permitted
   0.136388	Can't retrieve Report 00: Permission denied
   0.136409	Path: ff000001.ff000002, Type: Input, ReportID: 0x00, Offset: 0, Size: 8
   0.136642	libusb_get_report: error sending control message: Operation not permitted
   0.136682	Can't retrieve Report 00: Permission denied
   0.136722	Path: ff000001.ff000002, Type: Output, ReportID: 0x00, Offset: 0, Size: 8
   0.136765	Report descriptor retrieved (Reportlen = 27)
   0.136778	Found HID device
   0.136797	Detected a UPS: Salicru/ADV1500-RT
   0.136823	find_nut_info: unknown info type: load.off.delay
   0.136843	find_nut_info: unknown info type: load.on.delay
   0.136863	find_nut_info: unknown info type: load.off.delay
   0.136896	upsdrv_initinfo...
   0.136925	upsdrv_updateinfo...
   0.387520	libusb_get_interrupt: Connection timed out
   0.387560	Got 0 HID objects...
   0.387579	Quick update...
   0.387682	dstate_init: sock /run/nut/usbhid-ups-salicru open on fd 5
   0.387716	upsdrv_updateinfo...
   0.638247	libusb_get_interrupt: Connection timed out
   0.638304	Got 0 HID objects...
   0.638317	Quick update...
   2.388657	upsdrv_updateinfo...
   2.639625	libusb_get_interrupt: Connection timed out
   2.639669	Got 0 HID objects...
   2.639683	Quick update...
   4.390641	upsdrv_updateinfo...
   4.641499	libusb_get_interrupt: Connection timed out
   4.641543	Got 0 HID objects...
   4.641556	Quick update...
   6.392629	upsdrv_updateinfo...```

clepple, 2020-12-31:
`Report Descriptor size = 27` is too short to contain useful information for `usbhid-ups`, so this UPS is likely using HID as a replacement for a proper USB-to-serial protocol.

Issue #554 and the [nutdrv_qx](https://networkupstools.org/docs/man/nutdrv_qx.html) man page indicate that other Salicru models that use ViewPower should work with `nutdrv_qx`, so that might be worth a try. I have only used `nutdrv_qx` with hardware serial ports, not USB, so hopefully someone else (@zykh?) has additional information.

jimklimov, 2021-12-01:
Heads-up: with PR #1199 merged into NUT master branch, there should now be a new subdriver for Salicru devices "out of the box". Can you please check if it also works for your model?

pedromrg, 2023-08-09:
Hi there, bumping this up... My ADV 2000 RT2 is not being detected for some reason, using latest version of usbhid-ups driver. :/

jimklimov, 2023-08-10:
@pedromrg : Can you please elaborate? There are many definitions of "latest version", alas.
* Custom build of current master (or some feature branch) from Git
* Modern distro package that delivers 2.8.0 or newer
* Many stable distros still deliver 2.7.4 which is about 7 years old...

pedromrg, 2023-08-11:
> @pedromrg : Can you please elaborate? There are many definitions of "latest version", alas.
> 
> * Custom build of current master (or some feature branch) from Git
> * Modern distro package that delivers 2.8.0 or newer
> * Many stable distros still deliver 2.7.4 which is about 7 years old...

Thank you for the clarification @jimklimov this kind replies to my question.

Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Generic HID driver 0.41 (2.7.4)

now I need to find out how to update this to a modern distro.

jimklimov, 2023-08-11:
Yes, "by our records" the 2.8.0 release should include a dedicated subdriver. There may be more changes in git head, gotta check by comparison...