networkupstools/nut #835: 2.7.4.1 e24b9a84 fails to connect to see UPS

cschuber, 2020-10-21:
FreeBSD 13-CURRENT
ups.model: UPS RF1025      
ups.productid: 0501

Config that worked correctly under 2.7.4 and under 2.7.4.1 e24b9a84 but now causes an error under 79a2f758:

listening on 10.1.2.254 port 3493
listening on 10.1.1.254 port 3493
listening on 127.0.0.1 port 3493
Can't connect to UPS [cps] (usbhid-ups-cps): No such file or directory

Same configuration:

[cps]   
        driver = usbhid-ups
        port = auto
        # pollfreq = 5
        pollfreq = 30
        desc = "cwfw"

The ups matches this in the nut supplied devd.conf:

#  Dynex DX-800U?, CP1200AVR/BC1200D, CP825AVR-G, CP1000AVRLCD, CP1000PFCLCD, CP
1500C, CP550HG, etc.  - usbhid-ups
notify 100 {
        match "system"          "USB";
        match "subsystem"       "DEVICE";
        match "type"            "ATTACH";
        match "vendor"          "0x0764";
        match "product"         "0x0501";
        action "chgrp uucp /dev/$cdev; chmod g+rw /dev/$cdev";
};

cschuber, 2020-10-21:
usbhid-ups is dying on a bus error at dstate.c:583.

Program received signal SIGBUS, Bus error.
0x000000000021faa2 in dstate_poll_fds (timeout=..., extrafd=-1) at dstate.c:583
583		FD_SET(sockfd, &rfds);
(gdb) bt
#0  0x000000000021faa2 in dstate_poll_fds (timeout=..., extrafd=-1)
    at dstate.c:583
#1  0x000000000021efbb in main (argc=0, argv=0x7fffffffea48) at main.c:749
(gdb) p sockfd
$1 = -1
(gdb) p rfds  
$2 = {__fds_bits = {0 <repeats 16 times>}}
(gdb) p &rfds
$3 = (fd_set *) 0x7fffffffe5b8
(gdb)

clepple, 2020-10-22:
@aquette I'm thinking there might be a typo in 73c8644d5

`dump_data` seems to be initialized to 0, but this checks for `< 0` before calling `dstate_init()`:

https://github.com/networkupstools/nut/commit/73c8644d53d7dbaae46ec9d3b6bbee9ad23f9f70#diff-66826a7a6e7b34b0ac578931f6570583c73e5ca608faf9dda647ba561daa778dR705

cc @jimklimov since it sounds like your testing concerns on #820 are relevant

cschuber, 2020-10-22:
Yes, reverting 73c8644 resolves the issue.

I'll apply a patch reverting 73c8644 from the FreeBSD nut-devel port.

aquette, 2020-10-22:
@clepple indeed, confirmed. I'll make an amended PR. sorry for the issue guys, and thanks @cschuber for raising it

closed 2020-10-23 by jimklimov