networkupstools/nut #839: genericups but over GPIO?

dogtopus, 2020-10-27:
It would be nice to support this so devices without an RS232 port but has GPIO pins like Raspberry Pi could use genericups driver directly without a e.g. FTDI cable.

jimklimov, 2020-11-05:
For Raspberry Pi in particular, there is a PiJuice HAT driver (GPIO/i2c) merged recently with #730 

We also have one other i2c driver it seems (asem)

Otherwise, lacking the hardware to test at least some use-cases (including UPS-like hardware that has GPIO interface) I'm afraid I can only say "PRs welcome" :\

dogtopus, 2020-11-05:
> including UPS-like hardware that has GPIO interface

Well that would probably be the "next step". What I have in mind only applies to contact closure type UPSes. Basically if they have open-drain-like output (like using relays) then it could be possible to use GPIO input pins with pull-up resistors instead of RS232 flow control pins to at least receive the status signals from those UPSes.

clepple, 2020-11-08:
@dogtopus I think what Jim is getting at is that he doesn't have the hardware available (either a contact-closure UPS or something similar that uses GPIO) to test such a driver, were he to write it.

I have a SleepyPi that I am mostly using for the RTC, but it could probably be a good testbed for a GPIO driver: https://spellfoundry.com/docs/getting-the-sleepy-pi-to-shutdown-the-raspberry-pi/

That said, my time is fairly limited at the moment, and I don't want to give the impression that I can work on this until at least after we get all of the existing code tested for the next NUT release.

(Someone who is decent at Arduino coding could probably whip up something quickly for the SleepyPi CPU that talks to one of the existing NUT drivers like `nutdrv_qx` over serial.)

clepple, 2020-11-11:
Notes for future self, or someone else who has time to look at this now:

https://www.acmesystems.it/gpiod says "Since linux 4.8 the GPIO sysfs interface is deprecated."