networkupstools/nut #937: bcmxcp does not disable beeper with Powerware 9125

apankevych, 2020-12-16:
'upscmd' disable command says 'OK' but it does not change the status:

`upsc pw9125
ambient.temperature: 28
ambient.temperature.high: 52
battery.charge: 100
battery.charger.status: floating
battery.energysave.delay: 5
battery.energysave.load: 10
battery.packs: 0
battery.runtime: 6238
battery.runtime.low: 2
battery.voltage:  84.1
device.mfr: Eaton
device.model: PW9125          3000i
device.part: 103002723-5501  
device.serial: RX235A0480      
device.type: ups
driver.name: bcmxcp
...
ups.beeper.status: enabled`

'mute' status can be set but beeper is still ON.

apankevych, 2020-12-16:
System: Archlinux
community/nut 2.7.4-3
extra/linux-zen 5.9.9.zen1-1

apankevych, 2020-12-16:
It seems it can be related to https://github.com/networkupstools/nut/issues/9.

jimklimov, 2020-12-17:
Is this with a packaged release 2.7.4 (sadly quite aged, working on that) - not with a recent Git revision?

apankevych, 2020-12-17:
Correct, this is with packaged release 2.7.4. I'll try with git revision as well.

apankevych, 2020-12-18:
@jimklimov Doesn't work with recent Git revision neither(

The same behavior:
```
~> sudo upscmd -u upsuser -p password pw9125 beeper.disable
OK
~> sudo upsc pw9125 ups.beeper.status
enabled
~> sudo upscmd -u upsuser -p password pw9125 beeper.mute
OK
~> sudo upsc pw9125 ups.beeper.status
enabled
```
Version: Network UPS Tools - UPS driver controller 2.7.4-2060-gb7374ea6