networkupstools/nut #949: Synology cannot connect to network ups running macosx-ups (missing variables)

smagoun, 2020-12-29:
I have a Synology NAS and a Mac that are both powered from a UPS. The Mac is connected to the UPS via USB, and I've set up nut with the `macosx-ups` driver so I can connect the NAS as a nut slave. I have everything working at the command line level, but the Synology web interface complains that it cannot connect to the UPS server. I traced the problem to the lack of `ups.mfr` and `ups.model` variables in the `macosx-ups` driver. The Synology web interface is hardcoded to say the network UPS server is not available if these variables aren't set.

A fix is to add the `ups.mfr` and `ups.model` variables to the output of the `macosx-ups` driver. I'll put up a PR.

**macosx-ups output - not working w/ Synology:**
```
Aion:nut steve$ upsc ups
battery.charge: 100
battery.runtime: 3240
battery.voltage: 122.000
device.mfr: (unknown)
device.model: UPS CP1000AVRLCD
device.type: ups
driver.name: macosx-ups
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.internal: 1.2
ups.status: OL
```

**macosx-ups output - works w/ Synology:**
```
Aion:nut steve$ upsc ups
battery.charge: 100
battery.runtime: 3240
battery.voltage: 122.000
device.mfr: (unknown)
device.model: UPS CP1000AVRLCD
device.type: ups
driver.name: macosx-ups
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.internal: 1.2
ups.mfr: (unknown)
ups.model: UPS CP1000AVRLCD
ups.status: OL

```

clepple, 2020-12-29:
Thanks for tracking this down and providing a patch. I think we can merge this, though I'd consider it a bug in the Synology software since the `device.*` names have been documented as preferred for more than a decade. For tracking purposes, what version of the Synology software are you running?

For others who don't want to rebuild, it should be possible to manually add the mfr/model in `ups.conf` with `default.device.mfr: (unknown)` and `default.device.model: (unknown)`  (as described in the [man page](https://networkupstools.org/docs/man/ups.conf.html#_ups_fields)).

closed 2020-12-29 by clepple

smagoun, 2020-12-29:
The Synology NAS is running DSM 6.2.3-25426 Update 3. I agree this is really a Synology issue, and I've raised it with them.

I'd missed the `default.*` option, thank you! I set `default.ups.mfr` and `default.ups.model` in my `ups.conf`, and it worked - much easier than rebuilding nut. The only catch is that setting those values overwrites the `default.device.mfr` and `default.device.model` set by the driver. I don't think this is a big deal, but it surprised me.

There is a MacOS how-to for nut floating around, which I've forked+updated with instructions for setting up for use with Synology clients: https://gist.github.com/smagoun/7e976ae43d5e64a81e26d55ef3f94b44

Thank you for the help!

clepple, 2020-12-29:
Hmm, `default.ups.mfr` should not override `device.mfr` - it is meant to be a fallback in case the driver doesn't set a value for that variable. I wonder if the order is incorrect in the code that copies `device.mfr` from `ups.mfr`.

If it is more of a documentation thing, I'm going to try to find some time to rework the `default.*` and `override.*` documentation: #832 

Thanks for posting that howto. It might be easier to just do `LISTEN 0.0.0.0 3943` - if you need to exclude one interface, you'd list 127.0.0.1 and the other IPs. But a lot of Macs are probably doing DHCP and mDNS, so just listening on `0.0.0.0` should avoid any timing problems if the network isn't completely up when `upsd` starts.

commit 699bf8b4 (2020-12-29): drivers/macosx-ups.c: Add ups.mfr and ups.model legacy vars, fixes #949