networkupstools/nut #951: fightwarn: some code using SIZE_MAX fails on FreeBSD

clepple, 2020-12-29:
It looks like `server/upsd.c` might need another include file.

* http://buildbot.networkupstools.org/public/nut/builders/FreeBSD-x86/builds/1070
* http://buildbot.networkupstools.org/public/nut/builders/FreeBSD-x64/builds/694

Code: https://github.com/networkupstools/nut/commit/ebc6c42d48df2eed16b4d982035b6cff18789342#diff-f173de61621c390fe1dc46304983282d7771a8eb519b8376bc5fb3d4a64148b7R728

jimklimov, 2021-01-06:
That proved to be an interesting puzzle, seems the BSD headers do define the value but only optionally https://github.com/freebsd/freebsd-src/blob/master/sys/x86/include/_stdint.h - and it seems that by default the stars did not align for that.

Experimenting a fix along these lines:
* https://github.com/networkupstools/nut/commit/3bbf6634fab1000b19e435323cad88407b881944
* https://github.com/networkupstools/nut/commit/a7786ae27d5fd610cc5fbb0d2e274e7c87e31418

jimklimov, 2021-01-07:
Also, on one round I saw several "warning: ignoring prerequisites on suffix rule definition" lines, about un(dir)named Makefile lines 523, 528 and 533 on a BSD worker, but it was not reproduced in later iterations.

According to https://www.gnu.org/software/make/manual/html_node/Error-Messages.html it means that we had a suffix rule, and that rule also specified prerequisites which is invalid for POSIX make. I did not locate any such rules (looked for `%`) in our Makefile.am templates, so not sure where that came from... some mis-processing by autotools?..

clepple, 2021-01-07:
Thanks for tracking down the `SIZE_MAX` issue.

I do remember seeing the suffix rule warning, but I did not make a note of it. If it comes up again, we can deal with it then. (The FreeBSD ports tree uses - or at least, has used in the past - GNU Make in its build recipe for NUT.)

closed 2021-12-01 by jimklimov