networkupstools/nut #952: HCL Cyber Power OR500LCDRM1U supported by usbhid-ups just like OR700LCDRM1U

thomasvs, 2020-12-31:
I couldn't figure out how to run upsc - I saw no way to specify username and password I configured my nut server with.

It's working fine for me with nut running in hass.io and homeassistant using the nut integration to query it.

clepple, 2020-12-31:
username/password are not needed to see UPS stats, only to change variables (`upsrw`) or execute commands, such as test and shutdown commands (`upscmd`).

Have you tested the shutdown sequence? Sounds like it should work, but it's always good to confirm that nothing under the hood has changed, and that's why we appreciate the `upsc` output. https://networkupstools.org/stable-hcl.html#footnotes

clepple, 2020-12-31:
Looks like someone has already provided some of this information, so I'm going to close this in favor of #578. If you see any major differences between your OR500 and the earlier one, please reopen this issue.

closed 2020-12-31 by clepple