networkupstools/nut #953: Cannot shutdown APC SU-1000 with AP9617

dwmw2, 2021-01-03:
When I attempt to shut down I get an error:
```
   0.121544	Initiating UPS shutdown
   0.121547	[D1] upsdrv_shutdown...
   0.121552	[D2] entering su_setOID(instcmd, shutdown.return, (null))
   0.121559	[D3] su_find_info: "shutdown.return" found
   0.121565	[D1] entering nut_snmp_set(.1.3.6.1.4.1.318.1.1.1.6.1.1.0, t, 200)
   0.151626	[su1000] nut_snmp_set: can't set .1.3.6.1.4.1.318.1.1.1.6.1.1.0: Error in packet: wrongType (The set datatype does not match the data type the agent expects)
   0.151632	[D1] su_setOID: cannot execute command 'shutdown.return'
```

Using v2.7.4 from Fedora 33 the error is different:
```
   0.086717	Initiating UPS shutdown
   0.086720	upsdrv_shutdown...
   0.086726	Unknown template type: shutdown.return
   0.086732	entering su_instcmd(shutdown.return, (null))
   0.086740	su_find_info: "shutdown.return" found
   0.086745	entering nut_snmp_set(.1.3.6.1.4.1.318.1.1.1.6.1.1.0, i, 2)
   0.127311	[su1000] nut_snmp_set: can't set .1.3.6.1.4.1.318.1.1.1.6.1.1.0: Error in packet: (genError) A general failure occured
   0.127321	su_instcmd: cannot set value for shutdown.return
```

```
 $ snmpget  -u nut 192.168.122.104 .1.3.6.1.4.1.318.1.1.1.6.1.1.0
SNMPv2-SMI::enterprises.318.1.1.1.6.1.1.0 = INTEGER: 1

 $ snmpset  -u nut 192.168.122.104 .1.3.6.1.4.1.318.1.1.1.6.1.1.0 i 2
Error in packet.
Reason: (genError) A general failure occured
Failed object: SNMPv2-SMI::enterprises.318.1.1.1.6.1.1.0

 $ snmpset  -u nut 192.168.122.104 .1.3.6.1.4.1.318.1.1.1.6.1.1.0 t 200
Error in packet.
Reason: wrongType (The set datatype does not match the data type the agent expects)
Failed object: SNMPv2-SMI::enterprises.318.1.1.1.6.1.1.0
```
Using SNMPv1 doesn't make much difference.

```
 $ snmpset  -v1 -c private 192.168.122.104 .1.3.6.1.4.1.318.1.1.1.6.1.1.0 i 2
Error in packet.
Reason: (genError) A general failure occured
Failed object: SNMPv2-SMI::enterprises.318.1.1.1.6.1.1.0

 $ snmpset  -v1 -c private 192.168.122.104 .1.3.6.1.4.1.318.1.1.1.6.1.1.0 t 200
Error in packet.
Reason: (badValue) The value given has the wrong type or length.
Failed object: SNMPv2-SMI::enterprises.318.1.1.1.6.1.1.0
```

dwmw2, 2021-01-03:
Full output of `./snmp-ups -a su1000 -DDDDD   -k`: [shutdown-attempt.txt](https://github.com/networkupstools/nut/files/5761234/shutdown-attempt.txt)

Driver running with debugging: [driver-run.txt](https://github.com/networkupstools/nut/files/5761238/driver-run.txt)

```
 $ upsc su1000
ambient.1.humidity.alarm.high: 60
ambient.1.humidity.alarm.low: 30
ambient.1.temperature.alarm.high: 40
ambient.1.temperature.alarm.low: 10
battery.charge: 100
battery.charge.restart: 0
battery.current: 0
battery.date: 05/03/20
battery.packs: 0
battery.packs.bad: -1
battery.runtime: 2580
battery.runtime.low: 120
battery.voltage: 27.60
battery.voltage.nominal: -1
device.mfr: APC
device.model: SMART-UPS 1000
device.serial: QS0023121408
device.type: ups
driver.name: snmp-ups
driver.parameter.pollfreq: 15
driver.parameter.pollinterval: 2
driver.parameter.port: 192.168.122.104
driver.parameter.snmp_version: v3
driver.parameter.synchronous: no
driver.version: 2.7.4-2077-g699bf8b4
driver.version.data: apcc MIB 1.3
driver.version.internal: 1.12
input.frequency: 50
input.sensitivity: low
input.transfer.high: 253
input.transfer.low: 196
input.transfer.reason: selfTest
input.voltage: 237.90
input.voltage.maximum: 240.50
input.voltage.minimum: 236.60
output.current: 0
output.frequency: 50
output.voltage: 237.90
output.voltage.nominal: 230
ups.delay.shutdown: 20
ups.delay.start: 0
ups.firmware: 60.11.I
ups.id: SU 1000
ups.load: 3.60
ups.mfr: APC
ups.mfr.date: 06/03/00
ups.model: SMART-UPS 1000
ups.serial: QS0023121408
ups.status: OL
ups.temperature: 27.40
ups.test.date: 01/02/2021
ups.test.result: Ok

```

jimklimov, 2025-12-07:
Interesting, looking at various representations of the MIB, I see e.g. https://mibs.observium.org/mib/PowerNet-MIB/#upsBasicControlConserveBattery for `.1.3.6.1.4.1.318.1.1.1.6.1.1` (with no trailing `.0`) described there or in https://networkupstools.org/protocols/snmp/APC-Powernet.pdf as:

> The [upsBasicControl] category has one OID, which any Management Card or PowerNet Agent can use to put a UPS that is running on battery into “sleep mode.”
> [upsBasicControlConserveBattery] Cause a UPS running on battery to turn off its outlets to conserve battery runtime and then wait in “sleep mode” until acceptable input power returns.
> • noTurnOffUps (1): The value always returned for a GET. Setting this value has no effect.
> • turnOffUpsToConserveBattery (2): The UPS, if running on battery, waits in “sleep mode” until acceptable input power returns. If the UPS is not on battery, a badValue error is returned.

There is also an advanced tree with 9 operations as of https://mibs.observium.org/mib/PowerNet-MIB/#upsAdvControl seen as nodes under `.1.3.6.1.4.1.318.1.1.1.6.2` and many of those are seen in `drivers/apc-mib.c` also trailed with `.0` (so maybe that is supposed to be a shortcut for "node itself"? and maybe it is not ubiquitous?)

This spelling stems from ancient history like commit 78bdf003ab so apparently this worked at some point?..

@nahez : can you please re-test the `snmpset` experiment without a trailing `.0` in the OID? I suppose we can teach `snmp-ups` to fall back to chopping off such tail? I do not know OTOH if that would be a *correct* thing to do though.

CC @aquette : any ideas here, please? :-}

nahez, 2025-12-07:
Hello, thank you so much for the prompt reply. I am not familiar with how all this works but I tried the snmpset command for the OID without the trailing ".0" and a few other things. I am not too sure if they are useful but here they are.

snmpset -v2c -c private 192.168.151.2 .1.3.6.1.4.1.318.1.1.1.6.1.1 i 2  
Error in packet.
Reason: notWritable (That object does not support modification)
Failed object: iso.3.6.1.4.1.318.1.1.1.6.1.1

snmpwalk -v2c -c private 192.168.151.2 .1.3.6.1.4.1.318.1.1.1.6.1.1
iso.3.6.1.4.1.318.1.1.1.6.1.1.0 = INTEGER: 1
snmpwalk -v2c -c private 192.168.151.2 .1.3.6.1.4.1.318.1.1.1.6.1  
iso.3.6.1.4.1.318.1.1.1.6.1.1.0 = INTEGER: 1
nmpwalk -v2c -c private 192.168.151.2 .1.3.6.1.4.1.318.1.1.1.6  
iso.3.6.1.4.1.318.1.1.1.6.1.1.0 = INTEGER: 1
iso.3.6.1.4.1.318.1.1.1.6.2.1.0 = INTEGER: 1
iso.3.6.1.4.1.318.1.1.1.6.2.2.0 = INTEGER: 1
iso.3.6.1.4.1.318.1.1.1.6.2.3.0 = INTEGER: 1
iso.3.6.1.4.1.318.1.1.1.6.2.5.0 = INTEGER: 1
iso.3.6.1.4.1.318.1.1.1.6.2.6.0 = INTEGER: 1
iso.3.6.1.4.1.318.1.1.1.6.2.7.0 = INTEGER: 1
iso.3.6.1.4.1.318.1.1.1.6.2.8.0 = INTEGER: 1
iso.3.6.1.4.1.318.1.1.1.6.2.9.0 = INTEGER: 1

Since you mentioned mib, I looked it up what that is. I also found a mib file listed named "powernet458.mib" under this cards support page. I wonder if this is the structure definitions for this card and if this is useful?

[powernet458.txt](https://github.com/user-attachments/files/24016957/powernet458.txt)

Here is the support page for this card. 

https://www.se.com/ca/en/product/AP9631/ups-network-management-card-2-with-environmental-monitoring/

thanks again for looking into this.

jimklimov, 2025-12-07:
Yes, it is the formal definition of how different OID nodes in the SNMP data tree are named and related (as a further branch or leaf node contained in another). You can find the `upsBasicControlConserveBattery` definition in the text.

I googled around, and it seems the `snmpset <OID>.0` is a commonly used form, and failures are usually due to either lack of support in the firmware, or lack of permission to use a certain command (and/or write a variable). Perhaps double-check the SNMP card configuration, if it allow to manage constraints like this (e.g. community `private` may be not all-powerful by default?)

Generally, in relatively newer devices there may be an inclination to frown upon SNMP v1/v2 (almost lack of) security model, and to allow access to juicy features only with cryptographically stronger SNMPv3 (with real user accounts or equivalent concept). So one more idea to test is to try that, whether with `snmpset` or NUT.

nahez, 2025-12-07:
Already tried updating permissions for community private. snmpset works for something like device location. That's why in my upsc output the device.location is "hello".

One of the output messages said "No matching MIB found for sysOID '.1.3.6.1.4.1.318.1.3.27'!" So I took a guess that maybe because this card is not officially supported, iso.3.6.1.4.1.318.1.1.1.6.1.1.0 is just a fallback and is incorrect for this card. I didn't know exactly how to check if iso.3.6.1.4.1.318.1.1.1.6.1.1.0 is the correct value to set for shutdown, so I thought maybe I can come here to confirm. If someone can confirm this is indeed the correct OID to set for shutdown, then I believe you maybe right that this might be blocked on firmware level as I can set other values like location.

nahez, 2025-12-19:
Sorry I meant would this be considered a bug in the driver that the devs could consider updating or is the driver considered working as intended and I need to somehow update this myself. Thanks!