networkupstools/nut #955: Tripplite SMART500RT1U (09ae:3026) support

Sandman614, 2021-01-03:
I purchased a Tripplite SMART500RT1U which NUT HCL shows as supported.  I am running TrueNAS Core 12.0-U1 which has NUT functionality.  The UPS identifies immediately when plugged in with USB, driver set to usbhid-ups, port doesn't matter but set to AUTO.  It looks like this productid (09ae:3026) needs added to the driver. Until that happens can I add the -x ProductID somewhere to make it run correctly?

Problem: UPS not yet supported
% sudo upsdrvctl start
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
Duplicate driver instance detected! Terminating other driver!
This TrippLite device (09ae:3026) is not (or perhaps not yet) supported
by usbhid-ups. Please make sure you have an up-to-date version of NUT. If
this does not fix the problem, try running the driver with the
'-x productid=3026' option. Please report your results to the NUT user's
mailing list <nut-upsuser@lists.alioth.debian.org>.

No matching HID UPS found
Driver failed to start (exit status=1)

Driver ran with productid as suggested
% sudo /usr/local/libexec/nut/usbhid-ups -a SMART500RT1U -x productid=3026
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
Using subdriver: TrippLite HID 0.82

Sandman614, 2021-01-03:
David from the NUT mailing list offered the following solution which works.  In Truenas Core you can add the line to the "Auxiliary Parameters (ups.conf)" field of the UPS entry.

Add 'productid = 3026' to UPS.conf
https://alioth-lists.debian.net/pipermail/nut-upsdev/2020-June/007473.html

eyal0, 2021-05-04:
I have a TrippLite and a solution like this one worked for me.

In my `/et/nut/ups.conf` file I have:

[myups]
        driver = usbhid-ups
	productid = 3024
        port = auto
        desc = "Workstation"

"3024" is correct for my TrippLite but for yours, use 3026 probably.

ajschorr, 2024-02-16:
I'm having the same issue. I was able to get it approximately working by patching ups.conf as
suggested with "productid = 3026", but I also needed to patch /usr/lib/udev/rules.d/62-nut-usbups.rules
to add the following line in order to fix the permissions problem:

ATTR{idVendor}=="09ae", ATTR{idProduct}=="3026", MODE="664", GROUP="dialout"

Without that, I was getting this error:

libusb1: Could not open any HID devices: insufficient permissions on everything

I guess that could also have been addressed by setting "user = root".

But after this, I'm getting bogus values for output.voltage, and I'm getting no value for battery.runtime. Are other
people having more like? I guess that just as protocol 3024 required some patching, 3026 does as
well.

jimklimov, 2024-02-16:
With NUT v2.8.1 or newer, technically you can specify a new `subdriver` option (see `usbhid-ups -h` for details) to force matching a particular subdriver with your device, even if the IDs are not built into it. At least, this way you can quickly check if the protocols are sufficiently close (reasonable data appears).

That said, I'm not fully sure how it would behave here with the additional modifiers like `smart1500lcdt_scale` in the example line above (probably would use NULL in their place).

To build and test a customized driver, you can follow https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

ajschorr, 2024-02-16:
I've got version 2.8.0 on CentOS Stream 9, but I guess I can upgrade to 2.8.1.
Thanks for the pointer!
[smart500rt1u.txt](https://github.com/networkupstools/nut/files/14312727/smart500rt1u.txt)

ajschorr, 2024-02-16:
For the record, without modifying the code at all, I did see that the ups.load value rose above zero once I plugged
in some additional devices. I guess it's pretty granular and was rounding down to zero when I had only a  couple of
items plugged in. With respect to battery.runtime, I'm not optimistic that this UPS is actually reporting that info.
In drivers/tripplite-hid.c, I see that this item is simply copied from UPS.PowerSummary.RunTimeToEmpty, but
that item is not present in the debug output log that I shared. So I guess it's simply not available from this
low-end UPS. So I'm going to leave well enough alone.

ajschorr, 2024-02-16:
The output.voltage is not scaled properly, but I don't care so much about that value. I'll just assume that it's
correct. For the record, it looks like output.voltage for this unit would come from UPS.PowerConverter.Output.Voltage.
It's reporting 1.171e-05, and presumably should be in the 120 range, so should be scaled by a million, just
as in smart1500lcdt_scale. On the other hand, the battery voltage does not seem to need to be scaled,
so this unit can't simply use smart1500lcdt_scale. And the frequencies don't need to be scaled either. The
current scale looks roughly correct also.