networkupstools/nut #957: riello_usb (& riello_ser ) segmentation fault on missing vars

ianb61, 2021-01-12:
Hi,
I have installed nut on Debian 10 (Buster) from the release apt package. I have the riello_usb driver installed and configured and it works fine most of the time. When I use the instcmd ( upscmd ) to request a shutdown.return the riello_usb driver crashes with a segmentation fault.
I downloaded the source from github and chased the fault down to the driver requesting the ups.delay.shutdown variable. This variable is not established from my ups, so it doesn't exist unless specifically created by the user in ups.conf. Without the variable the riello_usb code is given a NULL pointer from the dstate_getinfo() call and this is passed into atoi( ) without a check, which then causes the crash. Also uses the variable for command load.off.delay and similarly crashes.
Variable ups.delay.reboot causes the same behaviour from the command load.on.delay.

Easy fix for me was to define the variables in my ups.conf. But life would be better if this was documented in the man page for riello_usb or fixed in the code so it handles the situation cleanly. I am new to github so I dont know how to make such fixes or manual updates. Happy to help if I can or if this is a pointless issue then please close it off and I am sorry to trouble you.
Thanks
Ian

biergaizi, 2021-01-29:
I'm not a project developer, but I think it's a real problem. I'm working on the driver for my own UPS for two weeks so I've read many drivers for my reference. It's my personal observation that there are a lot of unsafe uses of `atoi()` throughout the code (#966) that I'd like to fix. Also, I noticed that there is currently no standard to set startup and shutdown delays, and this is also responsible for your problem, so I opened #972 for that. Let's wait for comments from project maintainers.

geoghegan, 2022-02-11:
`ups.delay.reboot` and `ups.delay.shutdown` now (PR #1280) have a default value of 5 seconds, and can be set without a segfault.

I have an IDG1600 connected over usb, @ianb61 . shutdown.return works ok for me - no segfault using the updated version of the riello_usb driver (version 0.7)

```
upscmd -u admin -p ups ups shutdown.return
OK
```

closed 2022-02-17 by jimklimov