networkupstools/nut #958: Windows installation and usage manual is required

almol, 2021-01-14:
Hi.

I have a Windows 10 PC connected to UPS with serial port and I am trying to monitor it.

I download: http://www.networkupstools.org/package/windows/NUT-Installer-2.6.5-6.msi 
And install... And that's all... what are the next steps?
I can't find more indications to use...

jimklimov, 2021-01-16:
Hello, I am unfortunately not well acquainted with the latest builds of NUT
for Windows (which were based on 2.6.x releases so nearing a decade old;
there is a long-backlogged desire to re-achieve that with modern codebase).

I did use some NUT build for Windows when I was a mere community member,
and at that time it only included the client side: actual UPS monitoring
was done by Unix servers, and the Windows machines listened to their NUT
protocol server with upsmon or equivalent for doing safe shutdowns. Setup
was similar to that on Unix/Linux.

That's about as much as I can say now, hopefully someone else has a more
comprehensive and recent experience with NUT on Windows to answer better.

git3smurf, 2021-01-17:
Hi.

There is a readme.txt file in the WinNut folder that is created [ C:\Program Files (x86)\WinNUT ]  when you install the msi.
It is rather concise though, and it still mentions windows 9x....    :)

In **upsmon.conf** also in the WinNut folder which you have to edit to get it working there is some additional help. 

I use WinNut on several Windows 10 computers to monitor a UPS-service on a Truenas (as a slave), and it works fine. 

The link in the readme file from one of the (former?) maintainers (Andrew Delpha) to [http://csociety.ecn.purdue.edu/~delpha/winnut/](url) does not work anymore. Maybe someone knows how to restore the link to this resource?

clepple, 2021-01-23:
> it still mentions windows 9x.... :)

old documentation doesn't update itself :-)

As Jim mentioned, those builds are from an old branch of NUT (2.6.x vs 2.7.x) and given all of the complaints from users that we haven't made a new 2.7.x release in a while, I don't think we're going to have time to redo the Windows port in the near future. See the other issues tagged [Windows](https://github.com/networkupstools/nut/issues?q=is%3Aissue+is%3Aopen+label%3Awindows), especially #5, for more information.

closed 2021-01-23 by clepple

jimklimov, 2022-09-05:
FWIW, it seems the Q&A were about different products. MSI was from NUT's own effort as of 2.6.5, and WinNUT is (AFAIK) a separate project, initially abandoned in 2006 at version 2.0.0b and then re-hosted at Google Code in 2010 and abandoned in 2011. There is also a separate(?) https://github.com/gawindx/WinNUT-Client last touched Feb 2021.

The URL mentioned above was last archived at https://web.archive.org/web/20161222060711/http://csociety.ecn.purdue.edu/~delpha/winnut/ and refers to https://code.google.com/archive/p/winnut/ as the codebase (still available); not sure if there were further iterations on other code hosting.

Curiously, there are recent articles on the name (actually describing WinNUT-Client): https://www.gadgetvictims.com/2022/02/configuring-winnut-2x-on-windows-to.html as a follow-up to https://www.gadgetvictims.com/2010/02/battery-backup-solution-for-home.html (describing original WinNUT)