networkupstools/nut #960: alarm_commit() silently truncates a 1024-char message to 256 characters

biergaizi, 2021-01-19:
Currently, an alarm message is limited to 1024 characters (`LARGEBUF`), a developer may check and source and see `alarm_buf[LARGEBUF]` and naively assuming the alarm buffer is adequate.

    static char     status_buf[ST_MAX_VALUE_LEN], alarm_buf[LARGEBUF];

However, when `alarm_commit()` is called, it attempts to create a UPS variable, but a UPS variable is only limited to 256 characters (`ST_MAX_VALUE_LEN`), far below the 1024-character `alarm_buf`. If the UPS alarm string is too long, it's silently truncated without any warning.

    /* write the status_buf into the info array */
    void alarm_commit(void)
    {
            if (strlen(alarm_buf) > 0) { 
                    dstate_setinfo("ups.alarm", "%s", alarm_buf);
                    alarm_active = 1; 
            } else {
                    dstate_delinfo("ups.alarm");
                    alarm_active = 0; 
            }
    }

This is unexpected from a developer's point of view.

1. The alarm message is truncated when it's committed as a variable even if the alarm buffer itself is large enough. And when `alarm_commit()` truncates an alarm, it doesn't return any error code.

2. Nowhere in the documentation has a warning about the length of the alarm string. In fact, the documentation treats a `alarm_set()` call as something different from a `dstate_setinfo()`, without mentioning the fact that `alarm_set()`'s underlying implementation is a  `dstate_setinfo()` and the same restrictions apply.

## Proposed Solution(s)

1. Raise the length limit of a variable to 1024, the same size of `alarm_buf`, or
2. Lower the limit of the `alarm_buf` to 256, the same size of `status_buf`.
3. Don't silently truncates an alarm message, at least returns an error code, or
4. Keep the existing code, but improve the documentation `new-drivers.txt` to warn the developers about these limitations.

jimklimov, 2021-03-12:
I agree this is an unexpected trap. For the least surprise - being generally not sure what consumers, including external libs and binaries, exist for a dstate status variable, a safe bet currently seems to be the option number 2 and get these buffer sizes in sync, and maybe add debug messages at least for overflows in `alarm_set()`.

It seems that @aquette was the last to touch all lines with `alarm_buf` so I hope he's available to comment here.

Just looking at code, `alarm_set()` uses `snprintfcat()` to combine maybe several alarms into one (space-separated) string, and returns the length of the result as it would be if not truncated, or a negative value upon error (according to `vsnprintf()` used and its documented `snprintf()` behaviors). So indeed, this return value can be compared with `sizeof(alarm_buf)` to at least log about the lost alarms (maybe log their content too, though in case of some unterminated string and arbitrary overflow it does not seem like a safe idea), or to booleanize the function so it can report if the alarm-setting succeeded, if any callers were to check for that.

jimklimov, 2021-03-12:
@biergaizi : does PR #986 look like a reasonable solution (without API change from `void` to some sort of checkable status, so far)?

jimklimov, 2021-03-14:
@biergaizi : thanks for raising this issue, now it is hopefully solved satisfactorily :)

jimklimov, 2021-03-14:
Note: per current solution, the routines still return `void`. Is there a good case practically for returning a success/failure code of some sort (bool, snprintf-style, something)?

closed 2021-03-14 by jimklimov